openwrt/packages #17496: node: can not compile on host debian11 amd64

svlobanov, 2022-01-04:
Maintainer: @nxhack  
Environment: target: armvirt/64, OpenWrt version master 4a2cca78245e9291096e7c8c98627426df50ef58
host: debian11 amd64
packages version: b434cd71b950185cbf57308466eb0e3473ba484e

Description:

```
node build failed on debian11 amd64 for armvirt/64 target

$ make -j1 package/feeds/packages/node/compile V=sc
...
/home/sergey/openwrt2/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/sergey/openwrt2/build_dir/target-aarch64_cortex-a53_musl/node-v14.18.2/out/Release/obj.target/tools/v8_gypfiles/libv8_base_without_compiler.a: error adding symbols: malformed archive
collect2: error: ld returned 1 exit status
make[4]: *** [mkcodecache.target.mk:176: /home/sergey/openwrt2/build_dir/target-aarch64_cortex-a53_musl/node-v14.18.2/out/Release/mkcodecache] Error 1
rm 7b5fc5ee81ce088234765b79064235bf25dcbeb8.intermediate c5e5ccdf03bb1a32d27790790235168774ff2e70.intermediate c0111d5b0c3d72ad0610b58b1e42f313bd201dbb.intermediate d5363770ebb9ca494965a0c4d84db35637d42e62.intermediate d2f8b8e1c4bef2e5a5403aced1738cc5d1553afc.intermediate f9f8ef52de441648f5fff1919a8505995bebcd9d.intermediate
make[3]: *** [Makefile:109: node] Error 2
make[3]: Leaving directory '/home/sergey/openwrt2/build_dir/target-aarch64_cortex-a53_musl/node-v14.18.2'
make[2]: *** [Makefile:160: /home/sergey/openwrt2/build_dir/target-aarch64_cortex-a53_musl/node-v14.18.2/.built] Error 2
make[2]: Leaving directory '/home/sergey/openwrt2/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#33.06#2.36#35.18
    ERROR: package/feeds/packages/node failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/node/compile] Error 1
make[1]: Leaving directory '/home/sergey/openwrt2'
make: *** [/home/sergey/openwrt2/include/toplevel.mk:230: package/feeds/packages/node/compile] Error 2
```

build host
```
sergey@deb11k:~/openwrt2$ uname -a
Linux deb11k 5.10.0-10-amd64 #1 SMP Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux
sergey@deb11k:~/openwrt2$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

.config and log build attached
[config.txt](https://github.com/openwrt/packages/files/7808878/config.txt)
[node.log](https://github.com/openwrt/packages/files/7808879/node.log)

svlobanov, 2022-01-04:
Github CI is able to build node https://github.com/svlobanov/packages/runs/4707270364?check_suite_focus=true

I've tried on clean installation of Debian11 amd64 and Ubuntu 20.04 amd64, but it fails. I can not understand why

nxhack, 2022-01-05:
@svlobanov 

Please refer here.
https://github.com/openwrt/packages/issues/16729
https://github.com/openwrt/packages/issues/17164

svlobanov, 2022-01-05:
https://github.com/openwrt/packages/issues/16729 helpded me (ulimit -n 2048). Thank you. I'm closing this ticket

closed 2022-01-05 by svlobanov

brada4, 2022-01-05:
It is actually a bug in gcc ld (binutils) https://sourceware.org/bugzilla/show_bug.cgi?id=28138
The diff in the thread is quite simple if you dont add the new testcase.

nxhack, 2022-01-08:
@svlobanov @brada4 
A patch has been applied to binutils 2.37. Please check it.
https://github.com/openwrt/openwrt/commit/8c501bf9feeaadfb9d950806a312b328baabddcb

brada4, 2022-01-09:
works for me.