openwrt/packages #17501: transmission-daemon: crash

nitroshift, 2022-01-05:
Maintainer: @\<dangowrt>
Environment: x86_64, OpenWRT master

Description:

Transmission-daemon crashes when adding a torrent with the folowwing errors:

"Wed Jan  5 12:44:25 2022 daemon.err transmission-daemon[20935]: [2022-01-05 10:44:25.169] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Wed Jan  5 12:44:26 2022 daemon.err transmission-daemon[20935]: [2022-01-05 10:44:26.159] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Wed Jan  5 12:45:10 2022 daemon.err transmission-daemon[20935]: [2022-01-05 10:45:10.159] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Wed Jan  5 12:45:10 2022 daemon.err transmission-daemon[20935]: [2022-01-05 10:45:10.159] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Wed Jan  5 12:45:42 2022 daemon.err transmission-daemon[20935]: [2022-01-05 10:45:42.570] Couldn't save temporary file "/mnt/xxx/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)
Wed Jan  5 12:45:42 2022 daemon.err transmission-daemon[20935]: [2022-01-05 10:45:42.570] Couldn't save temporary file "/mnt/xxx/transmission/dht.dat.tmp.XXXXXX": Permission denied (variant.c:1235)"

svlobanov, 2022-01-05:
I've tried to reproduced the issue, but I can't. I can see similar errors, but no crash and downloading works fine

image: https://downloads.openwrt.org/snapshots/targets/x86/64/openwrt-x86-64-generic-ext4-combined.img.gz (Tue Jan 4 23:20:53 2022)

run in qemu on x86_64 linux host: 
```
qemu-system-x86_64 -m 1024 -smp 1 -cpu host -M q35 \
-enable-kvm \
-nographic \
-drive file=openwrt-x86-64-generic-ext4-combined.img,id=d0,if=none,bus=0,unit=0 \
-device ide-hd,drive=d0,bus=ide.0 \
-device virtio-net,netdev=net0 \
-netdev user,id=net0,net=192.168.1.0/24,hostfwd=tcp:127.0.0.1:3322-192.168.1.1:22 \
-device virtio-net,netdev=net1 -netdev user,id=net1,net=192.0.2.0/24
```

check:

```
# opkg update
# opkg install transmission-daemon transmission-cli 
# ln -s /tmp /root/Downloads
# cd /tmp
# wget https://cdimage.debian.org/debian-cd/current/amd64/bt-cd/debian-11.2.0-amd64-netinst.iso.torrent
# transmission-cli debian-11.2.0-amd64-netinst.iso.torrent
transmission-cli 3.00 (bb6b5a062e)
[2022-01-05 16:34:34.948] Transmission 3.00 (bb6b5a062e) started
[2022-01-05 16:34:34.949] RPC Server: Adding address to whitelist: 127.0.0.1
[2022-01-05 16:34:34.950] RPC Server: Adding address to whitelist: ::1
[2022-01-05 16:34:34.951] DHT: Generating new id
[2022-01-05 16:34:34.998] Saved "/root/.config/transmission/torrents/28c55196f57753c40aceb6fb58617e6995a7eddb.torrent"
[2022-01-05 16:34:35.000] debian-11.2.0-amd64-netinst.iso: Pausing
[2022-01-05 16:34:35.001] Saved "/root/.config/transmission/resume/28c55196f57753c40aceb6fb58617e6995a7eddb.resume"
[2022-01-05 16:34:35.002] debian-11.2.0-amd64-netinst.iso: Queued for verification
[2022-01-05 16:34:35.003] debian-11.2.0-amd64-netinst.iso: Verifying torrent
[2022-01-05 16:34:35.281] Port Forwarding (NAT-PMP): initnatpmp succeeded (0)
[2022-01-05 16:34:35.282] Port Forwarding (NAT-PMP): sendpublicaddressrequest succeeded (2)
Progress: 0.0%, dl from 0 of 0 peers (0 kB/s), ul to 0 (0 kB/s) [None][2022-01-05 16:34:43.292] Port Forwarding: Port Forwarding
[2022-01-05 16:34:43.333] web: will verify tracker certs using envvar CURL_CA_BUNDLE: none
[2022-01-05 16:34:43.334] web: NB: this only works if you built against libcurl with openssl or gnutls, NOT nss
[2022-01-05 16:34:43.335] web: NB: invalid certs will show up as 'Could not connect to tracker' like many other errors
Progress: 0.0%, dl from 0 of 35 peers (0 kB/s), ul to 0 (0 kB/s) [None][2022-01-05 16:34:55.000] debian-11.2.0-amd64-netinst.iso: Starting IPv4
Progress: 0.0%, dl from 9 of 48 peers (1 kB/s), ul to 0 (0 kB/s) [0.00][2022-01-05 16:34:56.259] tr_crypto_utils: CyaSSL error: Key size error,
[2022-01-05 16:34:56.261] tr_crypto_utils: CyaSSL error: unknown error number
[2022-01-05 16:34:56.631] tr_crypto_utils: CyaSSL error: Key size error, either too small or large
[2022-01-05 16:34:56.632] tr_crypto_utils: CyaSSL error: unknown error number
Progress: 59.2%, dl from 48 of 50 peers (9.65 MB/s), ul to 0 (0 kB/s) [0.00][2022-01-05 16:35:23.866] Couldn't connect socket 22 to 2a01:e34:e:
[2022-01-05 16:35:23.867] Couldn't connect socket 22 to 2a03:4000:9:30::1, port 51413 (errno 13 - Permission denied)
[2022-01-05 16:35:23.869] Couldn't connect socket 22 to 2a01:e0a:352:2450:211:32ff:fed8:cacb, port 63810 (errno 13 - Permission denied)
[2022-01-05 16:35:23.871] Couldn't connect socket 22 to 2607:5300:60:acb::1, port 51413 (errno 13 - Permission denied)
Progress: 98.8%, dl from 40 of 49 peers (9.17 MB/s), ul to 0 (0 kB/s) [0.00][2022-01-05 16:35:42.382] debian-11.2.0-amd64-netinst.iso: State ch
Seeding, uploading to 0 of 12 peer(s), 0 kB/s [0.00]                              
Stopping torrent...
[2022-01-05 16:35:49.818] debian-11.2.0-amd64-netinst.iso: Pausing
[2022-01-05 16:35:49.820] Saved "/root/.config/transmission/resume/28c55196f57753c40aceb6fb58617e6995a7eddb.resume"
[2022-01-05 16:35:49.821] Saved "/root/.config/transmission/settings.json"

[2022-01-05 16:35:49.822] DHT: Saving 79 (79 + 0) nodes
[2022-01-05 16:35:49.823] Saved "/root/.config/transmission/dht.dat"
[2022-01-05 16:35:49.824] Port Forwarding: Stopped
[2022-01-05 16:35:49.825] debian-11.2.0-amd64-netinst.iso: Removing torrent
[2022-01-05 16:35:49.825] debian-11.2.0-amd64-netinst.iso: Pausing
[2022-01-05 16:35:49.826] Saved "/root/.config/transmission/stats.json"

# sha256sum debian-11.2.0-amd64-netinst.iso
45c9feabba213bdc6d72e7469de71ea5aeff73faea6bfb109ab5bad37c3b43bd  debian-11.2.0-amd64-netinst.iso

# ldd /usr/bin/transmission-daemon
	/lib/ld-musl-x86_64.so.1 (0x7fed745f0000)
	libminiupnpc.so.17 => /usr/lib/libminiupnpc.so.17 (0x7fed745e1000)
	libnatpmp.so.1 => /usr/lib/libnatpmp.so.1 (0x7fed745dc000)
	libevent-2.1.so.7 => /usr/lib/libevent-2.1.so.7 (0x7fed745a1000)
	libcurl.so.4 => /usr/lib/libcurl.so.4 (0x7fed7454f000)
	libwolfssl.so.4.8.1.5fb91bea => /usr/lib/libwolfssl.so.4.8.1.5fb91bea (0x7fed74325000)
	libz.so.1 => /usr/lib/libz.so.1 (0x7fed74310000)
	libc.so => /lib/ld-musl-x86_64.so.1 (0x7fed745f0000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fed742fc000)
	libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x7fed742d7000)

```

nitroshift, 2022-01-06:
@svlobanov 

I'm using transmission-web-control as gui frontend and the error posted above is just a snippet, system log is flooded with the above lines. Also, I can't see the temporary file error in your log. Should you need any more information, please let me know.

nitroshift

svlobanov, 2022-01-06:
> @svlobanov
> 
> I'm using transmission-web-control as gui frontend and the error posted above is just a snippet, system log is flooded with the above lines and download doesn't work. Permission error was a bit not set on the directory, it is solved, please disregard it. Should you need any more information, please let me know.
> 
> nitroshift

Can you reproduce the issue on official snapshot build (the link in my previous comment) and provide step-by-step guide how to reproduce?

nitroshift, 2022-01-06:
@svlobanov 

I'm using openwrt-x86-64-generic-ext4-combined-efi.img.gz from https://downloads.openwrt.org/snapshots/targets/x86/64/
Steps to reproduce the issue: open transmission gui in any browser, add torrent. System log gets flooded with errors and download doesn't work. Here are my installed dependencies:

dd /usr/bin/transmission-daemon 
	/lib/ld-musl-x86_64.so.1 (0x7f37e08ec000)
	libminiupnpc.so.17 => /usr/lib/libminiupnpc.so.17 (0x7f37e08dd000)
	libnatpmp.so.1 => /usr/lib/libnatpmp.so.1 (0x7f37e08d8000)
	libevent-2.1.so.7 => /usr/lib/libevent-2.1.so.7 (0x7f37e089d000)
	libcurl.so.4 => /usr/lib/libcurl.so.4 (0x7f37e084b000)
	libwolfssl.so.4.8.1.5fb91bea => /usr/lib/libwolfssl.so.4.8.1.5fb91bea (0x7f37e0621000)
	libz.so.1 => /usr/lib/libz.so.1 (0x7f37e060c000)
	libc.so => /lib/ld-musl-x86_64.so.1 (0x7f37e08ec000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f37e05f8000)
	libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x7f37e05d3000)

Here is the link to the errors: https://pastebin.com/r1EE3yWT

nitroshift

dangowrt, 2022-01-06:
Does the torrent you are adding have trackers? And if so, are they maybe using HTTPS? That would explain why it doesn't happen to me when I add torrents which are without any tracker or with HTTP or UDP-based trackers.

svlobanov, 2022-01-06:
one more question. Is DHT enabled in torrent-file? (some torrents have "private" flag to disable DHT)

nitroshift, 2022-01-06:
@dangowrt 

Any torrent triggers the issue. I tried with this: https://cdimage.debian.org/debian-cd/current/amd64/bt-dvd/debian-11.2.0-amd64-DVD-1.iso.torrent

@svlobanov 

DHT is enabled in /etc/config/transmission: option dht_enabled true

nitroshift

svlobanov, 2022-01-06:
Also, I think that "Wed Jan 5 12:45:42 2022 daemon.err transmission-daemon[20935]: [2022-01-05 10:45:42.570] Couldn't save temporary file "/mnt/xxx/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)" is not normal, looks like you have permissions issues. Can you fix this?

nitroshift, 2022-01-06:
@svlobanov 

I did fix that as mentioned in previous comments.

svlobanov, 2022-01-06:
@nitroshift please provide `cat /proc/cpuinfo` and `cat /proc/meminfo`

nitroshift, 2022-01-06:
# cat /proc/cpuinfo

   "processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 60
model name	: Intel(R) Core(TM) i3-4160 CPU @ 3.60GHz
stepping	: 3
microcode	: 0x1c
cpu MHz		: 826.758
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm arat pln pts
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit srbds
bogomips	: 7183.98
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 60
model name	: Intel(R) Core(TM) i3-4160 CPU @ 3.60GHz
stepping	: 3
microcode	: 0x1c
cpu MHz		: 1960.944
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 2
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm arat pln pts
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit srbds
bogomips	: 7183.98
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 60
model name	: Intel(R) Core(TM) i3-4160 CPU @ 3.60GHz
stepping	: 3
microcode	: 0x1c
cpu MHz		: 1004.017
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 0
[... 60 lines cut ...]
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:         20324 kB
Mapped:            23056 kB
Shmem:             21024 kB
KReclaimable:      26880 kB
Slab:              79412 kB
SReclaimable:      26880 kB
SUnreclaim:        52532 kB
KernelStack:        2080 kB
PageTables:         1072 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     4020400 kB
Committed_AS:      77820 kB
VmallocTotal:   34359738367 kB
VmallocUsed:        8524 kB
VmallocChunk:          0 kB
Percpu:             1072 kB
DirectMap4k:       59284 kB
DirectMap2M:     2967552 kB
DirectMap1G:     6291456 kB"

diizzyy, 2022-01-06:
Looking at the logs it might be related toe WolfSSL. My general assumption is that it doesn't see much usage/testing outside of OpenWrt as OpenSSL is much more common. It also complains about libcurl...

```
[2022-01-05 16:34:43.334] web: NB: this only works if you built against libcurl with openssl or gnutls, NOT nss
```

I'd try using OpenSSL and see if the issue goes away.

nitroshift, 2022-01-07:
@svlobanov @dangowrt 

Last night i tried Openwrt 21.02.1 release (https://downloads.openwrt.org/releases/21.02.1/targets/x86/64/openwrt-21.02.1-x86-64-generic-ext4-combined-efi.img.gz) and Openwrt Snapshot  (https://downloads.openwrt.org/snapshots/targets/x86/64/openwrt-x86-64-generic-ext4-combined-efi.img.gz). Issue is *NOT* present in the 21.02.1 release but it *IS* present in Snapshot. I've also built an image from Openwrt Master (OpenWrt SNAPSHOT, r18503-7881dce7d8) which is what I'm running at the moment and the issue is still present. As @diizzyy said, I suspect it's a crypto library issue (hint libwolfssl).

nitroshift

nitroshift, 2022-01-13:
Just an update: reverting from wolfssl to openssl / polarssl solves the issue. No more spam in system logs and downloads are working again.

nitroshift

closed 2022-01-13 by nitroshift

luckydevil13, 2022-01-14:
same here, confirm on snapshot

```
Fri Jan 14 21:09:59 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:09:59.593] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Fri Jan 14 21:09:59 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:09:59.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:10:01 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:10:01.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:10:01 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:10:01.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:10:11 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:10:11.593] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Fri Jan 14 21:10:12 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:10:12.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:11:56 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:56.593] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Fri Jan 14 21:11:56 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:56.593] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Fri Jan 14 21:11:56 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 28:6c:07:83:64:5b
Fri Jan 14 21:11:57 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:57.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:11:57 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:57.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:11:57 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:57.594] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Fri Jan 14 21:11:57 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:57.594] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Fri Jan 14 21:11:57 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:57.594] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Fri Jan 14 21:11:57 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:57.594] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)
Fri Jan 14 21:11:58 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:58.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:11:58 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:58.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:11:58 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:58.593] tr_crypto_utils CyaSSL error: unknown error number (crypto-utils-cyassl.c:292)
Fri Jan 14 21:11:58 2022 daemon.err transmission-daemon[8444]: [2022-01-14 18:11:58.594] tr_crypto_utils CyaSSL error: Key size error, either too small or large (crypto-utils-cyassl.c:260)

```

svlobanov, 2022-01-14:
@luckydevil13 could you please report to upstream? https://github.com/transmission/transmission/issues 
wolfssl is supported by upstream so we should report them

nitroshift, 2022-01-25:
@NautilusChan 

Revert https://github.com/openwrt/packages/commit/6786e35ff87c4f72ec5343a9f3815842367b3c84#diff-3dc735e7e3d28d09b2f110d6d92050af27a448236d24f10ab3984074498f1fd9

root0x, 2023-01-02:
Hi, was this issue actually fixed because i am experiencing this issue.