openwrt/packages #17506: Issue building master after git pull update

erdoukki, 2022-01-06:
Description:

From some times, I get issue after a git pull (update).

I build packages for testing on MVEBU, ARM64, ESPRESSOBIN.

When I try to directly build a package, modified in its own branch, like for my upstream PR proposal in review.
I always get this error:

```
$ make packages/fail2ban/compile
...
time: package/kernel/gpio-button-hotplug/compile#0.11#0.03#0.14
    ERROR: package/kernel/gpio-button-hotplug failed to build.
make[1]: *** [package/Makefile:116: package/kernel/gpio-button-hotplug/compile] Error 1
```

I have find a personal tweak/fix with:
```
$ make toolchain/compile
...
$ make target/compile
...
```
Then my package can build successfully (or badly, if it gets errors, but no more with the `gpio-button-hotplug` error)
```
$ make packages/fail2ban/compile
```

Hot to reproduce:

I made update of my openwrt local fork with:

```
git pull
./scripts/feeds update -a
./scripts/feeds install -a
make defconfig
```

then, in my packages local fork:

```
git checkout master
git pull
git checkout <local-working-branch>
git pull origin master --rebase
<LOCAL MODIFICATION/BUGFIXES/EDIT/...>
```

then, back in my openwrt local fork:
```
$ make packages/fail2ban/compile
```
and here came the problem, BUT, only if updating with pull was issued or get results with new commits !

What am I doing wrong, and how can I fix it or help to fix upstream ?

erdoukki, 2022-01-15:
A quick fix with a single command;
`make {toolchain,target,package/fail2ban}/compile`

feckert, 2022-01-22:
I think you have solved your problem

closed 2022-01-22 by feckert

erdoukki, 2022-01-22:
I find a solution, but it is still an issue...

hnyman, 2022-01-26:
I have recently started to run this sometimes, when compiling individual packages. 
Strange.