openwrt/packages #17508: adguardhome: Go 1.17 support deprecation for v0.108.0 and removal in v0.109.0

jamesmacwhite, 2022-01-06:
Maintainer: @alaviss @dobo90
Environment: N/A

Description:

I'm not familiar with Go dependencies with OpenWrt buildroot, but noticed that the AdGuard team have mentioned that [Go 1.17 support is being deprecated and will be dropped in v0.109.0](https://github.com/AdguardTeam/AdGuardHome/releases/tag/v0.108.0-b.1)

Would this cause an issue for building the package with OpenWrt 21.02 or master? I don't know what Go version buildroot would use currently.

alaviss, 2022-01-06:
> Would this cause an issue for building the package with OpenWrt 21.02 or master?

Maybe. The only way to know is to test it out unfortunately.

> I don't know what Go version buildroot would use currently.

I believe its 1.17.1.

jamesmacwhite, 2022-01-06:
@alaviss Thanks. So potentially could be an issue in the next couple of stable releases maybe.

I'm not that experienced with buildroot operations, so maybe others can provide their thoughts. Interestingly I noticed Go 1.18 is an unstable release and 1.17.5 is latest stable, interesting that support is being dropped quite soon when 1.18 isn't even stable itself.

I guess we'll have to see, wanted to flag it anyway, for information.

predators46, 2022-01-13:
just wait for golang in the update version 1.18 and the current stable version is v0.107.2

alaviss, 2022-01-13:
I think the main concern should be when Go 1.18 is released and imported to OpenWrt. If https://github.com/openwrt/packages/issues/16698#issuecomment-997812738 is anything to go by, AdGuard Home does not like to be built with a Go version differing from what it was developed on.

predators46, 2022-03-25:
https://github.com/openwrt/packages/pull/18147

jamesmacwhite, 2022-04-02:
So, looks like updating Go to 1.18 has presented the scenario of issues building 0.107.5 under 22.03/master. I don't think there is a lot we can do, given AdGuardHome just doesn't play nice with different Go versions. It is my understanding that 0.107.5 will only build with 1.17, the beta 0.108 releases may build with 1.18.

closed 2025-10-30 by GeorgeSapkin