openwrt/packages #17564: Netdata Dashboard v1.30.1 non-functional on Mt7620a (MI Router R3)

jahidulislamjahid, 2022-01-10:
**Netdata v1.30.1** does not load anything when i dial 19999 followed by my routers IP.
currently i'm running openwrt 21.10 on **MI Router R3** which is based on **MT7620a** _Raimaps mipsel_24kc_ 
The feedback i get says **"This Site Can'nt Be Reached"** 
can we get any solution on that. i only need the netdata dashboard. i dont need netdata cloud right now. so it'll be appreciated if i get any solution just only on the dashboard.

BKPepe, 2022-01-11:
Hello,

Is there anything in the system log related to netdata, which could be helpful? Would it be possible to paste the output from the following file: ``/etc/openwrt_release``?

I tried netdata on several platforms like ``powerpc_8540``, ``mvebu/cortex-a9``, ``mvebu/cortex-a53`` (all Turris devices) and it works there. Based on your report, I tried it also on ``bcm2711`` (Raspberry Pi 4 CM) and it works there.

I do have Xiaomi Mi WiFi R3G, ramips, mt7621, but currently, it is not connected and I am not sure in which state the device it is as I lent it to my colleague at work. I will need to take a look, but not sure when.

diizzyy, 2022-01-11:
I'd seen musl related crashes on my ar71xx platforms, at that point I stopped bothering look it at. MIPS platforms are also in general very low of memory (128Mbyte isn't all that common) and it'll simply run out of memory anyway.

jahidulislamjahid, 2022-01-11:
@BKPepe Here is what i see in the `/etc/openwrt_release`

DISTRIB_ID='X-WRT'
DISTRIB_RELEASE='21.10_b202201032246'
DISTRIB_REVISION='r18621+1-6bbb4b1068'
DISTRIB_CODENAME='impish'
DISTRIB_TARGET='ramips/mt7620'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='X-WRT 21.10_b202201032246 Impish'
DISTRIB_TAINTS='busybox'

and as i don't know much about system logging or which information you need from system log therefore i am attaching my entire system log for you. 
and by the way openwrt does not officially support my device "MI Router R3" and I'm using a openwrt derivative which is a fork of the openwrt.

Thanks for your efforts.
[system.log](https://github.com/openwrt/packages/files/7845171/system.log)

THLIVSQAZ, 2022-01-12:
same problem

```
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: resources control: allowed file descriptors: soft = 1024, max = 4096
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: Adjusted my Out-Of-Memory (OOM) score from 0 to 1000.
Wed Jan 12 20:51:53 2022 daemon.err netdata[32230]: Cannot adjust netdata scheduling policy to idle (5), with priority 0. Falling back to nice.
Wed Jan 12 20:51:53 2022 daemon.err netdata[32230]: Cannot get my current process scheduling policy.
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: netdata started on pid 32230.
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: Initializing spawn client.
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: Executing /usr/lib/netdata/plugins.d/system-info.sh
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_CONTAINER_OS_NAME=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_CONTAINER_OS_ID=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_CONTAINER_OS_ID_LIKE=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_CONTAINER_OS_VERSION=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_CONTAINER_OS_VERSION_ID=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_CONTAINER_OS_DETECTION=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_HOST_OS_NAME=OpenWrt
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_HOST_OS_ID=openwrt
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_HOST_OS_ID_LIKE=lede openwrt
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_HOST_OS_VERSION=SNAPSHOT
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_HOST_OS_VERSION_ID=snapshot
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_HOST_OS_DETECTION=/etc/os-release
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_HOST_IS_K8S_NODE=false
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_KERNEL_NAME=Linux
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_KERNEL_VERSION=5.10.88
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_ARCHITECTURE=mips
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_VIRTUALIZATION=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_VIRT_DETECTION=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_CONTAINER=unknown
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_CONTAINER_DETECTION=none
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_CPU_LOGICAL_CPU_COUNT=4
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_CPU_VENDOR=
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_CPU_MODEL=
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_CPU_FREQ=unknown
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_CPU_DETECTION=sysfs procfs
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_TOTAL_RAM=256274432
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_RAM_DETECTION=procfs
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_TOTAL_DISK_SIZE=0
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: NETDATA_SYSTEM_DISK_DETECTION=sysfs
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: SQLite database /var/cache/netdata/netdata-meta.db initialization
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: SQLite database initialization completed
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: CONFIG: cannot load user config '/etc/netdata/stream.conf'. Will try stock config.
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: Host 'OpenWrt' (at registry as 'OpenWrt') with guid '9a9517e6-6d67-11ec-9f22-d4ee07610393' initialized, os 'linux', timezone 'UTC', tags '', program_name 'netdata', program_version 'v1.30.1', update every 2, memory mode save, history entries 4032, streaming disabled (to '' with api key ''), health disabled, cache_dir '/var/cache/netdata', varlib_dir '/var/lib/netdata', health_log '/var/lib/netdata/health/health-log.db', alarms default handler '/usr/lib/netdata/plugins.d/alarm-notify.sh', alarms default recipient 'root'
Wed Jan 12 20:51:53 2022 daemon.err netdata[32230]: child pid 32331 exited with code 127.
Wed Jan 12 20:51:53 2022 daemon.err netdata[32230]: /usr/lib/netdata/plugins.d/get-kubernetes-labels.sh exited abnormally. No kubernetes labels will be added to the host.
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: thread created with task id 32332
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: thread created with task id 32333
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: thread created with task id 32334
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: thread created with task id 32335
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: thread created with task id 32337
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: thread created with task id 32338
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: thread created with task id 32339
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: thread created with task id 32340
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: Initializing command server.
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: set name of thread 32333 to PLUGIN[diskspac
Wed Jan 12 20:51:53 2022 daemon.info netdata[32230]: set name of thread 32337 to BACKENDS
[... cut ...]
Wed Jan 12 20:51:54 2022 daemon.info netdata[32230]: Initializing file /var/cache/netdata/cpu.cpu2/system.db.
Wed Jan 12 20:51:54 2022 kern.info kernel: [552615.835127] do_page_fault(): sending SIGSEGV to PLUGIN[proc] for invalid read access from 000000c8
Wed Jan 12 20:51:54 2022 kern.info kernel: [552615.844200] epc = 77dd9498 in libc.so[77d5d000+a6000]
Wed Jan 12 20:51:54 2022 kern.info kernel: [552615.849348] ra  = 77ddb40c in libc.so[77d5d000+a6000]
Wed Jan 12 20:51:54 2022 daemon.info procd: Instance netdata::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
```

BKPepe, 2022-01-13:
> and by the way openwrt does not officially support my device "MI Router R3" and I'm using a openwrt derivative which is a fork of the openwrt.

Since, I don't have X-Wrt, I can not help you further than suggesting to disable proc plugin to see if it is better or try to update netdata version to the latest one yourself to see if it works. If not, then you should open an issue in netdata repository.

@THLIVSQAZ , same problem, but I see that you are using daily OpenWrt snapshots based on the input. Are you using the same router? :thinking:

THLIVSQAZ, 2022-01-13:
I rollback to the openwrt 21.02 branch, work fine now. Thanks

BKPepe, 2022-01-13:
In OpenWrt, there is used ``1.30.1`` version, which is available in all supported OpenWrt versions.

THLIVSQAZ, 2022-01-13:
yeah, no the netdata problem, may be musl or other cause crash

diizzyy, 2022-03-12:
Closing this is most likely not related to netdata itself and netdata is also now bumped

closed 2022-03-12 by diizzyy