openwrt/packages #17570: family-dns: need per zone controls

pprindeville, 2022-01-11:
Maintainer: @farmergreg
Environment: x86_64, generic, HEAD (83cb2f4dc8)

Description:

I have a firewall with 3 different VLAN's behind it, via several Ubiquiti AP's, each VLAN being associated with a different SSID.

I'd like to only intercept/DNAT DNS requests on `lan2` (aka `eth0.2`).  It would be nice to be able to pick and choose which interfaces/zones that `family-dns` affects, as the kids at home have their own SSID to use with different rules, from the rest of the household.

It would also be nice to see it work not just with `dnsmasq`, but with `isc-dhcp` as well.

farmergreg, 2022-01-11:
Thanks for the request, @pprindeville it would be a good feature to add, indeed. Are you using dnsmasq or isc-dhcp?

Thanks,
Greg

pprindeville, 2022-01-11:
> Thanks for the request, @pprindeville it would be a good feature to add, indeed. Are you using dnsmasq or isc-dhcp?

I'm the isc-dhcp maintainer, so eating my own dog food.

Happy to work with you on it.