openwrt/packages #17588: ntpd | Package ntp-utils is missing dependencies for the following libraries: libintl.so.8

jeetu28, 2022-01-12:
While compiling the ntpd package, I am getting the below error and i have tried specifying the "DEPENDS = +libiconv" also
But still same

```
Package ntp-utils is missing dependencies for the following libraries:
libintl.so.8
Makefile:161: recipe for target '......../bin/packages/powerpc64_e5500/packages/ntp-utils_4.2.8p15-3_powerpc64_e5500.ipk' failed

```

neheb, 2022-01-19:
You want $(INTL_DEPENDS)

BKPepe, 2022-03-13:
This is interesting, but I am not able to reproduce your issue. Can you please provide more details? It will help to know which OpenWrt version you are compiling, for which target/device you are trying to compile it, and if you modify options to compile OpenWrt.

jow-, 2022-03-14:
> This is interesting, but I am not able to reproduce your issue.

I think you need to build OpenWrt with full language support to trigger this error.

BKPepe, 2022-03-14:
Yes, you are right. For such issues, NLS needs to be enabled, but this is
what we have for Turris routers and we did not encounter this issue.