openwrt/packages #17607: [openwrt-21.02] tor: Build failing for arc_arc700

jefferyto, 2022-01-14:
Maintainer: @hauke, @tripolar (ping @rsalvaterra, @ja-pa, @neheb)
Environment: arc_arc700

Description:

https://downloads.openwrt.org/releases/faillogs-21.02/arc_arc700/packages/tor/basic/compile.txt
https://downloads.openwrt.org/releases/faillogs-21.02/arc_arc700/packages/tor/full/compile.txt

```
  CCLD     src/test/bench
{standard input}: Assembler messages:
{standard input}:9369: Error: operand out of range (-130 is not between -128 and 127)
lto-wrapper: fatal error: /builder/shared-workdir/build/sdk/staging_dir/toolchain-arc_arc700_gcc-8.4.0_glibc/bin/arc-openwrt-linux-gnu-gcc returned 1 exit status
compilation terminated.
/builder/shared-workdir/build/sdk/staging_dir/toolchain-arc_arc700_gcc-8.4.0_glibc/bin/../lib/gcc/arc-openwrt-linux-gnu/8.4.0/../../../../arc-openwrt-linux-gnu/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:9599: src/test/bench] Error 1
make[5]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-arc_arc700_glibc/tor-basic/tor-0.4.5.10'
make[4]: *** [Makefile:6321: all] Error 2
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-arc_arc700_glibc/tor-basic/tor-0.4.5.10'
make[3]: *** [Makefile:183: /builder/shared-workdir/build/sdk/build_dir/target-arc_arc700_glibc/tor-basic/tor-0.4.5.10/.built] Error 2
time: package/feeds/packages/tor/basic/compile#128.20#19.90#210.51
```

Looks similar to, but not exactly like, the issue in https://github.com/openwrt/packages/pull/13467#issuecomment-697290330.

neheb, 2022-01-14:
it's a GCC bug. Can't do much about it.

jefferyto, 2022-01-14:
Perhaps there is something in tor we can patch or disable to work around the bug, like https://github.com/openwrt/packages/pull/13467#issuecomment-700968333?

neheb, 2022-01-14:
right. mallinfo however is not related. it's something else. I wonder if that bench binary is even needed.

GeorgeSapkin, 2025-11-08:
Target removed. 21.02 is out of support. Closing.

closed 2025-11-08 by GeorgeSapkin