openwrt/packages #17642: prometheus-node-exporter-lua: configure ipv6 listen address to use ULA

NoctivagusObitus, 2022-01-18:
Hi

currently it is possible to configure prometheus exporter to listen on a single interface or more precisely on one ip address of that interface. in my case this always is some ipv6 GUA address where i would prefer to use the ULA address of the same interface. my goal is to scrape the routers via a controlled path through vpn and vlan from some server in the internet to the router and back.

is there any way to accomplish having prometheus node exporter to listening on ULA addresses specifically?

the setup creating my desire is as follows:
i am scraping multiple openwrt routers via single wireguard vpn gateway. there are multiple such gateways all managing there own /56 ipv6 ULA subnet where the ther routers get there ip adresses via dhcp from. prometheus is running on a server having wireguard tunnels configured to each gateway router and subnet. all the ipv6 sites are unified by the same /48 subnet. 
since all the routers register themself with the gateway routers those gateways know there host names and ip addresses. furthermore since ULA addresses are preferred over GUA addresses the vpn setup with semi static ULA subnets works perfect for me. using an upstream dns setup at the prometheus server i am able to easily scrape all routers from a centralized config on one machine.
on each site the routers are configured to join a mesh cloud to request there addresses from said gateway routers. within that mesh cloud exist multiple vlans where on of those is the internal one i like to use for scraping.

kind regards

champtar, 2022-04-25:
In https://github.com/openwrt/packages/pull/18360 I switch to uhttpd , which makes listening to multiple IP simpler.
We are still using `network_get_ipaddr6` so you will likely see no changes.
We could easily reintroduce a listen_addresses in a follow up PR.
Would putting static listen addresses in the conf fix your use case ?

NoctivagusObitus, 2022-04-27:
Thx a lot for that suggestion @champtar
I naively copied over the code of the files the reference commit is showing. I would guess this should work since it is lua but maybe I am wrong? If I simply start `service prometheus-node-exporter-lua start` afterwards I can see a uhttpd process listening on port 9100 but ipv4 only?
```sh
:/etc/config# netstat -tunlp | grep uhttpd$
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      2254/uhttpd
tcp        0      0 127.0.0.1:9100          0.0.0.0:*               LISTEN      4919/uhttpd
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      2254/uhttpd
tcp        0      0 :::443                  :::*                    LISTEN      2254/uhttpd
tcp        0      0 :::80                   :::*                    LISTEN      2254/uhttpd

:/etc/config# netstat -tunlp | grep 9100
tcp        0      0 127.0.0.1:9100          0.0.0.0:*               LISTEN      5356/uhttpd
```
when I then change my interface to the desired one I furthermore get some error starting the process (maybe some conflicts of not up to date packages in my case)
```sh
:/etc/config# cat prometheus-node-exporter-lua 
config prometheus-node-exporter-lua 'main'
	option listen_interface 'br-lan'
	option listen_port '9100'

:/etc/config# logread -e prome
Wed Apr 27 09:18:19 2022 daemon.info prometheus-node-exporter-lua: defering start until listen interface br-lan becomes ready
Wed Apr 27 09:18:20 2022 daemon.info procd: Not starting instance prometheus-node-exporter-lua::data, command not set
Wed Apr 27 09:43:01 2022 daemon.info prometheus-node-exporter-lua: defering start until listen interface br-lan becomes ready
Wed Apr 27 09:43:01 2022 daemon.info procd: Not starting instance prometheus-node-exporter-lua::data, command not set
```

so sadly simple copy and paste is not working for me at the moment but maybe some OpenWrt upgrade later in time fixes that?
The versions I am running:
```sh
:/etc/config# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.1'
DISTRIB_REVISION='r16325-88151b8303'
DISTRIB_TARGET='ramips/mt7620'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.1 r16325-88151b8303'
DISTRIB_TAINTS=''

:/etc/config# opkg list-installed | grep 'prometheus\|uhttpd'
falter-berlin-uhttpd-defaults - 1
prometheus-node-exporter-lua - 2022.04.18-1
prometheus-node-exporter-lua-wifi - 2022.04.18-1
prometheus-node-exporter-lua-wifi_stations - 2022.04.18-1
uhttpd - 2021-03-21-15346de8-2
uhttpd-mod-lua - 2021-03-21-15346de8-2
uhttpd-mod-ubus - 2021-03-21-15346de8-2

# Image in use is a Custom FF image
:/etc/config# cat /etc/banner
  _____        _  __             _
 |  ___|      (_)/ _|           | |
 | |_ _ __ ___ _| |_ _   _ _ __ | | __
 |  _| '__/ _ \ |  _| | | | '_ \| |/ /
 | | | | |  __/ | | | |_| | | | |   <
 \_| |_|  \___|_|_|  \__,_|_| |_|_|\_\ 

 Firmware Berlin (v1.2.1 - v1.2.1)
   ramips - mt7620
 https://wiki.freifunk.net/Berlin:Firmware
 https://github.com/freifunk-berlin/falter-packages
 -----------------------------------------------------

 If you find bugs please report them at:

   https://github.com/freifunk-berlin/falter-packages/issues/

 For questions write a mail to <berlin@berlin.freifunk.net>
 or check https://berlin.freifunk.net/contact for our weekly meetings.

```

champtar, 2022-04-30:
In the init file, you can try to remove
```
network_is_up "$interface" || {
	_log "defering start until listen interface $interface becomes ready"
	return 0
}
```
But still we are using `network_get_ipaddr6` so we will not listen on ULA I think.

champtar, 2022-05-01:
listen_interface should be 'lan', not 'br-lan', ie we want the OpenWrt config interface, not the linux interface

NoctivagusObitus, 2022-05-01:
all right
setting the correct interface name already helped. thanks for that. but sadly its still only listening on the ipv4 address
```sh
config prometheus-node-exporter-lua 'main'
	option listen_interface 'lan'
	option listen_port '9100'

root@Rudolf-Seiffert-25_2:/etc/config# netstat -tunlp | grep :9100
tcp        0      0 10.1.1.82:9100          0.0.0.0:*               LISTEN      8342/uhttpd
```

the setup in my case is a Prometheus running somewhere in the internet and I have a wireguard connection running with my personal ULA prefix. but the router I am referring to is just a secondary one (client) to the one with that interface. via firewalling I like to have the metrics only be scrapable from the 'lan' network but no from the 'fguest' network

imp1sh, 2023-04-27:
I'm suffering the same problem.
```
config prometheus-node-exporter-lua 'main'
        option listen_interface 'MGMT'
        option listen_ipv6 '1'
        option listen_port '9100'
```
The GUA is always chosen to listen on but since mine are dynamic I want it to listen to the ULA address.
Weird thing is, when I set `option listen_ipv6 '0'` it will still listen on IPv6. (My interface does not have IPv4).

NoctivagusObitus, 2024-02-21:
@imp1sh 
Although a bit late and not definitive answer... I would like to point you to this peace of code:
https://github.com/openwrt/packages/blob/e119a57e97e894f0636df999d82807ae5c494030/utils/prometheus-node-exporter-ucode/files/init#L34-L39
I have not looked into it in details but the surrounding if statement seams to only distinguish if interface is `*` or not and in the not case check if there is an ipv6 address depending on this function:
https://github.com/openwrt/chaos_calmer/blob/22375a2fb87271dadcdb30f5e825c168ba73d502/package/base-files/files/lib/functions/network.sh#L31-L44

danthonywalker, 2025-08-15:
I have just ran into this issue. My interface has IPv4 and IPv6. If I have `listen_ipv6 '0'`, then everything is fine and I can get metrics via IPv4. However, if I use `listen_ipv6 '1`, then I cannot get metrics with either IPv4 or IPv6. I presume it's because `listen_ipv6 '1'` is attached to the GUA, but I don't know why that cuts off IPv4...

It would be preferable if I specify the interface to listen on, it would listen to all the IPs on that interface.

NoctivagusObitus, 2025-09-07:
> It would be preferable if I specify the interface to listen on, it would listen to all the IPs on that interface.

I partially agree with you. It would make sense given the name of the config `listen_interface`. But that would also mean to may have your prometheus endpoint scrapable via the internet in case it gets a GUA ipv6. Sure this depends on your firewall config and may be mitigated. I just wanted to point it out. 
My understanding is that the code here is responsible for getting the used ip adresses.
https://github.com/openwrt/packages/blob/23475cb50ee281f1a753fd6b7c636475daf73b0c/utils/prometheus-node-exporter-lua/files/etc/init.d/prometheus-node-exporter-lua#L23-L30
which sources the file and uses functions from my previous comment.

Why you can not use your ipv4 target is not clear to me at the moment. You may check the logs and `netstat -ntulp` looking for your chosen port to see on what ip if any you are listening on. You may grep for `uhttpd` since the prometheus lua exporter utilizes uhttpd to as the web server.