openwrt/packages #17652: ddns-scripts: Allow specifying port to use with bind-nsupdate

TTornblom, 2022-01-20:
# Issue template (remove lines from top till here)

Maintainer: @hgl
Environment: arm, wrt3200acm, 19.07.6 (ROOTer)

Description:
I have an internal DNS server on which I would like to be able to update an A address from a remote OpenWRT router.
This DNS server can be updated through a port different from 53, however the bind-nsupdate service has no provision for specifying which port to use.
I would like to have update_nsupdate.sh and dynamic_dns_functions.sh modified to allow using a port different from 53.

hgl, 2022-01-20:
Sorry, but I’m not the maintainer.

daiaji, 2022-02-02:
It's strange that I didn't get any response to the BUG of ddns-scripts.

feckert, 2023-09-18:
I don't have this requirement, so I haven't implemented it yet due to lack of time.
It is not a bug, it is a feature, that has not implemented yet. 
Maybe someone else can create a pull request. I would then be happy to review it. If the feature gets added, then it should get its own option in the uci with the name `port`.