openwrt/packages #17674: Can't verify the torrents

NautilusChan, 2022-01-22:
Description:
After I change my router deviece frome Netgear R8000 to NanoPi R4s,  the transmission can't work well. Although I keep all the config files, resume files , and  torrents files, the USB disks are mouted as the same path(such as /mnt/sda1/PT), I still get "no data found" error. I have set "chown -R transmission:transmission /mnt ". I have tried to re-add the torrents, but the torrents can't be verified. I found the log as below:

```
Sat Jan 22 17:40:39 2022 daemon.err transmission-daemon[16398]: [2022-01-22 09:40:39.787] The.Nun.2018.BluRay.1080p.AVC.Atmos.TrueHD7.1-MTeam Queued for verification (verify.c:272)
Sat Jan 22 17:40:39 2022 daemon.err transmission-daemon[16398]: [2022-01-22 09:40:39.787] The.Nun.2018.BluRay.1080p.AVC.Atmos.TrueHD7.1-MTeam Verifying torrent (verify.c:220)
Sat Jan 22 17:40:39 2022 daemon.err transmission-daemon[16398]: [2022-01-22 09:40:39.788] The.Nun.2018.BluRay.1080p.AVC.Atmos.TrueHD7.1-MTeam verifying torrent... (verify.c:52)
Sat Jan 22 17:40:39 2022 daemon.err transmission-daemon[16398]: [2022-01-22 09:40:39.788] The.Nun.2018.BluRay.1080p.AVC.Atmos.TrueHD7.1-MTeam Verification is done. It took 0 seconds to verify 31968460800 bytes (31968460800 bytes per second) (verify.c:161)

gfrancesco, 2022-02-04:
Hi, what version of transmission are you using? I have a maybe related issue after upgrading to 3.00 (I'm on OpenWRT snapshot) from 2.94.

In short the problem I'm facing is that all my shared torrents where the content is a symlink to another location outside the download folder are not working. I get a **no data found** error.
I fixed it by disabling the **procd jail** in the init script, see code below, but of course that's more of a workaround rather than a fix.

The configuration, permissions and file system structure are the same I was using with 2.94. I'll give a few examples to clarify.

`"download-dir": "/mnt/stripe/transmission/downloads"`
Let's say I download `ubuntu.iso` in `/mnt/stripe/transmission/downloads/iso_linux/ubuntu.iso`, a subfolder of my download dir.

If I move the iso up one dir to `/mnt/stripe/transmission/downloads/ubuntu.iso` and then create a symlink:
**/mnt/stripe/transmission/downloads/iso_linux/ubuntu.iso -> /mnt/stripe/transmission/downloads/ubuntu.iso** it works.
By working I mean that I just refresh the torrent in transmission, without changing it's location, and it keeps on seeding.

However if I move the file to `/tmp` for example it does not work:
**/mnt/stripe/transmission/downloads/iso_linux/ubuntu.iso -> /tmp/ubuntu.iso**
When I refresh the torrent in transmission now I get the **no data found** errror.

I have many torrents that are symlinked to paths outside of `"download-dir": "/mnt/stripe/transmission/downloads"`, for example to `/mnt/mirror/linux_sw`.

I can post this as separate issue if needed.

Commented blocks on `/etc/init.d/transmission` that fix the problem:

```bash
	procd_open_instance
	procd_set_param command "$PROG"
	procd_append_param command -f -g "$config_dir"
	procd_set_param user "$user"
	procd_set_param group "$group"
	procd_set_param nice "$nice"
	procd_set_param stderr 1
	procd_set_param respawn

#	seccomp_path="/etc/seccomp/transmission-daemon.json"
#	if [ -f "$seccomp_path" ]; then
#		procd_set_param seccomp "$seccomp_path"
#	fi

	if [ -z "$USE" ]; then
		procd_set_param limits core="0 0"
	else
		procd_set_param limits core="0 0" as="$USE $USE"
		logger -t transmission "Starting with $USE virt mem"
	fi

	[ -d "$web_home" ] && procd_set_param env TRANSMISSION_WEB_HOME="$web_home"
	[ "$ca_bundle" -gt 0 ] && procd_set_param env CURL_CA_BUNDLE="$ca_bundle_file"

#	procd_add_jail transmission log
#	procd_add_jail_mount "$config_file"
#	procd_add_jail_mount_rw "$config_dir/resume"
#	procd_add_jail_mount_rw "$config_dir/torrents"
#	procd_add_jail_mount_rw "$config_dir/blocklists"
#	procd_add_jail_mount_rw "$config_dir/stats.json"
#	procd_add_jail_mount_rw "$download_dir"
	web_home="${web_home:-/usr/share/transmission/web}"
#	[ -d "$web_home" ] && procd_add_jail_mount "$web_home"
#	[ -f "$ca_bundle_file" ] && procd_add_jail_mount "$ca_bundle_file"
	procd_close_instance
```

NautilusChan, 2022-02-14:
@gfrancesco  The version of transmission which I am using is 3.00(bb6b5a062e). I can download new data, but I can't verify the data whic I had download before.

NautilusChan, 2022-02-14:
@gfrancesco  Unbelievable!  The transmission can verify the data when I run the transmission by ./etc/init.d/transmission trace .But I use "trace" instead of "start"  just beacuase I want to get the detail log of the verify process.

dangowrt, 2022-02-20:
Using `trace` skips all jail restrictions and writes a list of syscalls used for future use a seccomp filter.

However, this issue is not related to seccomp but rather a consequence of mount namespace ie. filesystem restrictions.
I guess the torrent data is not located in a folder which is made available to transmission by ujail, only `download_dir` is.
You can add 
```
procd_add_jail_mount /mnt/additional/folder
```
to the init script to make the folder available to transmission read-only.
If that solves the problem we may add a configuration option for additional seed folders which are then also made available to the jailed process.

gfrancesco, 2022-02-21:
That was also my guess, I think that your solution might be useful. It will partially fix my problem, but I can probably adapt my download folder structure to make it work with additional read-only paths.
I still cannot explain how my configuration was working fine with transmission 2.94 which, as far as I can understand, was also using procd jails.

dangowrt, 2022-02-21:
Probably `procd-ujail` was not installed and now became a part of the default OpenWrt build. In case of `procd-uajil` missing, processes are just run without jail.
Once I find time for that I'll add an option (or rather: a list) for additional read-only paths to `/etc/config/transmission`, so people (including myself) won't have to edit the init script.

gfrancesco, 2022-02-21:
That explain it, thanks for your help!

clayface, 2022-02-24:
@dangowrt one issue in particular is blocklist files. These are first downloaded to `config_dir` before being moved. This is the error I receive when trying with procd-ujail enabled:
`Feb 24 20:48:23 OpenWrt-RB5009 transmission-daemon[16060]: [2022-02-24 20:48:23.728] Couldn't save file "/tmp/transmission/blocklist.tmp.XXXXXX": Bad file descriptor (rpcimpl.c:1696)`

No blocklist is downloaded subsequently.

a different example is
`Feb 24 20:55:59 OpenWrt-RB5009 transmission-daemon[16385]: [2022-02-24 20:55:59.892] Couldn't save temporary file "/tmp/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)`

I believe making config_dir rw mounted may the only option here, or else patch transmission to make blocklist.tmp to use `$config_dir/blocklists`.

As for procd-seccomp, this is also causing issues. I receive the same crash loops message even with some extra calls added (set_tls, set_tid_address).

NautilusChan, 2022-03-21:
> procd

After I add this :
```
procd_add_jail_mount "/mnt/sda1"
procd_add_jail_mount "/mnt/sda2"
procd_add_jail_mount "/mnt/sdb1"

The transmission can read the data. But  that means I can't download any more to the folders ?

And  there was an error when I start transmission  after I added this:
```
procd_add_jail_mount_rw "/mnt/sda1"
procd_add_jail_mount_rw "/mnt/sda2"
procd_add_jail_mount_rw "/mnt/sdb1"

NautilusChan, 2022-03-21:
> Probably `procd-ujail` was not installed and now became a part of the default OpenWrt build. In case of `procd-ujail` missing, processes are just run without jail. Once I find time for that I'll add an option (or rather: a list) for additional read-only paths to `/etc/config/transmission`, so people (including myself) won't have to edit the init script.

I checked the software list . I'm sure  that `procd-ujail` had been installed. The version of `procd-ujail` is 2022-02-19-ad652490-3

dangowrt, 2022-03-21:
> And there was an error when I started transmission after I added this:
> 
> ```
> procd_add_jail_mount_rw "/mnt/sda1"
> procd_add_jail_mount_rw "/mnt/sda2"
> procd_add_jail_mount_rw "/mnt/sdb1"
> ```

This could be explained if you also have `/mnt` already listed as read-only jail mount (which is not possible due to current limitations in procd-ujail)

neheb, 2024-06-09:
This looks fixed based on all the references.

closed 2024-06-09 by neheb