openwrt/packages #17681: crowdsec/geoipupdate: enhancement request for offline database download

erdoukki, 2022-01-23:
Maintainer: Gérald Kerma / @erdoukki
Compile tested: (aarch64_cortex-a53, espressoBin, OpenWrt master and 21.02)
Run tested: (aarch64_cortex-a53, espressoBin, OpenWrt 21.02.x, tests done)

Description:

As refereed in [Idea/GeoIP: offline/update/share database](https://github.com/crowdsecurity/crowdsec/issues/1196)

> CrowdSec OpenWrt package is downloading the GeoIP databases at postinstall, this may be a huge download on some users ISP, but it also may not work at all in case users want to integrate the crowdsec package in the OpenWrt firmware image.

Thanks to @clayface the package geoipupdate is available since PR: https://github.com/openwrt/packages/pull/17171

The question here is like the same from https://github.com/crowdsecurity/crowdsec/issues/1196

Note: Sorry for crosslink between OpenWrt/CrowdSec but the both communities may be interested

erdoukki, 2022-01-26:
Build: ok
Install: ok

```
gke@PRECISION:~/DEVEL/OWRT/packages$ git push upstream
Énumération des objets: 16, fait.
Décompte des objets: 100% (16/16), fait.
Compression par delta en utilisant jusqu'à 8 fils d'exécution
Compression des objets: 100% (14/14), fait.
Écriture des objets: 100% (14/14), 37.45 Mio | 109.00 Kio/s, fait.
Total 14 (delta 2), réutilisés 0 (delta 0)
remote: Resolving deltas: 100% (2/2), completed with 2 local objects.
remote: warning: See http://git.io/iEPt8g for more information.
remote: warning: File net/crowdsec-geoipdatabase/files/GeoLite2-City_20220118/GeoLite2-City.mmdb is 69.78 MB; this is larger than GitHub's recommended maximum file size of 50.00 MB
remote: warning: GH001: Large files detected. You may want to try Git Large File Storage - https://git-lfs.github.com.
remote: 
remote: Create a pull request for 'geoipupdate' on GitHub by visiting:
remote:      https://github.com/erdoukki/packages/pull/new/geoipupdate
remote: 
To github.com:erdoukki/packages.git
 * [new branch]          geoipupdate -> geoipupdate
```

Is there any recommendation about maximum file size ?

Another potential issue is the license about redistribution of the DB, can anyone help to understand if there is any issue and if the package may be valid to be proposed upstream ?

ne20002, 2022-03-03:
I just tried to install CrowdSec packages  and faild on WRT3200ACM. The avaialable space for packages ins 55MB and with downloading the GeoLite2-City.mmdb with 26.3 MB size the installation failed with no space left on device.

I suggest to only install a minimum on OpenWrt and make it either downloadable after installation or even have it in a seperate package for installation (so it's optional for user).

erdoukki, 2022-03-04:
Thanks.for your feedback.
As this issue was open about particular problem of GeoIP download and time issue, you point a more bad effect of GeoIP size.
I will link your issue with upstream and look to a solution as you proposed.
They both make sense in terms of users view, but GeoIP is only a script call at install.
In this particular case you point that GeoIP have to be optional.

erdoukki, 2022-03-04:
I install by default the linux collection which contains :
-> https://hub.crowdsec.net/author/crowdsecurity/collections/linux
Which contains the geoip-enrich
-> https://hub.crowdsec.net/author/crowdsecurity/configurations/geoip-enrich

I will remove this from the post install script.

Thanks @ne20002

erdoukki, 2022-03-05:
https://github.com/openwrt/packages/pull/17989
this PR may fix the issue.

closed 2022-05-02 by erdoukki