openwrt/packages #17710: nano: some key default key binds are missing and there is no way to add them after deployment.

okibcn, 2022-01-27:
Maintainer: @hnyman 
Environment:  openwrt-21.02.1-ramips-mt76x8

Description: nano package doesn't have all the key binds. There are some missing key binds such as M-A, ^6, M-6 for select and copy selection when not all the line is selected, or multiple lines are selected. Without that feature, the nano editor package is severely limited.

An easy solution to that would be replacing --enable-tiny by all the disabling compile switches except for nanorc feature so end users could add any missing keybinds. Basically replacing the CONFIGURE_ARGS parameter in makefile with this one:
```
CONFIGURE_ARGS += \
                --disable-browser \
                --disable-color \
                --disable-comment \
                --disable-extra \
                --disable-help \
                --disable-histories \
                --disable-justify \
                --disable-libmagic \
                --disable-mouse \
                --disable-multibuffer \
                --disable-operatingdir \
                --disable-speller \
                --disable-tabcomp \
                --disable-wordcomp \
                --enable-linenumbers
```

hnyman, 2022-01-30:
I think that I will take another approach:

I will create a new "nano-plus" variant that removes the "tiny" build option to enable the key bindings that you are interested in, but also enables a few more selected features: multiple open files, nanorc, help, justify.

See #17736

There will be 
* old nano package, same as now, roughly 62 kB
* new nano-plus package, roughly 120 kB

hnyman, 2022-01-31:
Closed via 85cb71d8d

closed 2022-01-31 by hnyman