openwrt/packages #17766: dockerd compatibility problem with nftables

qiuzi, 2022-02-01:
@G-M0N3Y-2503 @neheb 
In the future, iptables will be replaced. Will dockerd be perfectly compatible with nftables?

brada4, 2022-02-08:
What is the exact compatibility problem you are referring to in the title? Docker works on all major Linux distributions, all of them being NFT for few years now.

qiuzi, 2022-02-08:
@brada4  should be transitioned to nftables use, iptables will slowly be phased out
the current startup script can remove iptables dependencies and reduce redundant components

G-M0N3Y-2503, 2022-02-08:
Thanks for the thought-provoking question, I'm moving house ATM so I haven't had much time to look into this.
I did find https://github.com/moby/moby/issues/26824 that I am yet to read, but, it might provide some guidance on this question in the interim.

brada4, 2022-02-08:
Docker supports iptables command line only. Enough is emulated via NFT iptables wrappers on any (?other?) system out there.
Thats is a question to Docker Inc. to support some novelty environment.

G-M0N3Y-2503, 2022-03-17:
It seems that docker still depends on `iptables` until something like https://github.com/moby/moby/issues/26824 is done.  
With the advent of https://github.com/openwrt/packages/issues/16818, I'm working on having `docker` use `iptables-nft`.

It does sound like a possible feature to remove any `iptables` dependencies and disable `iptables` in the `docker` config for a custom build, but that sounds like a separate issue to this one.

So from my perspective, this issue is a duplicate of #16818

brada4, 2022-03-17:
> @brada4 should be transitioned to nftables use, iptables will slowly be phased out the current startup script can remove iptables dependencies and reduce redundant components

In the official post iptables-nft is recommended as a compatibility option, on par with what official docker package repos do. (podman too, before you ask)

qiuzi, 2022-03-17:
iptables-nft is not the best solution, nft has other new features

G-M0N3Y-2503, 2022-03-17:
> In the official post iptables-nft is recommended as a compatibility option, on par with what official docker package repos do. (podman too, before you ask)

I'm not sure to what you are referring to, if you have links I'd appreciate them. 
Because from what I can tell using `iptables-nft` is how current official packages are supporting `netfilter`.  
see https://github.com/docker/docker-ce-packaging/blob/master/deb/common/control#L32

acooler15, 2022-03-18:
My Docker container has access to the local network but not the internet when use  `iptables-nft`😢

brada4, 2022-03-18:
There should be forward rule for each exposed port and one for all masquerade rule involving docker's internal bridge (in iptables-save)
Which way it does not work?

champtar, 2022-03-18:
With nftables you now have multiple tables, if a rule in fw4 table drop a packet,
the fact that it was accepted in the filter table (configured by iptables-nft) doesn't matter.
So you might need to add some allow rules in fw4 for docker to work.
can someone show the output of `iptables-nft-save` and `nft list ruleset` when running docker ?

brada4, 2022-03-18:
[centos8-tables.txt](https://github.com/openwrt/packages/files/8306429/centos8-tables.txt)
[centos8-nft.txt](https://github.com/openwrt/packages/files/8306437/centos8-nft.txt)
"how it should look"

champtar, 2022-03-19:
@brada4 you just don't have any firewall configured on your CentOS8 box

smallprogram, 2026-10-14:
[image]

[image]