openwrt/packages #17773: Tinc: Makefile syntax error

jauntywunderkind, 2022-02-01:
Maintainer: @neheb 
Environment: ipq8064, netgear r7800 (nighthawk x4s), trunk
trunk r7800

Description:

Trying to build tinc produces this error:

```
Makefile:801: *** missing separator.  Stop.
```

Fuller log:

```
configure: WARNING: unrecognized options: --disable-nls, --with-kernel
touch /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git/.configured_f0d39d1547a48a7936c082769259d686
rm -f /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git/.built
touch /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git=tinc-1.1-git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99  -I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git=tinc-1.1-git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99  -I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include " LDFLAGS="-L/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -znow -zrelro " make  -C /usr/local/src
/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git=tinc-1.1-git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" ;
Makefile:801: *** missing separator.  Stop.
make[4]: Entering directory '/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git'
make[4]: Leaving directory '/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git'
make[3]: *** [Makefile:78: /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/tinc-1.1-git/.built] Error 2
make[3]: Leaving directory '/usr/local/src/openwrt/feeds/packages/net/tinc'
time: package/feeds/packages/tinc/compile#9.26#1.98#11.29
    ERROR: package/feeds/packages/tinc failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/tinc/compile] Error 1
make[2]: Leaving directory '/usr/local/src/openwrt'
make[1]: *** [package/Makefile:110: /usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/usr/local/src/openwrt'
```

neheb, 2022-02-02:
Strange. Never seen this.

BKPepe, 2022-02-02:
@jauntywunderkind Can you share your Makefile, which you try to compile?

BKPepe, 2022-02-07:
Closing due to inactivity.

closed 2022-02-07 by BKPepe