openwrt/packages #17780: dbus: switch from cmake to autotools for packaging

arkanoid87, 2022-02-03:
Maintainer: @feckert 
Environment: mips_24kc, gl-x300b (all *big endian* devices affected), snapshot

Description:

while debugging a problem on ModemManager not booting on *big endian* platforms, it ended up as a build issue with `dbus` that does not define `WORDS_BIGENDIAN` when built with cmake instead of autotools

I've [opened an issue on dbus bug tracker,](https://gitlab.freedesktop.org/dbus/dbus/-/issues/375) and they suggested to use autotools instead of cmake, as AC_C_BIGENDIAN Autoconf macro handles this.

In the meanwhile just for a quick test I've been suggested tro try adding `TARGET_CFLAGS += $(if $(CONFIG_BIG_ENDIAN),-DWORDS_BIGENDIAN)` in current cmake Makefile, and that fixed the problem.

Please follow the whole story here:
https://github.com/openwrt/packages/issues/17284
https://gitlab.freedesktop.org/dbus/dbus/-/issues/375

aleksander0m, 2022-02-03:
> I've [opened an issue on dbus bug tracker,](https://gitlab.freedesktop.org/dbus/dbus/-/issues/375) and they suggested to use autotools instead of cmake, as AC_C_BIGENDIAN Autoconf macro handles [...]

Worth noting that upstream stated that the cmake port was targeting Windows setups. I would definitely switch to autotools instead of adding the `TARGET_CFLAGS` workaround, which has all the proper bits and pieces to detect BE systems. Also, who knows what else is not covered by the cmake port :/

arkanoid87, 2022-02-04:
In the meanwhike, the ticket https://gitlab.freedesktop.org/dbus/dbus/-/issues/375
Changed title in "cmake build effectively assumes all platforms are little-endian"

feckert, 2022-02-04:
So that we can fix this as soon as possible.
Because this also effects the openwrt-21.02 stable branch.
I think we have to switch from cmake to autotools for the modemmanager packages.
How are we going to do that?
@arkanoid87 Are you planning the changeover? Don't want us to do the work twice.

arkanoid87, 2022-02-04:
modemmanager package does not need any change, as ModemManager uses glib2 to build the messages sent to dbus and I've checked that they are correct on both LE and BE platforms. It's just dbus that needs to move from cmake to autotools to fix BE issue.

Answering you question specifically, I lack the knowledge to write a Makefile (it would be my first ever).

feckert, 2022-02-04:
@arkanoid87 Sorry, I misspoke. I meant dbus, of course.
Ok, then I'll do that migrade cmak to autotools.
I will create a PR, it would be nice if you could test it.

arkanoid87, 2022-02-04:
Sure! Thanks

feckert, 2022-02-04:
Please have a look at https://github.com/openwrt/packages/pull/17792

feckert, 2022-02-04:
It looks like openwrt-21.02 is not affected, as it still relies on autotools.
https://github.com/openwrt/packages/blob/openwrt-21.02/utils/dbus/Makefile

closed 2022-02-04 by feckert

arkanoid87, 2022-02-04:
on the bright side, I've potentially spotted a serious issue for upcoming releases
on the dark side, I'd have saved many hours just by git dissecting, or trying stable first.

But my the page for my device says it wants snapshot, so I did.