openwrt/packages #17803: Installing grep package causes ssh startup to display an error message

SiNONiMiTY, 2022-02-05:
I needed to install grep that supports PCRE. Upon installing, everytime I connect to the shell, ash is giving an error that fgrep is missing.

```
BusyBox v1.35.0 (2022-02-02 02:37:21 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r18728-fb6cf22866
 -----------------------------------------------------
-ash: fgrep: not found
```

```
Package: grep
Version: 3.7-2
Depends: libc, libpcre
Status: install user installed
Section: utils
Architecture: aarch64_generic
Size: 98858
Filename: grep_3.7-2_aarch64_generic.ipk
Description: The grep command searches one or more input files for lines
 containing a match to a specified pattern. By default, grep
 prints the matching lines.
Installed-Time: 1644035133
```

The grep package should have symlinks for the functions it replaces.

Uninstalling the grep package will remove the error.

neheb, 2022-02-05:
that is extremely strange. Try editing:

```
--- a/package/base-files/files/etc/profile
+++ b/package/base-files/files/etc/profile
@@ -3,7 +3,7 @@
 [ -f /etc/banner ] && cat /etc/banner
 [ -n "$FAILSAFE" ] && cat /etc/banner.failsafe

-fgrep -sq '/ overlay ro,' /proc/mounts && {
+grep -Fsq '/ overlay ro,' /proc/mounts && {
        echo 'Your JFFS2-partition seems full and overlayfs is mounted read-only.'
        echo 'Please try to remove files from /overlay/upper/... and reboot!'
 }
```

that's /etc/profile on the router.

SiNONiMiTY, 2022-02-06:
@neheb editing that line fixes the issue. Thanks for pushing a quick fix.

neheb, 2022-02-07:
fix was merged upstream. closing.

closed 2022-02-07 by neheb

xtremeperf, 2022-11-27:
This error happens when Bash is not installed.

`/usr/libexec/fgrep` and `/usr/libexec/egrep` are simple `grep` wrapper scripts which invoke Bash to `exec` the cmdline using grep instead (the shebang line specified bash as the interpreter to be used).  However, bash isn't installed by default... and the wrapper scripts are perfectly functional when invoked by busybox's ash (sh) shell installed by default.

It appears the simplest logical solution here is to change the "shebang lines" in those two files to be: `#!/bin/sh`

I've tested this solution myself and appears to resolve the issue completely.  Just figured I'd share my thoughts.  I'm not sure if that's an upstream grep issue or what.  I've not contributed to OpenWRT before, but if a pull request is needed let me know and I'll look into submitted the changes.

altasol, 2025-09-24:
Reopen, as 24.10.3 has regression. On ssh login:

BusyBox v1.36.1 (2025-09-19 21:19:38 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.3, r28872-daca7c049b
 -----------------------------------------------------
-ash: fgrep: not found

and 

/usr/libexec/fgrep does not exist on system.

System is sysupgrade from v23.