openwrt/packages #17806: asterisk: console logging truncated for lines above 996 characters

pprindeville, 2022-02-05:
# Issue template (remove lines from top till here)

Maintainer: @jslachta
Environment: x86_64, generic, HEAD

Description:

I'm seeing truncated output when tracing PJSIP such as:
```
SIP/2.0 200 OK
Call-ID: 41dd13f4-b0c2-4483-adce-4df18231eba0
CSeq: 22390 OPTIONS
From: <sip:syringa2-endpoint@voip.syringanetworks.net>;tag=8caf4647-ecfd-4b10-bf94-9907fd4be383
To: <sip:voip.syringanetworks.net>;tag=sip+1+32a700b1+d16ebf88
Via: SIP/2.0/UDP 45.33.216.244:5060;received=45.33.216.244;rport=5060;branch=z9hG4bKPj6d3f6a7a-1184-4d44-b4f1-00516626b4ad
Content-Length: 0
Supported: resource-priority, siprec, 100rel
Server: DC-SIP/2.0
Organization: Metaswitch Networks
Allow-Events: message-summary, refer, dialog, line-seize, presence, call-info, as-feature-event, calling-name, ua-profile
Allow: INVITE, ACK, CANCEL, BYE, REGISTER, OPTIONS, PRACK, UPDATE, SUBSCRIBE, NOTIFY, REFER, INFO, PUBLISH
Accept-Encoding: identity
Accept: application/sdp, application/simple-message-summary, message/sipfrag, application/isup, application/x-simple-call-service-info, multipart/mixed, application/broadsoft, application
```

Note the incomplete last line.

jslachta, 2022-02-05:
Hi @pprindeville,

this issue should go to the OpenWrt/telephony repository.

Please, try to replicate this issue on the non-OpenWrt system to determine whether it is an asterisk packaging/compilation issue or an upstream issue. If this is an Asterisk-related OpenWrt-only issue, please report an issue here https://github.com/openwrt/telephony.

I am closing this issue.

Thank you.

Kind regards,
Jiri

closed 2022-02-05 by jslachta

pprindeville, 2022-02-06:
Moved to [#737](https://github.com/openwrt/telephony/issues/737)