openwrt/packages #17809: ddns-scripts: Request addition of DDNS provider PAIRDOMAINS.COM

VA1DER, 2022-02-06:
Maintainer: @moormaster 
Environment: All

Description:

Request addition of ddns provider pairdomains.com.  They are a dyndns2-compatible provider.  Config file as follows:
**pairdomains.com.json**
```
{
	"name": "pairdomains.com",
	"ipv4": {
		"url": "https://[USERNAME]:[PASSWORD]@dynamic.pairdomains.com/nic/update?hostname=[DOMAIN]&myip=[IP]",
		"answer": "good|nochg"
	},
	"ipv6": {
		"url": "https://[USERNAME]:[PASSWORD]@dynamic.pairdomains.com/nic/update?hostname=[DOMAIN]&myip=[IP]",
		"answer": "good|nochg"
	}
}

```
[pairdomains.com.json.zip](https://github.com/openwrt/packages/files/8009366/pairdomains.com.json.zip)

moormaster, 2022-02-06:
@feckert I think the mention in the issue description was meant for you :)

rmilecki, 2022-09-27:
@VA1DER can you submit a pull request, please? It seem you have ready code for that.