openwrt/packages #17816: [21.02] tvheadend: does not execute first run

BKPepe, 2022-02-06:
Maintainer: @M95D
Environment: OpenWrt 21.02.1, [Seeedstudio mini router](https://www.seeedstudio.com/Dual-GbE-Carrier-Board-with-4GB-RAM-32GB-eMMC-RPi-CM4-Case-p-5029.html) (Raspberry Pi 4 CM, bcm27xx/aarch64_cortex-a7)

Description:

Once I started tvheadend for the first time, it was asking me for credentials, which I dont have.
```
root@OpenWrt:~# opkg install tvheadend
Installing tvheadend (4.2.8-5) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/aarch64_cortex-a72/packages/tvheadend_4.2.8-5_aarch64_cortex-a72.ipk
Installing libffi (3.3-2) to root...
Downloading https://downloads.openwrt.org/releases/21.02.1/packages/aarch64_cortex-a72/packages/libffi_3.3-2_aarch64_cortex-a72.ipk
Configuring libffi.
Configuring tvheadend.
root@OpenWrt:~# /etc/init.d/tvheadend start
root@OpenWrt:~# ps w | grep tvheadend
11850 root     61036 S    /usr/bin/tvheadend -B -c /etc/tvheadend --ipv6
11952 root      1228 S    grep tvheadend
```

M95D, 2022-02-06:
Already fixed here:
https://github.com/openwrt/packages/pull/16475

BKPepe, 2022-02-06:
I am afraid that it is fixed only for OpenWrt master, which makes this issue still valid for OpenWrt 21.02.

M95D, 2022-02-06:
I don't think I can help with that.

BKPepe, 2022-02-07:
Hmm. So, the issue is in the stable branch. What can be done about it:

1. Someone can send a pull request against that branch
2. Should we, I mean me, cherry-pick it from ``master`` branch to ``openwrt-21.02`` branch?
I didn't look at it closely, but in the pull request, you added tvheadend user and use it. Is it possible that some permissions migration would be necessary or not?

walterav1984, 2022-02-14:
I can confirm this is currently an issue on Raspberry Pi 1 with 21.02.1 release, but I'm almost certain this worked at the end of august 2021 as in 4.0 (old/previous tvheadend 4.0.x on 19.07.x style/way meaning running as ROOT no hdhomerun nor satipclient) or maybe an early 4.2.x version without @M95D improvements?

Than at that time in august 27 2021 the master branch of OpenWrt got a updated tvheadend pull #16475 request running not as Root user and some other settings like hdhomerun/satip/dev/dvb support. I guess somewhere in between some parts of the master commit which was working are not fully backported see yet to 21.02.1 leading to this broken release see August 17, 2021 and August 19, 2021 .

See current PR since then:
https://github.com/openwrt/packages/commit/fc835bcaac1cbf357c713511b3827fac2307afa2
https://github.com/openwrt/packages/commit/4f990b7cd66b63ad44674c0bcc138683fee1bf3c
https://github.com/openwrt/packages/commit/d7c78f83be33e2fce0c3cbb8335050f19968699d

Nov 1, 2021 some hdhomerun backporting
https://github.com/openwrt/packages/commit/f2300cd651abbe74d648637216bfc5be20c1e531

Aug 27, 2021 master tvheadend 4.2,x without root user and with hdhomerun/satip
https://github.com/openwrt/packages/pull/16475

Aug 19, 2021 @BKPepe initial 21.02.1 cherrypicked !!!
https://github.com/openwrt/packages/commit/0240320e8dee02e69886481290cd20250fbc1018

Aug 17, 2021 master tvheadend 4.2.8 with root user without hdhomerun/satip
https://github.com/openwrt/packages/commit/326da3dbbc57b6c44bc0d0401529531c7d5dfb89
https://github.com/openwrt/packages/commit/4ddc4a613551c90a665acfb7fe210eb87ae900e8

M95D, 2022-02-15:
There is a problem with v4.2.8 and in current tvheadend master branch too. The usual command run by openwrt package to create a new config with admin account is **tvheadend -C -A**, but tvheadend aborts before saving the new configuration to /etc/tvheadend. It's a tvheadend bug, not package or openwrt.
I included a workaround in https://github.com/openwrt/packages/pull/16475 , but that pull request also includes the changes to root -> regular user account, and it's all in the same startup command.
There are some reported problems that tvheadend when run as a regular user can't send/receive multicast messages used by SAT>IP. Probably that's why the pull request was not selected for the stable branch of OpenWrt.

WORKAROUND until this is fixed:
Manually run:
`tvheadend -c /etc/tvheadend -B -C  & TVH_PID=$! ; sleep 10 ; kill $TVH_PID`
It will start the server for 10s and then stop it without using -A switch (the bug).

If you don't have console access to the router, disable the tvheadend service, put the command in System / Startup / Local Startup, reboot once, remove it form startup (so it won't run again on startup) and re-enable the tvheand service.

M95D, 2022-02-15:
Good news: I just found out I can create pull requests to specific branches of openwrt. :-)
I also reported the problem to tvheadend bug tracker: https://tvheadend.org/issues/6140 . Let's hope someone there has the time to investigate and fix it.

walterav1984, 2022-02-16:
@BKPepe and @M95D this is indeed fixed now for 21.02.1 tested on the RPi1 by removing the earlier broken tvheadend package and deleting its settings, after that it just did a succesful 1st run to setup the config folder:
```
opkg remove tvheadend
rm -r /etc/tvheadend
#it only complained about /etc/config/tvheadend already existing so it created a /etc/config/tvheadend-conf
```

closed 2022-03-13 by BKPepe