openwrt/packages #17826: grep: fgrep / egrep uses /bin/bash

532910, 2022-02-08:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

SiNONiMiTY, 2022-02-08:
might be related
https://github.com/openwrt/packages/issues/17803

To resolve, try editing /etc/profile and replace the usage of fgrep to grep -F
```
-fgrep -sq '/ overlay ro,' /proc/mounts && {
+grep -Fsq '/ overlay ro,' /proc/mounts && {
        echo 'Your JFFS2-partition seems full and overlayfs is mounted read-only.'
        echo 'Please try to remove files from /overlay/upper/... and reboot!'
 }
```

brada4, 2022-02-08:
Suggestion:
```
#!grep -E
```

xtremeperf, 2022-11-27:
The '/usr/libexec/egrep-gnu' and '/usr/libexec/fgrep-gnu' wrappers for grep can be interpreted just fine under the busy shell, so logically their "shebang lines" should be:`#!/bin/sh` which resolves the issue whether bash is installed or not.

stangri, 2022-11-27:
@neheb @Zokormazo  Looks like, [this line](https://github.com/openwrt/packages/blob/c87537e1a752cd6f2966b313e1708b55881f87fa/utils/grep/Makefile#L26) has no effect on the resulting package.

neheb, 2022-11-27:
looks like it doesn't. I have no idea how to fix.

brada4, 2022-11-27:
You can call the target program in shebang line, no need to involve shell binary in process.

PR #26403 (2025-04-29): grep: fix that egrep and fgrep workaround doesn't work
Maintainer: Julen Landa Alustiza / @Zokormazo
Compile tested: (rockchip, NanoPi R2S, master branch)
Run tested: the same as above.

Description:

Commit 07b6eec21f57c3cb391b0daf89240b7632b2a49f doesn't work at least now, because `package.mk` initializes the variables to the default values.  You have to modify the variable after including `package.mk` as @zhanhb [commented](https://github.com/openwrt/packages/pull/16551#issuecomment-1073797207).
Fixes: #17826
Related: #16548, #17803, #23778, #16551
closed 2025-04-29 by 1715173329