openwrt/packages #17856: boost error: array: No such file or directory

predators46, 2022-02-13:
I'm using boost-1.59.0 and openwrt 19.07.x. the reason i use boost-1.59.0 i made mysql-5.7 in openwrt. how to fix this error

`target-aarch64_generic_glibc/usr/include/boost/lexical_cast/detail/converter_lexical.hpp:47:10: fatal error: array: No such file or directory
 #include <array>`

neheb, 2022-02-14:
ping @ClaymorePT what a strange error...

predators46, 2022-02-15:
Fix thanks

predators46, 2022-02-15:
> ping @ClaymorePT what a strange error...

i just finished making it and now i try it and it works

`root@OpenWrt:~# ./mysql -V
./mysql  Ver 14.14 Distrib 5.7.10, for Linux (aarch64) using  EditLine wrapper
root@OpenWrt:~# ./mysqld                                                  
mysqld: Can't change dir to '/usr/data/' (Errcode: 2 - No such file or directory)
2022-02-15T17:06:39.300869Z 0 [Warning] TIMESTAMP with implicit DEFAULT value is deprecated. Please use --explicit_defaults_for_timestamp server option (see documentation for more details).
2022-02-15T17:06:39.301022Z 0 [Warning] Insecure configuration for --secure-file-priv: Current value does not restrict location of generated files. Consider setting it to a valid, non-empty path.
2022-02-15T17:06:39.301126Z 0 [Note] ./mysqld (mysqld 5.7.10) starting as process 30682 ...
2022-02-15T17:06:39.301217Z 0 [ERROR] Can't find error-message file '/usr/share/errmsg.sys'. Check error-message file location and 'lc-messages-dir' configuration directive.
2022-02-15T17:06:39.306483Z 0 [Warning] Can't create test file /usr/data/OpenWrt.lower-test
2022-02-15T17:06:39.306555Z 0 [Warning] Can't create test file /usr/data/OpenWrt.lower-test
2022-02-15T17:06:39.306677Z 0 [ERROR] failed to set datadir to /usr/data/
2022-02-15T17:06:39.306739Z 0 [ERROR] Aborting

2022-02-15T17:06:39.306762Z 0 [Note] Binlog end
2022-02-15T17:06:39.306912Z 0 [Note] 
root@OpenWrt:~#`

predators46, 2022-02-15:
Hy @ClaymorePT 

Can you help me. I want to use boost-1.59.0 on openwrt 21 I see python2 is gone. i have changed boost makefile and deleted python2 and i rebuilt boost without python2 but it didn't work. can you help me to remove python2 in boost 1.59.0. thanks

[Makefile](https://github.com/ClaymorePT/packages/blob/fff766fecdc2f1906e6bc9d5aa4ca61ddff72b1f/libs/boost/Makefile)

ClaymorePT, 2022-02-16:
Try this [Makefile](https://github.com/ClaymorePT/packages/blob/boost_1.59.0_no_python2/libs/boost/Makefile). If it does not work, please tell me the target to which you are trying to build boost to.

I simple grabbed the most recent version and changed the version and checksum to the right values for boost 1.59.0 . I did not test this since this was just an over-the-knee fix.

predators46, 2022-02-16:
> @predators46 Try this [Makefile](https://github.com/ClaymorePT/packages/blob/boost_1.59.0_no_python2/libs/boost/Makefile). If it does not work, please tell me the target to which you are trying to build boost to.
> 
> I simple grabbed the most recent version and changed the version and checksum to the right values for boost 1.59.0 . I did not test this since this was just an over-the-knee fix.

i will try it now. and soon I will report the results. Thank you

predators46, 2022-02-16:
> @predators46 Try this [Makefile](https://github.com/ClaymorePT/packages/blob/boost_1.59.0_no_python2/libs/boost/Makefile). If it does not work, please tell me the target to which you are trying to [...]

doesn't seem to work. there is no dependency creation process that is done by boost. boost is made immediately. and python3 not created

predators46, 2022-02-16:
Hy @neheb 

i was forced to use python2 to boost no other option. but i found problem when i build python2 on openwrt 21.02.1. how to add zlib dependencies in python-base-src

`Package python-base-src is missing dependencies for the following libraries:
libz.so.1`

neheb, 2022-02-16:
ping @jefferyto

ClaymorePT, 2022-02-16:
> doesn't seem to work. there is no dependency creation process that is done by boost. boost is made immediately. and python3 not created

@predators46 let me see what I can do on my side. I don't have much free time at the moment, but it should be possible to build boost 1.59 in recent openwrt versions.

predators46, 2022-02-17:
> ping @jefferyto

this problem has been solved

predators46, 2022-02-17:
i've overcome boost python2. I will close this

@neheb @ClaymorePT  thanks

closed 2022-02-17 by predators46

ClaymorePT, 2022-02-17:
@predators46 cool!

predators46, 2022-02-18:
> @predators46 cool! Glad you were able to overcome.

i tried mysql 8 with boost-1.68.0

`
BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.4, r11208-ce6496d796
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt: ./mysql -V                                                
./mysql  Ver 8.0.15 for Linux on aarch64 (Source distribution)                                                        
root@OpenWrt: ./mysqld                                                  
2022-02-14T01:22:26.661572Z 0 [System] [MY-010116] [Server] /root/mysqld (mysqld 8.0.15) starting as process 3997
2022-02-14T01:22:26.661773Z 0 [ERROR] [MY-010338] [Server] Can't find error-message file '/share/errmsg.sys'. Check error-message file location and 'lc-messages-dir' configuration directive.
2022-02-14T01:22:26.670225Z 0 [Warning] [MY-010091] [Server] Can't create test file /data/OpenWrt.lower-test
2022-02-14T01:22:26.670296Z 0 [Warning] [MY-010159] [Server] Setting lower_case_table_names=2 because file system for /data/ is case insensitive
2022-02-14T01:22:26.670388Z 0 [ERROR] [MY-010123] [Server] Fatal error: Please read "Security" section of the manual to find out how to run mysqld as root!
2022-02-14T01:22:26.670690Z 0 [ERROR] [MY-010119] [Server] Aborting
2022-02-14T01:22:26.671569Z 0 [System] [MY-010910] [Server] /root/mysqld: Shutdown complete (mysqld 8.0.15)  Source distribution.
root@OpenWrt:`

ClaymorePT, 2022-02-18:
@predators46 Doesn't really ring a bell. Maybe running mysql with root permissions requires special configurations, as indicated in the error message.

predators46, 2022-02-18:
> @predators46 Doesn't really ring a bell. Maybe running mysql with root permissions requires special configurations, as indicated in the error message.

no problem. what I'm showing is just a quick try. Thanks