openwrt/packages #17869: PHP7 Keep it, please

reinerotto, 2022-02-15:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

brada4, 2022-02-15:
PHP8 release notes describe on how to disable new security to make PHP7 code work.
You still got time until upstream EOL on 28/11.

BKPepe, 2022-02-15:
@reinerotto You didn't even bother to fill the pull request template. :-( You just write your own pull request title, and that's it. 

PHP7 is in OpenWrt 19.07 and OpenWrt 21.02 branches. It was removed in the master branch (daily snapshots). For example, the version used in 19.07 is deprecated out of date. Would you like to take of maintainership?

~ Closing for given reasons.

closed 2022-02-15 by BKPepe