openwrt/packages #17871: miniupnpd: can't map ports and show "There are no active redirects." in LuCI UI

escape0707, 2022-02-15:
### Maintainer

@stintel @ldir-EDB0 @neheb

### Environment

* Model | Xiaomi Redmi Router AC2100
* Architecture | MediaTek MT7621 ver:1 eco:3
* Target Platform | ramips/mt7621
* Firmware Version | OpenWrt SNAPSHOT r18806-2120cad38d / LuCI Master git-22.025.79016-22e2bfb
* Kernel Version | 5.10.100

### Description

As the title says, miniupnpd can't map requested ports successfully for applications and shows "There are no active redirects." in the LuCI web interface.

I setup this environment by:

1. Flashed the official snapshot version just after I got my hand on this Redmi Router, with the official guide provided method. 
2. Setup 2.4G & 5G WiFi. 
3. Setup PPPoE on WAN. 
4. Installed `luci-app-upnp` through `opkg`.

### More logs / configs

[image]
[qbittorrent_log.txt](https://github.com/openwrt/packages/files/8069794/qbittorrent_log.txt)
[longread.txt](https://github.com/openwrt/packages/files/8069795/system_log.txt)
[teredo_log.txt](https://github.com/openwrt/packages/files/8069796/teredo_log.txt)
[etc_config_upnpd.txt](https://github.com/openwrt/packages/files/8069797/upnpd_conf.txt)
[ip_addr_show.txt](https://github.com/openwrt/packages/files/8069801/ip_addr_show.txt)
[nftables.txt](https://github.com/openwrt/packages/files/8069802/nftables.txt)
[opkg_info_miniupnpd.txt](https://github.com/openwrt/packages/files/8069803/opkg_info_miniupnpd.txt)

stintel, 2022-02-15:
Redirects are there:
```
table inet miniupnpd {
	chain forward {
		type filter hook forward priority -25; policy accept;
		iif "pppoe-wan" th dport 49965 @nh,128,32 0xc0a8010b @nh,72,8 0x11 accept
		iif "pppoe-wan" th dport 9564 @nh,128,32 0xc0a8010b @nh,72,8 0x6 accept
		iif "pppoe-wan" th dport 9564 @nh,128,32 0xc0a8010b @nh,72,8 0x11 accept
	}
}
table ip miniupnpd {
	chain prerouting {
		type nat hook prerouting priority dstnat; policy accept;
		iif "pppoe-wan" udp dport 49965 dnat to 192.168.1.11:49965
		iif "pppoe-wan" tcp dport 9564 dnat to 192.168.1.11:9564
		iif "pppoe-wan" udp dport 9564 dnat to 192.168.1.11:9564
	}

	chain postrouting {
		type nat hook postrouting priority srcnat; policy accept;
	}
}
table ip6 miniupnpd {
	chain prerouting {
		type nat hook prerouting priority dstnat; policy accept;
	}

	chain postrouting {
		type nat hook postrouting priority srcnat; policy accept;
	}
}

```
Please report a LuCI issue instead.

escape0707, 2022-02-15:
@stintel Could you suggest me some more accurate ways to test which part is malfunctioning? Thanks!

stintel, 2022-02-15:
> That's what confused me, too. As application's like qbittorrent and windows teredo all says they can't be connected from outside.

Don't trust the application, trust tcpdump. I actually verified miniupnpd-nftables like that recently, because somewhere in some horrible 1500+ message thread in the forum where different issues are discussed, making it impossible to follow anything, and reminding me why I used to stay away from forums, someone complained that it didn't work.

On a remote host:
```
$ echo foo | nc -u 94.225.x.x 3074
```

```
# tcpdump -ni switch.54 port 3074
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on switch.54, link-type EN10MB (Ethernet), capture size 262144 bytes
16:40:37.990057 IP 94.225.x.x.38515 > 192.168.54.35.3074: UDP, length 4
```

escape0707, 2022-02-15:
> trust tcpdump

I'll go and learn about this tool now! Thanks for the information!

stintel, 2022-02-15:
If in fact that confirms the port forward doesn't work, please change the title of this issue to reflect just that. The fact that LuCI doesn't show the redirects should be fixed in LuCI so is not relevant for this issue (tracker).

escape0707, 2022-02-15:
> The fact that LuCI doesn't show the redirects should be fixed in LuCI so is not relevant for this issue (tracker).

 Duly noted

stintel, 2022-02-15:
My gut feeling says it's related to using ppp. @dangowrt reported a similar problem [here](https://github.com/openwrt/packages/pull/17094#issuecomment-968299497).

escape0707, 2022-02-15:
> My gut feeling says it's related to using ppp.

Do you think if I "use another router to do the PPPoE dial, connect my Redmi OpenWRT router to the first one's LAN port and use DHCP to connect to the Internet, then manually forward all ports from the first router to OpenWRT" will help diagnose this problem?

escape0707, 2022-02-15:
I'll have to try that tomorrow, parents are about to sleep. Thank you for helping me trouble shooting, nice sir!

escape0707, 2022-02-16:
Sadly, with this config, I get a lot of:

```
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: HTTP REQUEST from [::ffff:192.168.1.11]:33287 : POST /ctl/IPConn (HTTP/1.1)
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: Host: 192.168.1.1:5000
Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:2#AddPortMapping
Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: AddPortMapping: ext port 10659 to 192.168.1.11:10659 protocol TCP for: qBittorrent/4.5.0alpha1 leaseduration=604800 rhost=
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: UPnP permission rule 0 matched : port mapping accepted
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: Check protocol tcp for port 10659 on ext_if wan 192.168.2.116, 7402A8C0
Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: redirecting port 10659 to 192.168.1.11:10659 protocol TCP for: qBittorrent/4.5.0alpha1
Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: Returning UPnPError 501: ActionFailed
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: HTTP REQUEST from [::ffff:192.168.1.11]:33957 : POST /ctl/IPConn (HTTP/1.1)
Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: Host: 192.168.1.1:5000
```

snakwu, 2022-02-16:
> Sadly, with this config, I get a lot of:
> 
> ```
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
> Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: HTTP REQUEST from [::ffff:192.168.1.11]:33287 : POST /ctl/IPConn (HTTP/1.1)
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: Host: 192.168.1.1:5000
> Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:2#AddPortMapping
> Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: AddPortMapping: ext port 10659 to 192.168.1.11:10659 protocol TCP for: qBittorrent/4.5.0alpha1 leaseduration=604800 rhost=
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: UPnP permission rule 0 matched : port mapping accepted
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: Check protocol tcp for port 10659 on ext_if wan 192.168.2.116, 7402A8C0
> Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: redirecting port 10659 to 192.168.1.11:10659 protocol TCP for: qBittorrent/4.5.0alpha1
> Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: Returning UPnPError 501: ActionFailed
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: rule with label 'qBittorrent Enhanced/4.4.0.10' is not a IGD pinhole
> Wed Feb 16 02:45:44 2022 daemon.info miniupnpd[6819]: HTTP REQUEST from [::ffff:192.168.1.11]:33957 : POST /ctl/IPConn (HTTP/1.1)
> Wed Feb 16 02:45:44 2022 daemon.debug miniupnpd[6819]: Host: 192.168.1.1:5000
> ```
> 
> The first level router is set to dial up through PPPoE, then set the OpenWrt router as DMZ. When I manually forward all ports also in OpenWrt, I can connect to my machine from the Internet. But when I switch to UPnP, UPnP just won't work, and applications reports so, too.
> 
> [etc_config_upnpd.txt](https://github.com/openwrt/packages/files/8076376/etc_config_upnpd.txt)
> 
> [ip_addr_show.txt](https://github.com/openwrt/packages/files/8076393/ip_addr_show.txt)

me too！openwrt pppoe

escape0707, 2022-02-16:
@snakwu I think you are facing a completely different problem. You should open an independent issue addressing it.

escape0707, 2022-02-16:
Actually, I don't know what's the meaning of this table:

```
table inet miniupnpd {
	chain forward {
		type filter hook forward priority -25; policy accept;
		iif "pppoe-wan" th dport 49965 @nh,128,32 0xc0a8010b @nh,72,8 0x11 accept
		iif "pppoe-wan" th dport 9564 @nh,128,32 0xc0a8010b @nh,72,8 0x6 accept
		iif "pppoe-wan" th dport 9564 @nh,128,32 0xc0a8010b @nh,72,8 0x11 accept
	}
}
```

The default policy in its base chain is `ACCEPT`, while all rules in it says `ACCEPT`, too. What does the `@nh` filtering verifies? This table currently does nothing because so far there isn't anything malicious that needed to be rejected, yet?

escape0707, 2022-02-16:
I manually specified my `external_ip` in `/etc/config/upnpd` and UPnP set up the firewall rules successfully now. But I'm still getting the similar behavior that tcpdump shows incoming testing packets only on `wan` not `br-lan`.

escape0707, 2022-02-16:
See https://stackoverflow.com/questions/38162173/how-is-the-order-of-tables-chains-in-nftables-arranged

I think the reason this won't work is that currently `fw4` and `miniupnpd` both create their own (ipv4) table, which both contain a `hook prerouting priority dstnat` base chain. Then, one of them is ignored. It seems that the `miniupnpd` one comes later, so it is ignored.

escape0707, 2022-02-16:
I think I know where the problem is. If I add any port forward rule manually and UPnP seems to work. but I still need to do more tests.

jow-, 2022-02-16:
When at least one port forward is defined, certain additional rules are enabled by firewall4, like the rule that automatically accepts all inbound traffic that is related to a DNAT'ed connection

escape0707, 2022-02-16:
Yes, I diffed those nft rulesets and saw several 		`ct status dnat accept`. That is the reason why only enabling miniupnpd is not working.

escape0707, 2022-02-16:
After set up UPnP and add any firewall rules through LuCI, everything works just fine.

escape0707, 2022-02-17:
@jow-
I mean, when I only installled `luci-app-upnp` and enabled upnp, then launch an app (qbittorrent) to make upnp port forward request, I will get a nftables ruleset like this:

<summary><details>
<pre><code>
table inet fw4 {
	chain input {
		type filter hook input priority filter; policy accept;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
		iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
		iifname "pppoe-wan" jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
	}

	chain forward {
		type filter hook forward priority filter; policy drop;
		ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
		iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
		iifname "pppoe-wan" jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
		jump handle_reject
	}

	chain output {
		type filter hook output priority filter; policy accept;
		oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
		ct state established,related accept comment "!fw4: Allow outbound established and related flows"
		oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
		oifname "pppoe-wan" jump output_wan comment "!fw4: Handle wan IPv4/IPv6 output traffic"
	}

	chain handle_reject {
		meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic"
		reject comment "!fw4: Reject any other traffic"
	}

	chain syn_flood {
		limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
		drop comment "!fw4: Drop excess packets"
	}

	chain input_lan {
		jump accept_from_lan
	}

	chain output_lan {
		jump accept_to_lan
	}

	chain forward_lan {
		jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
		jump accept_to_lan
	}

	chain accept_from_lan {
		iifname "br-lan" counter packets 12 bytes 845 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
	}

	chain accept_to_lan {
		oifname "br-lan" counter packets 108 bytes 51604 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
	}

	chain input_wan {
		meta nfproto ipv4 udp dport 68 counter packets 0 bytes 0 accept comment "!fw4: Allow-DHCP-Renew"
		icmp type echo-request counter packets 3 bytes 252 accept comment "!fw4: Allow-Ping"
		meta nfproto ipv4 meta l4proto igmp counter packets 0 bytes 0 accept comment "!fw4: Allow-IGMP"
		ip6 saddr fc00::/6 ip6 daddr fc00::/6 udp dport 546 counter packets 0 bytes 0 accept comment "!fw4: Allow-DHCPv6"
		ip6 saddr fe80::/10 icmpv6 type . icmpv6 code { mld-listener-query . no-route, mld-listener-report . no-route, mld-listener-done . no-route, mld2-listener-report . no-route } counter packets 0 bytes 0 accept comment "!fw4: Allow-MLD"
		icmpv6 type { destination-unreachable, time-exceeded, echo-request, echo-reply, nd-router-solicit, nd-router-advert } limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-ICMPv6-Input"
		icmpv6 type . icmpv6 code { packet-too-big . no-route, parameter-problem . no-route, nd-neighbor-solicit . no-route, nd-neighbor-advert . no-route, parameter-problem . admin-prohibited } limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-ICMPv6-Input"
[... 302 lines cut ...]
		type nat hook postrouting priority srcnat; policy accept;
	}
}
</code></pre>
</details></summary>

A diff between those two shows that two

```
ct status dnat accept
```

rules was inserted just before 

```
jump reject_from_wan
```

and

```
jump reject_to_wan
```

which I believe are what make the port forwarded by miniupnpd works again.

dangowrt, 2022-02-17:
@stintel I'm not using PPPoE, got IPoE FTTH with the ISP-provided router operating in bridge-mode and OpenWrt device connected to it receiving a public IPv4 via DHCP. Yet rules added by miniupnpd didn't have any effect.

tiagogaspar8, 2022-02-17:
Same here, but my issue is with PCP and ipv6.
I see miniupnpd responding to the port open request, and I see the rules in nftables yet they do nothing.

Also, the lease file stays empty (I believe this is another separate issue)

stintel, 2022-02-17:
@jow already explained the reason [here](https://github.com/openwrt/packages/issues/17871#issuecomment-1041242806)
I have static port forwards so I have that rule and that's why things work for me.

tiagogaspar8, 2022-02-17:
I'm sorry, I'm confused, do we need to add something then?

I also saw that miniupnpd v2.3.0 fixes some issues with nftables and might be related to this, no?

escape0707, 2022-02-18:
@tiagogaspar8 https://github.com/openwrt/packages/issues/17871#issuecomment-1042615077

I think those rules added by luci firewall setting is the key point.

tiagogaspar8, 2022-02-18:
Yes, but what I mean is that I bleive the new release of miniupnpd fixes this issue because it corrects the location and the jumps that are currently broken.

Also, the port forwarding rules don't explain the reason why ipv6 PCP doesn't work.

escape0707, 2022-02-18:
Oh, I don't have IPv6 currently. And i guess i should update my OpenWrt and miniupnpd to test it again.

tiagogaspar8, 2022-02-18:
Yeah, I wanted to update my miniupnpd version too to see if this issue is fixed magically (hoping)

If you'd like you can try ipv6 with a he.net free tunnel 😁

tiagogaspar8, 2022-02-18:
So, I updated upnpd to the latest version and it built without issue.
Yet, now it is giving me errors:
```
Fri Feb 18 12:53:16 2022 daemon.notice miniupnpd[9995]: Listening for NAT-PMP/PCP traffic on port 5351
Fri Feb 18 12:53:18 2022 daemon.err miniupnpd[9995]: Failed to remove PCP mapping internal port 31425, protocol TCP
Fri Feb 18 12:53:18 2022 daemon.err miniupnpd[9995]: Failed to remove PCP mapping internal port 31425, protocol TCP
Fri Feb 18 12:53:18 2022 daemon.err miniupnpd[9995]: Failed to remove PCP mapping internal port 31425, protocol UDP
Fri Feb 18 12:53:18 2022 daemon.err miniupnpd[9995]: Failed to remove PCP mapping internal port 31425, protocol UDP
Fri Feb 18 12:53:18 2022 daemon.err miniupnpd[9995]: Failed to remove PCP mapping internal port 31425, protocol TCP
Fri Feb 18 12:53:18 2022 daemon.err miniupnpd[9995]: Failed to remove PCP mapping internal port 31425, protocol UDP
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: send_batch: mnl_cb_run returned -1
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: nft_send_rule(0xb6f59c60, 6, 0) send_batch failed -4
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: PCP MAP: failed to add mapping TCP 31425->IPV6_address 'PCP MAP ec0ac7716fc4d8826bae8213'
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: send_batch: mnl_cb_run returned -1
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: nft_send_rule(0xb6f59c50, 6, 0) send_batch failed -4
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: PCP MAP: failed to add mapping TCP 31425->IPV6_address 'PCP MAP 9cc757b08ef14bd005bbfe9f'
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: send_batch: mnl_cb_run returned -1
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: nft_send_rule(0xb6f59c60, 6, 0) send_batch failed -4
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: PCP MAP: failed to add mapping TCP 31425->IPV6_address 'PCP MAP 3e9a7f85248a0d9649c4c34d'
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: send_batch: mnl_cb_run returned -1
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: nft_send_rule(0xb6f59c50, 6, 0) send_batch failed -4
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: PCP MAP: failed to add mapping UDP 31425->IPV6_address 'PCP MAP e5cb295fb65fe8c9a5534f47'
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: send_batch: mnl_cb_run returned -1
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: nft_send_rule(0xb6f59c60, 6, 0) send_batch failed -4
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: PCP MAP: failed to add mapping UDP 31425->IPV6_address 'PCP MAP fb06d932a8c33d937f52b9f1'
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: send_batch: mnl_cb_run returned -1
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: nft_send_rule(0xb6f59c50, 6, 0) send_batch failed -4
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: PCP MAP: failed to add mapping UDP 31425->IPV6_address 'PCP MAP d305d6a5b481f8b4769d21ac'
```
Just to clarify, these errors are new:
```
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: send_batch: mnl_cb_run returned -1
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: nft_send_rule(0xb6f59c50, 6, 0) send_batch failed -4
Fri Feb 18 12:53:26 2022 daemon.err miniupnpd[9995]: PCP MAP: failed to add mapping UDP 31425->IPV6_address 'PCP MAP d305d6a5b481f8b4769d21ac'
```

And these were present in the previous version:
`Fri Feb 18 12:53:18 2022 daemon.err miniupnpd[9995]: Failed to remove PCP mapping internal port 31425, protocol TCP`

tiagogaspar8, 2022-02-18:
Another update, managed to get miniupnp to work with the 2.3 version, I had to add the following lines to the configuration file:
upnp_table_name=fw4
upnp_nat_table_name=fw4
upnp_forward_chain=forward_wan

That fixed the forwarding and the "new" issues with the miniupnpd adding the rules, yet it didn't fix the failed to remove error.
Now, we can't just add these lines like they are to the config file, because the "forwarding" section must be obtained dynamically from the firewall config.
Also, I'm not sure if there should be a table just for miniupnpd or if it should just hook itself onto the default forwarding tables.

nicefile, 2022-02-25:
Adding valid manual portforward via "http://router/cgi-bin/luci/admin/network/firewall/forwards" make upnpd-nftables starting to actually forward traffic. I learned this by accident now confirmed in this thread 
OpenWrt SNAPSHOT, r18809-5a0975f7ef @mt7621

trippleflux, 2022-03-06:
Does anyone have any progress?

@tiagogaspar8 
> Another update, managed to get miniupnp to work with the 2.3 version, I had to add the following lines to the configuration file:
> 
> ```
> upnp_table_name=fw4
> upnp_nat_table_name=fw4
> upnp_forward_chain=forward_wan
> ```
> 
I assume the configuration file that you were referring is ../files/miniupnpd.init?

stintel, 2022-03-07:
> upnp_table_name=fw4
> upnp_nat_table_name=fw4
> upnp_forward_chain=forward_wan

This is a bad idea, as restarting fw4 will result in all rules added by miniupnpd being removed. One of the advantages that nftables brings is that you do *not* need to use the same table.

trippleflux, 2022-03-08:
My WAN is a PPPOE client, when miniupnpd-nftables bring up somehow my ext_ifname inside /var/etc/miniupnpd.conf is "br-lan", after quick search, found out the following to fixes my ext_ifname & ext_ifname6 :
```
        local conf ifname ifname6

        . /lib/functions/network.sh

        if [ -n "$external_iface" ] ; then
                network_get_device ifname "$external_iface"
                ifname=$(ubus call network.interface.wan status | grep \"l3_device\" | grep -oE '[^:]+$' | grep -o '"[^"]\+"' | sed 's/"//g' 2>/dev/null | head -1)
        else
                if [ -n "$external_zone" ] ; then
                        #ifname=$(fw3 -q zone "$external_zone" 2>/dev/null | head -1)
                        ifname=$(ubus call network.interface.wan status | grep \"l3_device\" | grep -oE '[^:]+$' | grep -o '"[^"]\+"' | sed 's/"//g' 2>/dev/null | head -1)
                else
                        ifname=$(ubus call network.interface.wan status | grep \"l3_device\" | grep -oE '[^:]+$' | grep -o '"[^"]\+"' | sed 's/"//g' 2>/dev/null | head -1)
                        network_find_wan external_iface && \
                                network_get_device ifname "$external_iface"
                fi
        fi
        if [ -n "$external_iface6" ] ; then
                network_get_device ifname6 "$external_iface6"
        else
                if [ -n "$external_zone" ] ; then
                        #ifname6=$(fw3 -q zone "$external_zone" 2>/dev/null | head -1)
                        ifname6=$(ubus call network.interface.wan status | grep \"l3_device\" | grep -oE '[^:]+$' | grep -o '"[^"]\+"' | sed 's/"//g' 2>/dev/null | head -1)

                else
                        ifname6=$(ubus call network.interface.wan status | grep \"l3_device\" | grep -oE '[^:]+$' | grep -o '"[^"]\+"' | sed 's/"//g' 2>/dev/null | head -1)
                        network_find_wan6 external_iface6 && \
                                network_get_device ifname6 "$external_iface6"
                fi
        fi
```

Environments : 
```
OpenWrt SNAPSHOT, r19053-921392e216 on x86-64, NFTABLES
```

ptpt52, 2022-03-11:
to follow this thread

escape0707, 2022-03-13:
> to follow this thread

Just click subscribe button, no need to add a comment here that everyone will get an email notification of.

tiagogaspar8, 2022-03-18:
> I assume the configuration file that you were referring is ../files/miniupnpd.init?

Nop, I actually meant I created another config file manually and started miniupnpd manually.

> This is a bad idea, as restarting fw4 will result in all rules added by miniupnpd being removed. One of the advantages that nftables brings is that you do not need to use the same table.

True, that's an issue I thought of, but I currently don't see a way of adding a jump to the miniupnpd section before all of the rules that fw4 usually creates. Or is there a way I don't know about?

> Another issue found

I have mentioned this before, I also have no idea of what it is and I'm honestly scared to ask upstream 😅 

Sorry for the delay btw guys.

msylgj, 2022-03-19:
Hello Everyone. I've just modify the miniupnp-nftables package to use nft while create firewall rules. And uploaded it to this repo: https://github.com/msylgj/miniupnpd.git
Seems to be worked.😅 However, would you like to have a test? 
[image]

escape0707, 2022-03-20:
@msylgj It will be better for you to fork from the miniupnpd project and then commit your patches, rather than download a snapshot to you local drive then start a git repo afresh.

msylgj, 2022-03-20:
> @msylgj It will be better for you to fork from the miniupnpd project and then commit your patches, rather than download a snapshot to you local drive then start a git repo afresh.

Thank you. I know that and the repo is just for test before a PR

tiagogaspar8, 2022-03-21:
@msylgj I gotta say you're close to the solution! Indeed port forwardings work! Yet when qbittorrent tries to remove them this happens:
```
Mon Mar 21 21:46:10 2022 daemon.err miniupnpd[6032]: Failed to remove PCP mapping internal port 31425, protocol TCP
Mon Mar 21 21:46:10 2022 daemon.err miniupnpd[6032]: Failed to remove PCP mapping internal port 31425, protocol UDP
Mon Mar 21 21:46:10 2022 daemon.err miniupnpd[6032]: Failed to remove PCP mapping internal port 31425, protocol TCP
Mon Mar 21 21:46:10 2022 daemon.err miniupnpd[6032]: Failed to remove PCP mapping internal port 31425, protocol UDP
Mon Mar 21 21:46:10 2022 daemon.err miniupnpd[6032]: Failed to remove PCP mapping internal port 31425, protocol TCP
Mon Mar 21 21:46:10 2022 daemon.err miniupnpd[6032]: Failed to remove PCP mapping internal port 31425, protocol UDP
```
And, when a reboot is performed on the firewall tis happens:
```
root@router1:~# /etc/init.d/firewall restart
Section @rule[9] (Support-UDP-Traceroute) is disabled, ignoring section
Section @rule[10] (31425) is disabled, ignoring section
Hardware flow offloading unavailable, falling back to software offloading
/proc/self/fd/0:51:45-55: Error: Could not process rule: Not supported

/proc/self/fd/0:150:45-57: Error: Could not process rule: Not supported

/proc/self/fd/0:156:46-58: Error: Could not process rule: Not supported
```
And firewall4 doesn't start...

Yet, you're closer than ever, do you have any idea of what this can be?

msylgj, 2022-03-22:
Thanks for your feedback.@tiagogaspar8 
I also see those 'Failed to remove PCP mapping' errors, and duplicate rules will be add.
For the second,  if use 'fw4 reload' replace of '/etc/init.d/firewall restart'(after restart miniupnpd), everything is ok.
However, I push a new commit to my test repo, no hook on fw4, so the second error will be gone. But it still far away from really work, Because there has been a big problem like below, all the time. I'm trying to read the source code of miniupnpd, and find the reason.
```
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: level=0 type=8
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: ifindex = 5  192.168.2.1
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: ST: upnp:rootdevice (ver=0)
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: SSDP M-SEARCH from 192.168.2.3:1900 ST: upnp:rootdevice
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: Single search found
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: SendSSDPResponse(): 0 bytes to 192.168.2.3:1900 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: upnp:rootdevice
 USN: uuid:9b1e2954-07c8-4697-ac7f-8b24e082f39a::upnp:rootdevice
 EXT:
 SERVER: OpenWrt/5.15.30 UPnP/1.1 MiniUPnPd/2.3.0
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1647922782
 BOOTID.UPNP.ORG: 1647922782
 CONFIGID.UPNP.ORG: 1337
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: PCP request received from 192.168.2.3:36261 60bytes
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: UPnP permission rule 0 matched : port mapping accepted
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Check protocol tcp for port 6881 on ext_if pppoe-wan <My Public WAN IP>, 0B701424
Tue Mar 22 12:22:03 2022 daemon.err miniupnpd[24500]: PCP MAP: failed to add mapping TCP 6881->192.168.2.3:6881 'PCP MAP c1217244571b6e5699a108ef'
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: PCP request received from 192.168.2.3:36261 60bytes
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: UPnP permission rule 0 matched : port mapping accepted
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Check protocol udp for port 6881 on ext_if pppoe-wan <My Public WAN IP>, 0B701424
Tue Mar 22 12:22:03 2022 daemon.err miniupnpd[24500]: PCP MAP: failed to add mapping UDP 6881->192.168.2.3:6881 'PCP MAP d92ad01048919d21eded711c'
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: PCP request received from 192.168.2.3:36261 60bytes
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: UPnP permission rule 0 matched : port mapping accepted
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Check protocol tcp for port 8083 on ext_if pppoe-wan <My Public WAN IP>, 0B701424
Tue Mar 22 12:22:03 2022 daemon.err miniupnpd[24500]: PCP MAP: failed to add mapping TCP 8083->192.168.2.3:8083 'PCP MAP a4b5478e687e11137efe4662'
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: HTTP REQUEST from 192.168.2.3:34810 : GET /rootDesc.xml (HTTP/1.1)
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Host: 192.168.2.1:5000
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: HTTP REQUEST from 192.168.2.3:34812 : POST /ctl/IPConn (HTTP/1.1)
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Host: 192.168.2.1:5000
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:1#GetExternalIPAddress
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: HTTP REQUEST from 192.168.2.3:35302 : POST /ctl/IPConn (HTTP/1.1)
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Host: 192.168.2.1:5000
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:1#AddPortMapping
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: AddPortMapping: ext port 6881 to 192.168.2.3:6881 protocol TCP for: qBittorrent Enhanced/4.3.9.10 leaseduration=604800 rhost=
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: UPnP permission rule 0 matched : port mapping accepted
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Check protocol tcp for port 6881 on ext_if pppoe-wan <My Public WAN IP>, 0B701424
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: redirecting port 6881 to 192.168.2.3:6881 protocol TCP for: qBittorrent Enhanced/4.3.9.10
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: Returning UPnPError 501: ActionFailed
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: HTTP REQUEST from 192.168.2.3:55095 : POST /ctl/IPConn (HTTP/1.1)
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Host: 192.168.2.1:5000
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:1#AddPortMapping
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: AddPortMapping: ext port 45368 to 192.168.2.3:6881 protocol TCP for: qBittorrent Enhanced/4.3.9.10 leaseduration=604800 rhost=
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: UPnP permission rule 0 matched : port mapping accepted
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Check protocol tcp for port 45368 on ext_if pppoe-wan <My Public WAN IP>, 0B701424
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: redirecting port 45368 to 192.168.2.3:6881 protocol TCP for: qBittorrent Enhanced/4.3.9.10
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: Returning UPnPError 501: ActionFailed
[... cut ...]
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: redirecting port 46971 to 192.168.2.3:6881 protocol UDP for: qBittorrent Enhanced/4.3.9.10
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: Returning UPnPError 501: ActionFailed
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: HTTP REQUEST from 192.168.2.3:60035 : POST /ctl/IPConn (HTTP/1.1)
Tue Mar 22 12:22:03 2022 daemon.debug miniupnpd[24500]: Host: 192.168.2.1:5000
Tue Mar 22 12:22:03 2022 daemon.info miniupnpd[24500]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:1#AddPortMapping
```

tiagogaspar8, 2022-03-22:
@msylgj OK, if you find anything and need some testing tell me!
BTW, I have disabled upnp, I only use PCP because I only have IPv6 with NAT64 in my network.

stintel, 2022-03-22:
For this bug, please try https://git.openwrt.org/38423fae4ba0
For the miniupnpd error, please report them upstream, or update https://github.com/miniupnp/miniupnp/issues/582 and optionally create a new issue for those. And I would prefer not to have to resort to using custom scripts at all for adding nftables rules. The less custom code we have to maintain, the better.

not-the-nrc, 2022-04-22:
> For this bug, please try https://git.openwrt.org/38423fae4ba0 

@stintel I'm running the latest miniupnpd-nftables_2.3.0-1.  Without your patch, my Xbox Series X reports that my NAT type is moderate (Your network is behind a UPnP port-restricted NAT).

With your patch, my Xbox reports that my NAT type is open (Your network is behind a cone NAT).

neilsan1366, 2022-04-22:
hi how apply the patch on belkin rt3200 please is complicate you think ? thanks

not-the-nrc, 2022-04-23:
> hi how apply the patch on belkin rt3200 please is complicate you think ? thanks

@neilsan1366 If you compile OpenWrt yourself, you can download the patch above at https://git.openwrt.org/?p=project/firewall4.git;a=patch;h=38423fae4ba0f116ae7b5853b1c459202fe2c9a4 and place it in package/network/config/firewall4/patches.

If you don't compile OpenWrt yourself, install a text editor on your router, like nano.  Then, on your router, edit the file /usr/share/firewall4/templates/ruleset.uc and delete the 4 lines deleted by the patch.  Reboot your router for the changes to take effect.  You will lose these changes every time you flash a new version of OpenWrt on your router, so you'll have to modify the file each time.

Please note that they recently reverted the latest changes to miniupnpd and downgraded it to the previous version.  I'm currently waiting for the build to finish to see if the results are the same.

neilsan1366, 2022-04-23:
Yes thanks for the response i will replace the files with winscp 👍

nicefile, 2022-05-03:
@not-the-nrc I can also confirm this patch open ports without adding any manual port forward . This is easy fix for  already  flashed devices .  Tested witu build r18809  @dir-853-r1

neilsan1366, 2022-05-10:
Yes the patch work 👌 thanks for all

jckefan, 2022-07-13:
> For this bug, please try https://git.openwrt.org/38423fae4ba0 For the miniupnpd error, please report them upstream, or update [miniupnp/miniupnp#582](https://github.com/miniupnp/miniupnp/issues/582) and optionally create a new issue here to track those. And I would prefer not to have to resort to using custom scripts at all for adding nftables rules. The less custom code we have to maintain, the better.

Also fixed my UPnP issue thanks.

firstred, 2022-08-06:
> For this bug, please try https://git.openwrt.org/38423fae4ba0 For the miniupnpd error, please report them upstream, or update [miniupnp/miniupnp#582](https://github.com/miniupnp/miniupnp/issues/582) [...]

I can confirm that the patch works, although I had to update the line numbers for 22.03.0-rc6 (patch placed in `package/network/config/firewall4/patches/` folder of OpenWrt src):

```patch
From 38423fae4ba0f116ae7b5853b1c459202fe2c9a4 Mon Sep 17 00:00:00 2001
From: Stijn Tintel <stijn@linux-ipv6.be>
Date: Tue, 22 Mar 2022 21:20:40 +0200
Subject: [PATCH] test: unconditionally allow ct status dnat

---
 root/usr/share/firewall4/templates/ruleset.uc | 4 ----
 1 file changed, 4 deletions(-)

diff --git a/root/usr/share/firewall4/templates/ruleset.uc b/root/usr/share/firewall4/templates/ruleset.uc
index b402315..c7121aa 100644
--- a/root/usr/share/firewall4/templates/ruleset.uc
+++ b/root/usr/share/firewall4/templates/ruleset.uc
@@ -214,9 +214,7 @@
 {%  for (let rule in fw4.rules(`input_${zone.name}`)): %}
 		{%+ include("rule.uc", { fw4, rule }) %}
 {%  endfor %}
-{%  if (zone.dflags.dnat): %}
 		ct status dnat accept comment "!fw4: Accept port redirections"
-{%  endif %}
 {%  fw4.includes('chain-append', `input_${zone.name}`) %}
 		jump {{ zone.input }}_from_{{ zone.name }}
 	}
@@ -235,9 +233,7 @@
 {%  for (let rule in fw4.rules(`forward_${zone.name}`)): %}
 		{%+ include("rule.uc", { fw4, rule }) %}
 {%  endfor %}
-{%  if (zone.dflags.dnat): %}
 		ct status dnat accept comment "!fw4: Accept port forwards"
-{%  endif %}
 {%  fw4.includes('chain-append', `forward_${zone.name}`) %}
 		jump {{ zone.forward }}_to_{{ zone.name }}
 	}
-- 
2.30.2

```

mastum, 2022-08-06:
Thanks, also for me the path has solved the problem.

firewire10000, 2022-08-08:
I've just compiled 22.03.0-rc6 with miniupnpd 2.3.0 (modified the Makefile to download the latest version) and applied the patch but I'm still not getting any open ports nor anything show under the 'Active UPnP Redirects' table.

edrikk, 2022-08-08:
> ... but I'm still not getting any open ports nor anything show under the 'Active UPnP Redirects' table.

Yes.  As I've noted in [Issue 6578](https://github.com/openwrt/luci/issues/5678), both the Luci status -> overview page’s “Active UPnP Redirects” as well as luci-app-upnp package’s “Active UPnP Redirects” sections are empty.

I believe that these two issues are going to cause confusion if 22.03 is released without these 2 GUI fixes.

axet, 2022-08-11:
https://github.com/miniupnp/miniupnp/commit/b8d66c5f7c7cad61241eca5332fda49b242de664

rules suppose to be deleted

tiagogaspar8, 2022-08-24:
Hi guys, can you please give this a test? it should fix and improve the functionality of miniupnpd.
https://github.com/openwrt/packages/pull/19154

not-the-nrc, 2022-08-25:
> Hi guys, can you please give this a test? it should fix and improve the functionality of miniupnpd. #19154

@tiagogaspar8 I removed the patch above from @stintel and applied your pull request.  I'm now running `miniupnpd-nftables_2022-08-06-fa42d8f9` on `r20381-fe86b2ffaa`.  After booting my Xbox, I can see the redirect in `/var/run/miniupnpd.leases` and when running `nft list ruleset`, and my Xbox reports that my NAT type is open ("Your network is behind a cone NAT").

tiagogaspar8, 2022-08-25:
That's good right @not-the-nrc ?
Please test it out so we can close this issue, as this commit is merged and should fix all issues

not-the-nrc, 2022-08-25:
> That's good right @not-the-nrc ? Please test it out so we can close this issue, as this commit is merged and should fix all issues

Yes, that is good.  Your patch works for me and my Xbox.

[19 more replies]

closed 2023-03-10 by stintel