openwrt/packages #17875: ddns-scripts: Simultaneous update of A and AAAA records - Broken IPv6 support on dyn.com

lepeuvedic, 2022-02-16:
Maintainer: @\svlobanov
Environment: Linux OpenWrt 4.14.221 #0 SMP Mon Feb 15 15:22:37 2021 armv7l GNU/Linux
                      OpenWrt 19.07.7 r11306-c4a6851c72
                      ddns-scripts 2.7.8-14
Description:

The script dynamic_dns_updater.sh does not support updating both the A records, and the AAAA record.
With at last dyn.com, the two updates cannot be made independently. 
If myip=nnn.nnn.nnn.nnn, the A record is updated and the AAAA record is deleted.
If myip=2011:.....:abcd, the AAAA record is updated and the A record is deleted.

The correct URL syntax required to update both records is: myip=IPv4,IPv6
Alternately, it seems that repeating the myip parameter works too, but there will be only one response good/nochg using the format IPv4,IPv6

It is not currently possible to update both records on dyn.com, and probably a number of other. The need to update only the AAAA record is probably very rare, and out of that rare situation, IPv6 support on dyn.com is broken, since it deletes the A record.

Suggested enhancement:

The updater script currently has the option use_ipv6 (Luci radiobuttons IPv4-Address and IPv6-Address) which controls which address it attempts to get using nslookup on the defined Lookup hostname (lookup_host). If a third option Both-Addresses was defined, it would be rather simple to get them from the output of nslookup:
```
root@OpenWrt:/usr/local/bin# nslookup critical-optimisation.com
Server:		127.0.0.1
Address:	127.0.0.1#53

Name:      critical-optimisation.com
Address 1: 176.137.81.126
Address 2: 2001:861:3205:7e13::905e
```
As far as I could see, if a custom script also returns  both addresses, the remaining logics should be fine. There could be an issue with expansion of the IPv6, which takes place before the addresses are compared, but the custom script could take care of that as well.

In general a custom script to read the "local" IPs will be required for IPv6, because the ISP often only allocates a prefix. In my case, the DNS AAAA record points directly to a server in the LAN, placed behind the firewall. There is usually no NAT with IPv6. 

My custom script is:
```
#!/bin/sh 

CONFIG=/etc/get_ipv6.conf
DEFAULTS=/etc/get_ipv6.conf.defaults

config_read_file() {
    (grep -E "^${2}=" -m 1 "${1}" 2>/dev/null || echo "VAR=__UNDEFINED__") | head -n 1 | cut -d '=' -f 2-;
}

config_get() {
    val="$(config_read_file "${CONFIG}" "${1}")";
    if [ "${val}" = "__UNDEFINED__" ]; then
        val="$(config_read_file "${DEFAULTS}" "${1}")";
    fi
    printf -- "%s" "${val}";
}

main() {
    interface=$(config_get wan)
    interface6=$(config_get wan6)
    server=$(config_get server)
    if [ "${server}" = "__UNDEFINED__" ]; then
	logger "Configure target LAN server in /etc/get_ipv6.conf"
	exit 1
    fi
    #echo ${server}
    address=$(ifstatus ${interface} | jsonfilter -e '@["ipv4-address"][0].address')
    prefix=$(ifstatus ${interface6} | jsonfilter -e '@["ipv6-prefix"][0].assigned["lan"].address')
    #echo ${prefix}
    ip=$(nslookup ${server} | grep ${prefix} | cut -d " " -f 3)
    echo ${address},${ip}
}
 
main
```
The script reads the prefix, then uses it to filter the correct public IPv6 from the list returned by nslookup for the parameterized ${server}. The IPv4 address is read normally on the wan interface.

I am not sure that I would be able to devise a fix, and my suggestion might not work.

SebTM, 2022-03-12:
Same issue when using INWX.de for DynDNS

renne, 2024-05-05:
Still not working in current OpenWRT/ddns-scripts versions (INWX in my case).

On my Debian machines I run this script every minute:

```bash
#!/bin/bash

# Include config file
[ -f "${1}" ] || {
  echo -e "\n\tThe configuration file ${1} does not exist!\n"
  exit 1
}

[ -r "${1}" ] || {
  echo -e "\n\tThe configuration file ${1} is not readable!\n"
  exit 2
}

. "${1}"

# Check if variables are set in configuration file
for VARIABLE in UPDATE_URL HOSTNAME USERNAME PASSWORD; do {
  [ '1' = "${DEBUG}" ]  && echo -e "${VARIABLE}=${!VARIABLE}"
  [ -z "${!VARIABLE}" ] && {
    echo -e "\n\tVariable \"${VARIABLE}\" missing or empty in configuration file ${1}\n"
    exit 3
  }
}; done

# Define commands
CURL="`which curl` --silent --user ${USERNAME}:${PASSWORD}"
DIG="`which dig` +short -t"

# Get Primary nameserver of hostname
PRIMARY_NAMESERVER="`${DIG} SOA ${HOSTNAME}`"
PRIMARY_NAMESERVER="${PRIMARY_NAMESERVER%%. *}"

# Get IP addresses in DNS from primary nameserver
DNS_IPv4="`${DIG} A    ${HOSTNAME} @${PRIMARY_NAMESERVER}`"
DNS_IPv6="`${DIG} AAAA ${HOSTNAME} @${PRIMARY_NAMESERVER}`"

# Get current IP addresses
NET_IPv4="`${DIG} A     -4 myip.opendns.com @resolver1.opendns.com`"
NET_IPv6="`${DIG} AAAA  -6 myip.opendns.com @resolver1.opendns.com`"

[ '1' = "${DEBUG}" ] && echo -e "PRIMARY_NAMESERVER: ${PRIMARY_NAMESERVER}\nDNS IPv4: ${DNS_IPv4}\nNET IPv4: ${NET_IPv4}\nDNS IPv6: ${DNS_IPv6}\nNET IPv6: ${NET_IPv6}"

# Exit if no changes
[ "${DNS_IPv4}" = "${NET_IPv4}" ] && [ "${DNS_IPv6}" = "${NET_IPv6}" ] && exit

declare -a UPDATE_URL_ARRAY=()
[ -n "${NET_IPv4}" ] && UPDATE_URL_ARRAY+=("myip=${NET_IPv4}")
[ -n "${NET_IPv6}" ] && UPDATE_URL_ARRAY+=("myipv6=${NET_IPv6}")

## Exit if no changes
#[ '0' = "${#UPDATE_URL_ARRAY[@]}" ] && exit

UPDATE_URL+="?${UPDATE_URL_ARRAY[*]}"
UPDATE_URL="${UPDATE_URL// /\&}"

if [ '1' = "${DEBUG}" ]
then
  echo -e "${CURL} ${UPDATE_URL}"
else
  ${CURL} "${UPDATE_URL}"
fi
```
It would be great to have some similar solution for OpenWRT.
The `dig` queries to `resolver1.opendns.com` can be replaced with the IP addresses from the interface.

pomeloy, 2025-04-27:
Still an issue.