openwrt/packages #17888: prometheus-node-exporter-lua: snmp6 plugin tries to read nonexistent proc files

stintel, 2022-02-17:
Maintainer: @champtar 
Environment: OpenWrt master r18878-a93dfff10e on qoriq 

Description:

The snmp6 plugin tries to read /proc/net/dev_snmp6/$device even if this file does not exist. Can be reproduced with this config:

```
config device 'eth1_wan3'
        option type 'macvlan'
        option name 'wan3'
        option ifname 'eth1'
        option macaddr '00:11:22:33:44:55'

config interface 'wan3'
        option ifname 'wan3'
        option proto 'dhcp'
        option ip4table '33'
        option auto '0'
```

```
Thu Feb 17 23:34:19 2022 daemon.info prometheus-node-exporter-lua[42: /usr/lib/lua/prometheus-collectors/snmp6.lua:25: bad argument #1 to 'lines' (/proc/net/dev_snmp6/wan3: No such file or directory)
```

champtar, 2024-06-05:
@stintel is this still an issue ? If yes can you share `ls /proc/net/dev_snmp6/` and `ubus call network.device status`

stintel, 2024-08-03:
I've replaced prometheus-node-exporter-lua with prometheus-node-exporter-ucode.

closed 2024-08-03 by stintel