openwrt/packages #17892: <share-mime-info>: linking error

trippleflux, 2022-02-18:
error when linking, because of lintl not found?
```
ninja: Entering directory `/home/username/works/openwrt/build_dir/target-x86_64_glibc/shared-mime-info-2.1/openwrt-build'
[1/5] Linking target src/update-mime-database
FAILED: src/update-mime-database
x86_64-openwrt-linux-gnu-gcc  -o src/update-mime-database src/update-mime-database.p/update-mime-database.c.o -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-10.3.0_glibc/usr/lib -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-10.3.0_glibc/lib -Wl,--as-needed -Wl,--no-undefined -znow -zrelro -Wl,-rpath,/home/username/works/openwrt/staging_dir/target-x86_64_glibc/usr/lib -Wl,-rpath-link,/home/username/works/openwrt/staging_dir/target-x86_64_glibc/usr/lib -Wl,--start-group /home/username/works/openwrt/staging_dir/target-x86_64_glibc/usr/lib/libglib-2.0.so -L/home/username/works/openwrt/staging_dir/target-x86_64_glibc/usr/lib -lintl /home/username/works/openwrt/staging_dir/target-x86_64_glibc/usr/lib/libxml2.so -Wl,--end-group
/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-10.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/10.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: cannot find -lintl
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:56: /home/username/works/openwrt/build_dir/target-x86_64_glibc/shared-mime-info-2.1/.built] Error 1
make[3]: Leaving directory '/home/username/works/openwrt/feeds/packages/utils/shared-mime-info'
time: package/feeds/packages/shared-mime-info/compile#0.30#0.23#0.46
    ERROR: package/feeds/packages/shared-mime-info failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/shared-mime-info/compile] Error 1
make[2]: Leaving directory '/home/username/works/openwrt'
make[1]: *** [package/Makefile:110: /home/username/works/openwrt/staging_dir/target-x86_64_glibc/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:230: world] Error 2
```

flyn-org, 2022-02-18:
Would you kindly provide a little more context? I just built shared-mime-info using a fresh SDK from the snapshot builds, and it worked fine. I built for x86_64.

BKPepe, 2022-02-18:
I see that for Turris OS, we have the same issue. We are using full NLS.

```
Applying ./patches/001-shared-mime-info-2.1-fix-build.patch using plaintext: 
patching file data/meson.build
patching file meson.build

Applying ./patches/002-shared-mime-info-2.1-no-xmlto.patch using plaintext: 
patching file data/meson.build
patching file meson.build
The Meson build system
Version: 0.61.1
Source dir: /mox-master/build/build_dir/target-aarch64_cortex-a53_musl/shared-mime-info-2.1
Build dir: /mox-master/build/build_dir/target-aarch64_cortex-a53_musl/shared-mime-info-2.1/openwrt-build
Build type: cross build
Project name: shared-mime-info
Project version: 2.1
C compiler for the host machine: ccache_cc (gcc 11.2.0 "aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r18880+54-e9e61d76fd) 11.2.0")
C linker for the host machine: ccache_cc ld.bfd 2.37
C compiler for the build machine: ccache gcc (gcc 8.3.0 "gcc (Debian 8.3.0-6) 8.3.0")
C linker for the build machine: gcc ld.bfd 2.31.1
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: aarch64
Host machine cpu: generic
Target machine cpu family: aarch64
Target machine cpu: generic
Program ./xdgmime/src/print-mime-data found: NO
Program ./xdgmime/src/test-mime-data found: NO
Program ./xdgmime/src/test-mime found: NO
../../../../build_dir/target-aarch64_cortex-a53_musl/shared-mime-info-2.1/meson.build:35: WARNING: 
***************************************************************************
*** xdgmime not compiled, test suite cannot run. Check HACKING for info ***
***************************************************************************
    
Checking for function "fdatasync" : YES 
Found pkg-config: /mox-master/build/staging_dir/host/bin/pkg-config (1.7.3)
Run-time dependency libxml-2.0 found: YES 2.9.12
Run-time dependency glib-2.0 found: YES 2.70.0
Run-time dependency gio-2.0 found: YES 2.70.0
Program check_translations.sh found: YES (/mox-master/build/build_dir/target-aarch64_cortex-a53_musl/shared-mime-info-2.1/po/check_translations.sh)
Configuring config.h using configuration
Configuring shared-mime-info.pc using configuration
Build targets in project: 82
NOTICE: Future-deprecated features used:
 * 0.56.0: {'meson.source_root'}

shared-mime-info 2.1

  User defined options
    Cross files : /mox-master/build/build_dir/target-aarch64_cortex-a53_musl/shared-mime-info-2.1/openwrt-cross.txt
    Native files: /mox-master/build/build_dir/target-aarch64_cortex-a53_musl/shared-mime-info-2.1/openwrt-native.txt
    buildtype   : plain

Found ninja-1.10.2 at /mox-master/build/staging_dir/host/bin/ninja
ninja: Entering directory `/mox-master/build/build_dir/target-aarch64_cortex-a53_musl/shared-mime-info-2.1/openwrt-build'
[1/82] Generating po/af/LC_MESSAGES/shared-mime-info-af.mo with a custom command
[2/82] Generating po/ar/LC_MESSAGES/shared-mime-info-ar.mo with a custom command
[3/82] Generating po/as/LC_MESSAGES/shared-mime-info-as.mo with a custom command
[4/82] Generating po/ast/LC_MESSAGES/shared-mime-info-ast.mo with a custom command
[5/82] Generating po/az/LC_MESSAGES/shared-mime-info-az.mo with a custom command
[6/82] Generating po/be/LC_MESSAGES/shared-mime-info-be.mo with a custom command
[7/82] Generating po/be@latin/LC_MESSAGES/shared-mime-info-be@latin.mo with a custom command
[8/82] Generating po/bg/LC_MESSAGES/shared-mime-info-bg.mo with a custom command
[9/82] Generating po/bn_IN/LC_MESSAGES/shared-mime-info-bn_IN.mo with a custom command
[10/82] Generating po/ca/LC_MESSAGES/shared-mime-info-ca.mo with a custom command
[11/82] Generating po/ca@valencia/LC_MESSAGES/shared-mime-info-ca@valencia.mo with a custom command
[12/82] Generating po/cs/LC_MESSAGES/shared-mime-info-cs.mo with a custom command
[13/82] Generating po/cy/LC_MESSAGES/shared-mime-info-cy.mo with a custom command
[14/82] Generating po/da/LC_MESSAGES/shared-mime-info-da.mo with a custom command
[... 48 lines cut ...]
[63/82] Generating po/sr/LC_MESSAGES/shared-mime-info-sr.mo with a custom command
[64/82] Generating po/sr@latin/LC_MESSAGES/shared-mime-info-sr@latin.mo with a custom command
[65/82] Generating po/sv/LC_MESSAGES/shared-mime-info-sv.mo with a custom command
[66/82] Generating po/ta/LC_MESSAGES/shared-mime-info-ta.mo with a custom command
[67/82] Generating po/te/LC_MESSAGES/shared-mime-info-te.mo with a custom command
[68/82] Generating po/th/LC_MESSAGES/shared-mime-info-th.mo with a custom command
[69/82] Generating po/tr/LC_MESSAGES/shared-mime-info-tr.mo with a custom command
[70/82] Generating po/uk/LC_MESSAGES/shared-mime-info-uk.mo with a custom command
[71/82] Generating po/vi/LC_MESSAGES/shared-mime-info-vi.mo with a custom command
[72/82] Generating po/wa/LC_MESSAGES/shared-mime-info-wa.mo with a custom command
[73/82] Generating po/zh_CN/LC_MESSAGES/shared-mime-info-zh_CN.mo with a custom command
[74/82] Generating po/zh_HK/LC_MESSAGES/shared-mime-info-zh_HK.mo with a custom command
[75/82] Generating po/zh_TW/LC_MESSAGES/shared-mime-info-zh_TW.mo with a custom command
[76/82] Compiling C object src/update-mime-database.p/update-mime-database.c.o
[77/82] Linking target src/update-mime-database
FAILED: src/update-mime-database 
ccache_cc  -o src/update-mime-database src/update-mime-database.p/update-mime-database.c.o -L/mox-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/usr/lib -L/mox-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/lib -Wl,--as-needed -Wl,--no-undefined -DPIC -fPIC -spec [... cut ...]
/mox-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: cannot find -lintl
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:56: /mox-master/build/build_dir/target-aarch64_cortex-a53_musl/shared-mime-info-2.1/.built] Error 1
time: package/feeds/packages/shared-mime-info/compile#4.61#0.77#5.61
```

I think I know what could be wrong, I will send PR.

PR #17893 (2022-02-19): shared-mimo-info: fix compilation with BUILD_NLS
Maintainer: @flyn-org 
Compile tested: Turris Omnia, mvebu, OpenWrt master
Run tested: N/A

When you have enabled full Native Language Support, it could not be
compiled due to this error:

ccache_cc  -o src/update-mime-database src/update-mime-database.p/update-mime-database.c.o -L/mox-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/usr/lib -L/mox-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/lib -Wl,--as-needed -Wl,--no-undefined -DPIC -fPIC -specs=/mox-master/build/include/hardened-ld-pie.specs -zn [... cut]
closed 2022-02-19 by BKPepe