openwrt/packages #17905: miniupnpd-iptables: chain MINIUPNPD not found

trippleflux, 2022-02-19:
Environment: x86-64 OpenWrt SNAPSHOT, r18882-48ace62114, miniupnpd 2.2.3-4
Description:

Getting the following while 
```
Sun Feb 20 01:00:44 2022 daemon.notice miniupnpd[12483]: HTTP listening on port 5000
Sun Feb 20 01:00:44 2022 daemon.warn miniupnpd[12483]: no HTTP IPv6 address, disabling IPv6
Sun Feb 20 01:00:44 2022 daemon.notice miniupnpd[12483]: Listening for NAT-PMP/PCP traffic on port 5351
Sun Feb 20 01:01:17 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:17 2022 daemon.err miniupnpd[12483]: Failed to remove PCP mapping internal port 50342, protocol TCP
Sun Feb 20 01:01:17 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:17 2022 daemon.err miniupnpd[12483]: Failed to remove PCP mapping internal port 50342, protocol UDP
Sun Feb 20 01:01:17 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:17 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:17 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:17 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: PCP MAP: failed to add mapping TCP 50342->10.10.8.142:50342 'PCP MAP 7e43988923c43d78d59bd861'
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: PCP MAP: failed to add mapping UDP 50342->10.10.8.142:50342 'PCP MAP 908051ae52b1b6098b510d28'
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: chain MINIUPNPD not found
Sun Feb 20 01:01:26 2022 daemon.err miniupnpd[12483]: addnatrule() : chain MINIUPNPD not found
```

The chain is exist already in the iptables : 
```
root@username:~# iptables -t nat -L -n -v
# Warning: iptables-legacy tables present, use iptables-legacy to see them
Chain PREROUTING (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain OUTPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain POSTROUTING (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain MINIUPNPD (0 references)
 pkts bytes target     prot opt in     out     source               destination

Chain MINIUPNPD-POSTROUTING (0 references)
 pkts bytes target     prot opt in     out     source               destination
```

/etc/config/upnpd
```
root@username:~# cat /etc/config/upnpd

config upnpd 'config'
        option external_iface 'wan'
        option internal_iface 'lan'
        option port '5000'
        option upnp_lease_file '/var/run/miniupnpd.leases'
        option igdv1 '1'
        option download '170000'
        option upload '180000'
        option uuid '48bee918-020a-4efe-8fc1-b6bd0fcbd74f'
        option enabled '1'

config perm_rule
        option action 'allow'
        option ext_ports '1024-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '1024-65535'
        option comment 'Allow high ports'

config perm_rule
        option action 'deny'
        option ext_ports '0-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '0-65535'
        option comment 'Default deny'
```
Config file:
[.config.zip](https://github.com/openwrt/packages/files/8102723/default.config.zip)

wulfy23, 2022-02-20:
let me guess you installed sqm-scripts?

```
fgrep -r iptables /usr/lib/opkg/info/*.control
realpath $(command -v iptables)
```

trippleflux, 2022-02-20:
Yes, but previously self-compiled OpenWRT snapshot is working fine with SQM & SQM-extra being enabled.

**fgrep -r iptables /usr/lib/opkg/info/*.control** :
```
root@username:~# fgrep -r iptables /usr/lib/opkg/info/*.control
/usr/lib/opkg/info/banip.control:Depends: libc, librt, libpthread, jshn, jsonfilter, ip, ipset, iptables, ca-bundle
/usr/lib/opkg/info/fail2ban.control:Depends: libc, librt, libpthread, iptables, python3-light, python3-ctypes, python3-distutils, python3-email, python3-logging, python3-sqlite3, python3-urllib, python3-pkg-resources
/usr/lib/opkg/info/ip6tables.control:Depends: libc, librt, libpthread, kmod-ip6tables, iptables
/usr/lib/opkg/info/ip6tables.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/ip6tables.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-mod-conntrack-extra.control:Package: iptables-mod-conntrack-extra
/usr/lib/opkg/info/iptables-mod-conntrack-extra.control:Depends: libc, librt, libpthread, iptables, kmod-ipt-conntrack-extra, kmod-ipt-raw
/usr/lib/opkg/info/iptables-mod-conntrack-extra.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/iptables-mod-conntrack-extra.control:SourceName: iptables-mod-conntrack-extra
/usr/lib/opkg/info/iptables-mod-conntrack-extra.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-mod-conntrack-extra.control:Description:  Extra iptables extensions for connection tracking.
/usr/lib/opkg/info/iptables-mod-extra.control:Package: iptables-mod-extra
/usr/lib/opkg/info/iptables-mod-extra.control:Depends: libc, librt, libpthread, iptables, kmod-ipt-extra
/usr/lib/opkg/info/iptables-mod-extra.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/iptables-mod-extra.control:SourceName: iptables-mod-extra
/usr/lib/opkg/info/iptables-mod-extra.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-mod-extra.control:Description:  Other extra iptables extensions.
/usr/lib/opkg/info/iptables-mod-filter.control:Package: iptables-mod-filter
/usr/lib/opkg/info/iptables-mod-filter.control:Depends: libc, librt, libpthread, iptables, kmod-ipt-filter
/usr/lib/opkg/info/iptables-mod-filter.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/iptables-mod-filter.control:SourceName: iptables-mod-filter
/usr/lib/opkg/info/iptables-mod-filter.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-mod-filter.control:Description:  iptables extensions for packet content inspection.
/usr/lib/opkg/info/iptables-mod-ipopt.control:Package: iptables-mod-ipopt
/usr/lib/opkg/info/iptables-mod-ipopt.control:Depends: libc, librt, libpthread, iptables, kmod-ipt-ipopt
/usr/lib/opkg/info/iptables-mod-ipopt.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/iptables-mod-ipopt.control:SourceName: iptables-mod-ipopt
/usr/lib/opkg/info/iptables-mod-ipopt.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-mod-ipopt.control:Description:  iptables extensions for matching/changing IP packet options.
/usr/lib/opkg/info/iptables-mod-ipsec.control:Package: iptables-mod-ipsec
/usr/lib/opkg/info/iptables-mod-ipsec.control:Depends: libc, librt, libpthread, iptables, kmod-ipt-ipsec
/usr/lib/opkg/info/iptables-mod-ipsec.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/iptables-mod-ipsec.control:SourceName: iptables-mod-ipsec
/usr/lib/opkg/info/iptables-mod-ipsec.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-mod-ipsec.control:Description:  iptables extensions for matching ipsec traffic.
/usr/lib/opkg/info/iptables-mod-physdev.control:Package: iptables-mod-physdev
/usr/lib/opkg/info/iptables-mod-physdev.control:Depends: libc, librt, libpthread, iptables, kmod-ipt-physdev
/usr/lib/opkg/info/iptables-mod-physdev.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/iptables-mod-physdev.control:SourceName: iptables-mod-physdev
/usr/lib/opkg/info/iptables-mod-physdev.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-mod-physdev.control:Description:  The iptables physdev match.
/usr/lib/opkg/info/iptables-mod-tproxy.control:Package: iptables-mod-tproxy
/usr/lib/opkg/info/iptables-mod-tproxy.control:Depends: libc, librt, libpthread, iptables, kmod-ipt-tproxy
/usr/lib/opkg/info/iptables-mod-tproxy.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/iptables-mod-tproxy.control:SourceName: iptables-mod-tproxy
/usr/lib/opkg/info/iptables-mod-tproxy.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-mod-tproxy.control:Description:  Transparent proxy iptables extensions.
/usr/lib/opkg/info/iptables-nft.control:Package: iptables-nft
/usr/lib/opkg/info/iptables-nft.control:Alternatives: 300:/usr/sbin/iptables:/usr/sbin/xtables-nft-multi, 300:/usr/sbin/iptables-restore:/usr/sbin/xtables-nft-multi, 300:/usr/sbin/iptables-save:/usr/sbin/xtables-nft-multi
/usr/lib/opkg/info/iptables-nft.control:Source: package/network/utils/iptables
/usr/lib/opkg/info/iptables-nft.control:SourceName: iptables-nft
/usr/lib/opkg/info/iptables-nft.control:CPE-ID: cpe:/a:netfilter_core_team:iptables
/usr/lib/opkg/info/iptables-nft.control:Description:  Extra iptables nftables nft binaries.
/usr/lib/opkg/info/iptables-nft.control: iptables-nft
/usr/lib/opkg/info/iptables-nft.control: iptables-nft-restore
/usr/lib/opkg/info/iptables-nft.control: iptables-nft-save
[... cut ...]
/usr/lib/opkg/info/strongswan.control:Depends: libc, librt, libpthread, libpthread, ip, kmod-crypto-authenc, kmod-ipsec, kmod-ipsec4, kmod-ipsec6, kmod-ipt-ipsec, iptables-mod-ipsec
/usr/lib/opkg/info/tc-full.control:Depends: libc, librt, libpthread, kmod-sched-core, libxtables12, tc-mod-iptables, libbpf0
/usr/lib/opkg/info/tc-mod-iptables.control:Package: tc-mod-iptables
/usr/lib/opkg/info/tc-mod-iptables.control:SourceName: tc-mod-iptables
/usr/lib/opkg/info/tc-mod-iptables.control:Description:  Traffic control module - iptables action
```

solidus1983, 2022-03-20:
I made a patch for miniupnpd-iptables that i can confirm works with SQM enabled.

```
From 0eb7ed7791707993ad47ca86791ebf0990482798 Mon Sep 17 00:00:00 2001
From: solidus1983 <8018745+solidus1983@users.noreply.github.com>
Date: Tue, 15 Mar 2022 23:07:47 +0000
Subject: [PATCH] Fixing MiniUPNP Iptables Legacy

Fixing iptables-legacy verison of MiniUPNP to resolve the issue below by symlinking xtables-legacy-multi as both iptables and ip6tables

```
daemon.err miniupnpd[8052]: chain MINIUPNPD not found
daemon.err miniupnpd[8052]: addnatrule() : chain MINIUPNPD not found
```
---
 package/base-files/files/etc/init.d/miniupnp-fix | 10 ++++++++++
 1 file changed, 10 insertions(+)
 create mode 100755 package/base-files/files/etc/init.d/miniupnp-fix

diff --git a/package/base-files/files/etc/init.d/miniupnp-fix b/package/base-files/files/etc/init.d/miniupnp-fix
new file mode 100755
index 000000000000..3c831adb964a
--- /dev/null
+++ b/package/base-files/files/etc/init.d/miniupnp-fix
@@ -0,0 +1,10 @@
+#!/bin/sh /etc/rc.common
+
+USE_PROCD=0
+
+START=20
+STOP=31
+ln -s -f /usr/sbin/xtables-legacy-multi /usr/sbin/iptables
+ln -s -f /usr/sbin/xtables-legacy-multi /usr/sbin/ip6tables
+/etc/init.d/miniupnp restart
+/etc/init.d/sqm restart
+logger "MiniUPNP Fix Applied"
```

solidus1983, 2022-03-22:
Think i found the issue to the problem

https://github.com/openwrt/packages/blob/master/net/miniupnpd/files/firewall3.include

```
IPTABLES="/usr/sbin/iptables"
IP6TABLES="/usr/sbin/ip6tables"
IPTARGS="-w 1"
```

Shouldn't it be 

```
IPTABLES="/usr/sbin/xtables-legacy-multi"
IP6TABLES="/usr/sbin/xtables-legacy-multi"
IPTARGS="-w 1"
```

champtar, 2022-03-22:
There was a bunch of commits to iptables recently: https://github.com/openwrt/openwrt/commits/master/package/network/utils/iptables
What's the latest version you tested with ? can you retry with today's master ?

solidus1983, 2022-03-22:
> There was a bunch of commits to iptables recently: https://github.com/openwrt/openwrt/commits/master/package/network/utils/iptables What's the latest version you tested with ? can you retry with today's master ?

r19170 was the last build i did that was currently flash.

I am going to do a rebuild, just struggling to drop the patch i made before hand that was hacky so i can test the new one.

I did check the PR's and didn't notice anything to do with the firewall3.include file for the issue though.

trippleflux, 2022-03-22:
@solidus1983 
Thanks, it was working with that workaround but there were some problems with miniupnpd-iptables with the latest snapshot before, I haven't have the chances to test miniupnpd-iptables anymore, so I have moved into using @msylgj's fork of miniupnpd-nftables.

solidus1983, 2022-03-23:
@trippleflux i am currently compiling a new build myself with the fix from the patch i made to test.

I have R19177 built ready and R19180 compiling now. R19177 has not patches for miniupnp in it so will be able to test then both to see what happens. If the new patch solves the issue then i will ask for assistance in how to make a PR to openwrt standard.

solidus1983, 2022-03-24:
Right i have managed to solve the issue.

After running make menuconfig and heading into firewall making sure that all iptables-nft parts are set to module fixes the issue. So no patch required. User error.

trippleflux, 2022-03-28:
Closed as requested.

closed 2022-03-28 by trippleflux