openwrt/packages #17912: findutils on OpenWRT 21.02 fails to compile

D43m0n, 2022-02-21:
Maintainer: @bernhard-voelker
Environment: armv7, Netgear R7800, OpenWRT 21.02.1

Description:
I cloned a [repository of OpenWRT 21.02](https://github.com/ACwifidude/openwrt), I then checked out [a branch](https://github.com/ACwifidude/openwrt/tree/openwrt-21.02-nss-qsdk11.0) of the 21.02 stable release. I then updated feeds, installed feeds, applied my .config and made a definitive config:

```
./scripts/feeds update -a && ./scripts/feeds install -a && cp diffconfig .config && make defconfig && ./scripts/getver.sh
```

After that, I run make with a low number and some additional verbosity: `make -j2 V=s`. Compiling starts and after a minute or so it cuts out with the following error:

```
make[7]: Entering directory '/home/dude/openwrt/openwrt/build_dir/host/findutils-4.7.0/find'
gcc -DHAVE_CONFIG_H -I. -I..  -I../gl/lib -I../lib -I../gl/lib -DLOCALEDIR=\"/home/dude/openwrt/openwrt/staging_dir/host/share/locale\" -I/home/dude/openwrt/openwrt/staging_dir/host/include   -O2 -I/home/dude/openwrt/openwrt/staging_dir/host/include  -MT ftsfind.o -MD -MP -MF .deps/ftsfind.Tpo -c -o ftsfind.o ftsfind.c
mv -f .deps/ftsfind.Tpo .deps/ftsfind.Po
gcc -DHAVE_CONFIG_H -I. -I..  -I../gl/lib -I../lib -I../gl/lib -DLOCALEDIR=\"/home/dude/openwrt/openwrt/staging_dir/host/share/locale\" -I/home/dude/openwrt/openwrt/staging_dir/host/include   -O2 -I/home/dude/openwrt/openwrt/staging_dir/host/include  -MT finddata.o -MD -MP -MF .deps/finddata.Tpo -c -o finddata.o finddata.c
mv -f .deps/finddata.Tpo .deps/finddata.Po
gcc -DHAVE_CONFIG_H -I. -I..  -I../gl/lib -I../lib -I../gl/lib -DLOCALEDIR=\"/home/dude/openwrt/openwrt/staging_dir/host/share/locale\" -I/home/dude/openwrt/openwrt/staging_dir/host/include   -O2 -I/home/dude/openwrt/openwrt/staging_dir/host/include  -MT fstype.o -MD -MP -MF .deps/fstype.Tpo -c -o fstype.o fstype.c
mv -f .deps/fstype.Tpo .deps/fstype.Po
gcc -DHAVE_CONFIG_H -I. -I..  -I../gl/lib -I../lib -I../gl/lib -DLOCALEDIR=\"/home/dude/openwrt/openwrt/staging_dir/host/share/locale\" -I/home/dude/openwrt/openwrt/staging_dir/host/include   -O2 -I/home/dude/openwrt/openwrt/staging_dir/host/include  -MT parser.o -MD -MP -MF .deps/parser.Tpo -c -o parser.o parser.c
parser.c: In function 'parse_user':
parser.c:75:20: error: expected expression before ')' token
   75 | # define endpwent ()
      |                    ^
parser.c:2463:7: note: in expansion of macro 'endpwent'
 2463 |       endpwent ();
      |       ^~~~~~~~
make[7]: *** [Makefile:1945: parser.o] Error 1
make[7]: Leaving directory '/home/dude/openwrt/openwrt/build_dir/host/findutils-4.7.0/find'
make[6]: *** [Makefile:2008: all-recursive] Error 1
make[6]: Leaving directory '/home/dude/openwrt/openwrt/build_dir/host/findutils-4.7.0/find'
make[5]: *** [Makefile:2109: all-recursive] Error 1
make[5]: Leaving directory '/home/dude/openwrt/openwrt/build_dir/host/findutils-4.7.0'
make[4]: *** [Makefile:2048: all] Error 2
make[4]: Leaving directory '/home/dude/openwrt/openwrt/build_dir/host/findutils-4.7.0'
make[3]: *** [Makefile:28: /home/dude/openwrt/openwrt/build_dir/host/findutils-4.7.0/.built] Error 2
make[3]: Leaving directory '/home/dude/openwrt/openwrt/tools/findutils'
time: tools/findutils/compile#34.43#5.03#40.68
    ERROR: tools/findutils failed to build.
make[2]: *** [tools/Makefile:159: tools/findutils/compile] Error 1
make[2]: Leaving directory '/home/dude/openwrt/openwrt'
make[1]: *** [tools/Makefile:155: /home/dude/openwrt/openwrt/staging_dir/host/stamp/.tools_compile_yyyyyynnyyynyyyyyynyynnyyyynyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/home/dude/openwrt/openwrt'
make: *** [/home/dude/openwrt/openwrt/include/toplevel.mk:230: world] Error 2
```
I have attached the complete compile log to this issue:
[findutils fails to compile.txt](https://github.com/openwrt/packages/files/8107646/findutils.fails.to.compile.txt)

bernhard-voelker, 2022-02-21:
[adding @neheb ]
First of all, I'm **not** the maintainer here.  Instead, I'm a upstream GNU findutils maintainer.
I don't know anything about building for OpenWRT, and it seems that the build there is quite special.

The problem seems to arise from the fact that the function getpwent() is not defined.
Still, code in parser.c looks a bit odd.  Does the following work for you?

```
diff --git a/find/parser.c b/find/parser.c
index c728fd6b..5b5e52a2 100644
--- a/find/parser.c
+++ b/find/parser.c
@@ -67,12 +67,12 @@
 #include "findutils-version.h"
 #include "system.h"
 
-
-#ifndef HAVE_ENDGRENT
-# define endgrent ()
+#if ! HAVE_ENDGRENT
+# define endgrent() ((void) 0)
 #endif
-#ifndef HAVE_ENDPWENT
-# define endpwent ()
+
+#if ! HAVE_ENDPWENT
+# define endpwent() ((void) 0)
 #endif
 
 static bool parse_accesscheck   (const struct parser_table*, char *argv[], int *arg_ptr);

```

D43m0n, 2022-02-21:
I’m sorry if I tagged you by mistake, should I have tagged @BKPepe instead when creating this issue?

BKPepe, 2022-02-21:
Why me? I am not findutils maintainer. :-)

D43m0n, 2022-02-21:
To be honest; I’m trying to contribute by following a template when creating an issue. The template suggests to find the package maintainer by looking at it’s Makefile.

To get some sort of an authoritative lead on that, I searched for the package `findutils` on OpenWRT’s [package search](https://openwrt.org/packages/table/start?dataflt%5BName_pkg-dependencies*%7E%5D=Findutils). On the result page I see links to the source and to the place where to add bug reports. When I click on the link of [source](https://github.com/openwrt/packages/tree/openwrt-21.02/utils/findutils), I see two contributors and I see the Makefile all in one place, as suggested by the GitHub issue template? That’s how I sort of concluded I found the two maintainers of the “findutils package for OpenWRT”. But if both of you are not the maintainer of this package for OpenWRT, than who is?

neheb, 2022-02-21:
@D43m0n what system is this on? This is for tools/findutils , not the findutils package.

I can whip up a patch based on @bernhard-voelker 's suggestion, but I need more information.

D43m0n, 2022-02-21:
I’m compiling an OpenWRT 21.02 build for a [Netgear R7800](https://openwrt.org/toh/netgear/r7800). This is a device with a Qualcomm Atheros IPQ8065 chipset with a ARMv7 Processor rev 0 (v7l), arm_cortex-a15_neon-vfpv4 architecture.

Is this the kind of information you are looking for? 

Are you the package maintainer for findutils on OpenWRT?

neheb, 2022-02-21:
No and No.
I’m looking for host OS information.

D43m0n, 2022-02-21:
I’m trying to compile this on a Ubuntu 20.04 lts host (x86_64)

BKPepe, 2022-02-21:
You should rather use upstream repositories instead of forked, outdated repository.

D43m0n, 2022-02-21:
The fork I’m referring too has 3 additional branches, the last rebase on the branch I referred to was last rebased 25 days ago, on a stable release of 21.02.1. The 21.02.2 service release (stable) is about to be announced, based on conversations on the OpenWRT forum. I wouldn’t qualify that as outdated?

neheb, 2022-02-21:
@D43m0n try testing @bernhard-voelker 's patch.

D43m0n, 2022-02-22:
I don't know how to apply that patch in this large tree for OpenWRT, for a learning curve it's one of the steepest I can think off.

neheb, 2022-02-22:
In your openwrt tree, run

```
curl https://github.com/neheb/openwrt/commit/1e2eecf6623a5be7d33afc15e849f0d2e06bba12.patch | git apply
```

D43m0n, 2022-02-27:
compile succeeds, thanks!

closed 2022-02-27 by D43m0n