openwrt/packages #17915: mdnsresponder: mdnsd (IETF104-4) segfaults on startup on 21.02.2

maertsen, 2022-02-21:
# Issue template (remove lines from top till here)

Maintainer: @neheb ?
Environment: x86/64, 21.02.2

Description:
I've used the combination of mdnsd and ohybridproxy for multiple years. I recently upgraded from 19.07 to 21.02.2 and now see mdnsd segfaulting on startup.
```
mdnsd[17254]: segfault at 0 ip 000000000040429c sp 00007ffee50f30a0 error 4 in mdnsd[402000+43000]
Code: 1a 4d 06 00 44 8b 20 48 8b 5c 24 08 48 85 db 0f 84 a1 00 00 00 45 85 e4 0f 85 9e 00 00 00 45 31 ff 31 ed 45 31 f6 48 8b 43 18 <66> 8b 00 83 e0 f7 66 83 f8 02 75 54 8b 43 10 83 e0 11 ff c8 75 4a
```
I'm not familiar with debugging of software in the C family of languages, but certainly willing to provide additional information to further narrow this down. I could use some guidance how to proceed though :)

neheb, 2022-02-22:
If I had to guess it broke because of a musl update. I don't use this software so I can't really help.

maertsen, 2022-02-25:
I looked into this. This is a null-pointer dereference while looping over all available interfaces during setup. I happen to have an interface without an assigned address. There is a missing NULL check and this was previously found and fixed in https://github.com/IETF-Hackathon/mDNSResponder/commit/1fb07b9524b4afed3a826c087db4dc48a7bfdb8. That repo appears to be the (newer) upstream for this code, albeit without a formal release.

I can either submit a copy of this patch, or can try to get a latest git version of that repo working as an openwrt package. I would like to get a fix backported into 21.01.x and I'm seeking guidance on how to proceed.

@duckhorn, @StuartCheshire do you have an opinion as upstream?

Abhayakara, 2022-02-25:
The best place to put a PR would be here:

https://github.com/Abhayakara/mdnsresponder

I think this is a known issue, and may even be fixed there, but it doesn't hurt to submit a pull request if the fix isn't present in the repo. We don't have a formal process for keeping this repo in sync with our internal repo, so the fix for the issue might well not be in the above repo (or I might be wrong that it's fixed).

maertsen, 2022-02-25:
thanks @Abhayakara, it is indeed fixed in that repo. And it appears https://github.com/Abhayakara/mdnsresponder has much more recent work and should be the correct upstream for this package in openwrt in master. Would the https://github.com/Abhayakara/mdnsresponder/releases/tag/20201119-release release make sense?

The open question that remains for my personal use-case is how to get the fix for this particular nullpointer dereference into openwrt's stable 21.01.x branch.

Abhayakara, 2022-02-25:
The release tags are not particularly major, so actually ToT might be better. I can put a tag on it if that makes it easier to use as an upstream.

PR #17943 (2022-03-04): mdnsresponder: Fix nullpointer dereference while parsing interface list
This patch was backported from https://github.com/IETF-Hackathon/mDNSResponder/commit/1fb07b9524b4afed3a826c087db4dc48a7bfdb8

Fixes: #17915

Maintainer: unknown, @neheb would you willing to merge?
Compile tested: x86-64
Run tested: apu2, x86-64
closed 2022-03-04 by neheb