openwrt/packages #17922: xz dosen't have xzcat only have xz

leonpano2006, 2022-02-22:
E: Extracting .//var/cache/apt/archives/base-files_11ubuntu5_arm64.deb requires the xzcat command, which is not available
MastermixStudios@GT-AX11000-A2B0:/mnt/JFFS# opkg install xz
Package xz (5.2.5-3) installed in root is up to date.
MastermixStudios@GT-AX11000-A2B0:/mnt/JFFS# xz
xz: Compressed data cannot be written to a terminal
xz: Try `xz --help' for more information.
MastermixStudios@GT-AX11000-A2B0:/mnt/JFFS# xzcat

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
E: Extracting .//var/cache/apt/archives/base-files_11ubuntu5_arm64.deb requires the xzcat command, which is not available
MastermixStudios@GT-AX11000-A2B0:/mnt/JFFS# opkg install xz
Package xz (5.2.5-3) installed in root is up to date.
MastermixStudios@GT-AX11000-A2B0:/mnt/JFFS# xz
xz: Compressed data cannot be written to a terminal
xz: Try `xz --help' for more information.
MastermixStudios@GT-AX11000-A2B0:/mnt/JFFS# xzcat
bash: xzcat: command not found
MastermixStudios@GT-AX11000-A2B0:/mnt/JFFS# ls /opt/bin
addr2line         egrep             install-debian    mktemp            rbash             strip
ar                elfedit           ld                mount             readelf           umount
as                fgrep             ld.bfd            mountpoint        readlink          uname
ash               find              less              netstat           rm                who
bash              findmnt           locale.new        nm                sh                whoami
binutils-strings  gprof             localedef.new     objcopy           sha1sum           xargs
c++filt           grep              mc                objdump           size              xz
chroot            id                mcdiff            opkg              start-debian      zsh
cp                install           mcedit            ranlib            stress
MastermixStudios@GT-AX11000-A2B0:/mnt/JFFS# opkg install xz-utils```
![Screenshot 2022-02-22 183701](https://user-images.githubusercontent.com/58368166/155187316-02f9d850-bfc3-4ded-bf48-7edcb78132cf.png)
fixed by add syslink but that should done by package

BKPepe, 2022-03-06:
@leonpano2006 What are you trying to do? I see that you are running unknown OpenWrt version and you are trying to something with ``.deb`` package. 

I see that xzcat is compiled as subpackage:
https://github.com/openwrt/packages/blob/3e75dc582e1cc68da20c42f90160c45a1b38cc20/utils/xz/Makefile#L119

For me, it is not clear if you installed xz-utils or not.

Please next time, try to fill at least the issue template.

As everything was answered, closing.

closed 2022-03-06 by BKPepe

leonpano2006, 2022-03-07:
not openwrt 
is asuswrt
and deb just example how it should be

leonpano2006, 2022-03-07:
> @leonpano2006 What are you trying to do? I see that you are running unknown OpenWrt version and you are trying to something with `.deb` package.
>
> I see that xzcat is compiled as subpackage: [...]
is not openwrt 
is asuswrt
and that .deb is for debootstrap to make debian chroot