openwrt/packages #17923: collectd rrd files lost with reboot

emdete, 2022-02-22:
# Issue template

Maintainer: @\<jow->
Environment: arch Atheros AR7161 rev 2, model Netgear WNDR3700, OpenWrt version 21.02.1

Description:

In `/etc/collectd.conf` the "DataDir" is set to "/tmp/rrd" which is a ramdisk and the database is lost on a reboot. I think this is not desired.

So i set that setting to "/var/lib/collectd" (after creating that directory) and restarted the service.

The path must be adjusted as well in "/etc/config/luci_statistics" under "DataDir".

jow-, 2022-02-22:
This is intentional to avoid flash wear.

closed 2022-02-22 by jow-

emdete, 2022-02-23:
how about copying it back&forth on start/stop of the service? would that be a solution (at least for ordinary reboots)?

jow-, 2022-02-23:
Many devices do not have enough overlay space available to store the RRD archives. There really is no suitable global default, you do need to adjust the configuration to your needs.