openwrt/packages #17927: unbound: /var/lib/unbound/unbound.conf differs from /etc/config/unbound

it-jonas, 2022-02-23:
Maintainer: @EricLuehrsen
Environment: AMD GX-412TC SOC  OpenWrt 21.02.0, r16279-5cc0535800

Description:
The content of the config which was set within luci differs from the generated config. 
[image]
[image]

brada4, 2022-02-24:
I assume it maintains focus on embedded systems, multi-thread option is there for testing its stability while most systems will be saturated by default anyway.
https://github.com/openwrt/packages/blob/94858ece3120a866da97e0b891f313fd032c7bc6/net/unbound/files/unbound.sh#L706
You can edit that uci script and measure throughput (cached), it needs no iptables and multiqueue netcard to jump above 10k rps on normal systems.

it-jonas, 2022-03-20:
thank you for your answer. my report was about the options not matching when i set them. 
are you telling me that setting this option is not implemented at all??

brada4, 2022-03-20:
Script limits unbound to TWO threads no matter what you have/set. You can get more by editing that script on your system. 
and two more likely bottlenecks before measuring multiprocessor performance:
* multiqueue netcard
* iptables --ports 53 -j NOTRACK

EricLuehrsen, 2022-04-01:
`num_threads` can have caveats on different hardware and requires configuring memory slabs correctly. It can be complicated. One slab and two threads works well on MVEBU for example. UCI is limited to reliable and easy configuration. If something more is desired, then it would be better to use the manual configuration mode.

it-jonas, 2022-06-10:
Thanks for you replies. Seems like I need to dig way more deeper more into the documentation.

closed 2022-06-10 by it-jonas