openwrt/packages #17939: netifd: Use uqmi timeout option (-t) in qmi.sh

alpha197, 2022-02-25:
Description:
When devices with Quectel LTE modules loose connectivity (e.g. due to weak LTE cell connectivity) we observe that reconnects fail because an uqmi  process (triggered by netifd/qmi.sh) stucks. This can be tracked down to some kind of "stall" mode of Quectel LTE modems as already described [here](https://git.openwrt.org/?p=project/uqmi.git;a=commit;h=0a19b5b77140465c29e2afa7d611fe93abc9672f): Requests issued by uqmi hang forever and therefore connects/reconnects triggered by netifd via qmi.sh will fail and we have no LTE connection anymore. 

Solution:
Since 2020 uqmi has an timeout parameter to handle this problem, see [commit 0a19b5b77140465c29e2afa7d611fe93abc9672f](https://git.openwrt.org/?p=project/uqmi.git;a=commit;h=0a19b5b77140465c29e2afa7d611fe93abc9672f). Unfortunately qmi.sh does not make use of this parameter. As described in the above patch, qmi.sh should invoke an early dummy access to "unlock" the modem.

This can be simply archieved by inserting the following line in qmi.sh as line 85
```
uqmi -d "$device" --get-pin-status -t 3000 >/dev/null 2>&1
```
Another solution might be to add the "-t" parameter to each uqmi command in qmi.sh.

dr-ni, 2023-08-21:
This fix in /lib/netifd/proto/qmi.sh was helpfull for my SIMCOM SIM7600G modem:

[qmi.sh.txt](https://github.com/openwrt/packages/files/12397007/qmi.sh.txt)

dr-ni, 2023-08-21:
see also https://github.com/openwrt/openwrt/pull/13324