openwrt/packages #17964: shorewall6: recursive dependency errors

ldir-EDB0, 2022-03-01:
make defconfig is reporting recursive dependency errors:

```
tmp/.config-package.in:103626:error: recursive dependency detected!
tmp/.config-package.in:103626:	symbol PACKAGE_shorewall6 depends on PACKAGE_shorewall6
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:103662:error: recursive dependency detected!
tmp/.config-package.in:103662:	symbol PACKAGE_shorewall6-lite depends on PACKAGE_shorewall6-lite
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

WAN-Noob, 2022-03-01:
Can confirm. Started happening today. Getting the same error.

```
`tmp/.config-package.in:103584:error: recursive dependency detected!
tmp/.config-package.in:103584:	symbol PACKAGE_shorewall6 depends on PACKAGE_shorewall6
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:103620:error: recursive dependency detected!
tmp/.config-package.in:103620:	symbol PACKAGE_shorewall6-lite depends on PACKAGE_shorewall6-lite
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"`
```

test012012, 2022-03-01:
Sorry, I can't investigate this issue in a short time. Please revert my change

@zhanhb @stintel

svlobanov, 2022-03-01:
Sorry, I can't investigate this issue in a short time. Please revert my change

@zhanhb @stintel

justinkb, 2022-03-03:
oddly enough, reverting those recent commits from @svlobanov doesn't get rid of this issue for me

justinkb, 2022-03-03:
@champtar tagging you

champtar, 2022-03-03:
Thanks, will try to have a look

justinkb, 2022-03-03:
> Thanks, will try to have a look

https://github.com/openwrt/packages/blob/master/net/shorewall6/Makefile#L40 start here I suppose?

champtar, 2022-03-07:
https://github.com/openwrt/openwrt/issues/9407

closed 2022-06-26 by ldir-EDB0