openwrt/packages #17971: auc :  Segmentation fault when trying to update Linksys WRT1200 AC

notthinkingtinker, 2022-03-02:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @aparcar  (find it by checking history of the package Makefile)
Environment: POpenwrt 21.02.1

Description:
root@Openwrt~# auc
auc (0.2.4-4)
Server:    chef.libremesh.org
Running:   21.02.1 r16325-88151b8303 on mvebu/cortexa9 (linksys,wrt1200ac)
Segmentation fault

notthinkingtinker, 2022-03-13:
@aparcar  Please help with this issue .

dangowrt, 2022-03-13:
I guess you manage to crash it by using only a hostname instead of URL ie. protocol+hostname. Try setting url in `/etc/config/attendedsysupgrade` to `https://sysupgrade.openwrt.org`.

commit c0d2c825 (2022-03-13): auc: don't segfault on invalid URL

Show error message instead of segfaulting in case of an invalid URL
being read from UCI config.

Fixes: #17971
commit 793e7ee4 (2022-03-14): auc: don't segfault on invalid URL

Show error message instead of segfaulting in case of an invalid URL
being read from UCI config.

Fixes: #17971

(cherry picked from commit c0d2c82528e19a304164dade96e9b019114b8fb0)
closed 2022-03-13 by dangowrt