openwrt/packages #17978: apfree-wifidog requires libiptext library from iptables

BKPepe, 2022-03-03:
Maintainer: @liudf0716
Environment: Turris Omnia, mvebu/cortex-a9, OpenWrt master and Turris MOX, mvebu/cortex-a53, OpenWrt master

Description:

```
-- Install configuration: "Release"
-- Installing: /foo/build_dir/target-aarch64_cortex-a53_musl/apfree-wifidog-4.08.1771/ipkg-install/usr/bin/wifidogx
-- Installing: /foo/build_dir/target-aarch64_cortex-a53_musl/apfree-wifidog-4.08.1771/ipkg-install/usr/bin/wdctlx
Package apfree-wifidog is missing dependencies for the following libraries:
libiptext.so
libiptext4.so
```

Same happens on [buildbots](https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/apfree-wifidog/compile.txt):

```
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/apfree-wifidog-4.08.1771/src/wd_client.c: In function 'wd_get_orig_url':
/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/apfree-wifidog-4.08.1771/src/wd_client.c:49:34: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   49 |         struct evhttp_uri *uri = evhttp_request_get_evhttp_uri(req);
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[25/27] Building C object src/CMakeFiles/wifidogx.dir/wd_util.c.o
[26/27] Building C object src/CMakeFiles/wifidogx.dir/mqtt_thread.c.o
[27/27] Linking C executable src/wifidogx
ninja: Entering directory `/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/apfree-wifidog-4.08.1771'
[0/1] Install the project...
-- Install configuration: "Release"
-- Installing: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/apfree-wifidog-4.08.1771/ipkg-install/usr/bin/wifidogx
-- Installing: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/apfree-wifidog-4.08.1771/ipkg-install/usr/bin/wdctlx
Package apfree-wifidog is missing dependencies for the following libraries:
libiptext.so
libiptext4.so
make[3]: *** [Makefile:89: /builder/shared-workdir/build/sdk/bin/packages/aarch64_cortex-a53/packages/apfree-wifidog_4.08.1771-6_aarch64_cortex-a53.ipk] Error 1
time: package/feeds/packages/apfree-wifidog/compile#13.81#2.53#19.49
```

Related to switch to nftables instead of iptables:
https://github.com/openwrt/packages/issues/16818

liudf0716, 2022-04-05:
@BKPepe ok，I will fix it as soon as possible

liudf0716, 2022-04-06:
@BKPepe pls review the PR, thx
https://github.com/openwrt/packages/pull/18244

PolynomialDivision, 2022-06-01:
Issue can be closed?

liudf0716, 2022-06-01:
@PolynomialDivision already fixed

closed 2022-06-01 by PolynomialDivision