openwrt/packages #17979: aria2 generally breaks as of the employment of jail

hsghost, 2022-03-03:
**Maintainer**: @kuoruan
**Environment**: Snapshot@2022-02-27 on _GL.iNet GL-B1300_

### Description: 
The init script of aria2 is basically not working as of the current snapshots. It meant to run this program as a service while it fails under current builds. The log complaints about missing arguments (at least one URL blablalba...). I can see that although there are code in the script assigning parameters as arguments prior to the creation of a jail, it seems ineffective for the most part, except for a very few options like assigning the daemon user, etc. The others like `enable-rpc=true` seems tnever applied, causes the missing URL complaint mentioned above. Moreover, even if I manually add arguments like `--enable-rpc` onto the `procd_append_param` line, making the service to start successfully, some more options like `--rpc-certificate` still complaints about missing files.

I can see that the script is utilizing the `prod-ujail` functionality, and there are lines setting up mounts in order for the program to read configs, etc. So I suspect that these code might still have bugs which causes those mounts to be ineffective. Also, it's obviously missing mounts to the regular certificate paths like `/etc/ssl`, which no wonder made my attempts to add certificate options fail.

Currently I don't have time to look deep into the code, so please help to fix!

Thank you.

Acris, 2022-03-03:
Same as: https://github.com/openwrt/packages/issues/16963

hsghost, 2022-03-05:
Got it fixed. Thanks.

```
251a252,255
>               local rpc_certificate_dir
>               rpc_certificate_dir="$(dirname "$rpc_certificate")"
>               local rpc_private_key_dir
>               rpc_private_key_dir="$(dirname "$rpc_private_key")"
256a261,263
>               ca_certificate=${ca_certificate:-"/etc/ssl/cert.pem"}
>               local ca_certificate_dir
>               ca_certificate_dir="$(dirname "$ca_certificate")"
344,345c351,353
<       procd_add_jail_mount "$config_file"
<       procd_add_jail_mount_rw "$dir" "$config_dir" "$log"
---
>       # procd_add_jail_mount "$config_file"
>       procd_add_jail_mount_rw "$dir" "$config_dir" "$log" \
>               "$rpc_certificate_dir" "$rpc_private_key_dir" "$ca_certificate_dir"
```

closed 2022-03-05 by hsghost

ringtailedfox, 2022-10-09:
thank you!  whatever updates were made to the program between the time i reported this bug and today... whatever changes were made fixed it becuase i didn't even need to edit the /etc/init.d/aria2 file to comment out that rogue line about jail... :D