openwrt/packages #17981: Request Luci package

ne20002, 2022-03-03:
There is no luci package for crowdsec.

I would like to request the following:
* a page where I can enable disable CrowdSec 
* a way to configure the interfaces for the bouncer on which to block (for me only WAN and WAN6 is neccessary) or
* a firewall block rule where I can define the interfaces to be applied on and that I can move in the firewall rules in the correct position

The ipset block shall by after e.g. DHCP but before port forwarding rules or opened ports for exposed services.

hnyman, 2022-03-06:
LuCI is handled in its own repo, so this should be there.

But I do not think that any of the active LuCI developers is using that crowdsec app, so it is mainly upto somebody using the app to craft the LuCI support for it and then issue PR in the LuCI repo.

closed 2022-03-06 by hnyman

ne20002, 2023-11-09:
A LucCI package is now available in PR # . Also a UCI package implementing the crowdsec firewall filter with UCI config.