openwrt/packages #17984: obfs4proxy: Golang strip & upx

rabits, 2022-03-04:
Actually this is pretty generic question, so not sure the package is relevant... Anyway - why golang packages (like obfs4proxy) don't use strip and upx to lower the disk space requirements?

For example what I've done to install obfs4proxy to my router:
```
$ go get gitlab.com/yawning/obfs4
$ cd pkg/mod/gitlab.com/yawning/obfs4\@v0.0.0-20220204003609-77af0cba934d/
$ GOARCH=mips GOMIPS=softfloat go build -ldflags="-s -w" -o obfs4proxy/obfs4proxy ./obfs4proxy
# Get UPX from https://github.com/upx/upx/actions (the latest version 3.96 have a mips bug where the app is running forever)
# Unpack UPX and run
$ ./upx obfs4proxy/obfs4proxy
$ ls -lh obfs4proxy/obfs4proxy
-rwxr-xr-x 1 root root 1.8M Mar  4 05:52 obfs4proxy/obfs4proxy
```

The original obfs4proxy from obfs4proxy_0.0.11-1_mips_24kc.ipk is 6.3M.

Thank you

jefferyto, 2022-03-05:
If you have [searched](https://github.com/openwrt/packages/search?q=upx&type=issues) here for "upx" you would have found #8000 - I suggest you read that first.

OpenWrt uses compressed file systems. You need to look at the actual disk space used, not the reported size of the binary.

jefferyto, 2022-03-05:
Also, all binaries built for OpenWrt are stripped by default, unless you manually selected [`CONFIG_NO_STRIP`](https://github.com/openwrt/openwrt/blob/c2d7896a650edd6ff4f37dbb33d45027001c23f4/config/Config-build.in#L169). I suggest you examine the ipk binary for symbols, instead of just looking at file size.

rabits, 2022-03-05:
Oh I see, thank you for clarification with #8000 and stripping.

In general I was not able to squeeze the original binary to my 2.4 MB of overlay (even if there is compression) - so the only quick option for me was to build it myself (because I had to use the client static port https://github.com/rabits/obfs4/commit/231dd59117caa5289592bd0fe8efd8b1bdb4346d) - and upx really helped to pack it tight and put to the system. Now, with additional openvpn and other small packages I have just 140KB, but at least the system is working properly)

closed 2022-03-13 by BKPepe