openwrt/packages #17991: libreswan: build fails, missing unbound depency

oskarirauta, 2022-03-05:
Maintainer: Lucian Christian / @lucize 
Environment: x86_64, recent snapshot

Description:

libreswan fails to build. unbound library is required by libreswan but there are options for **libunbound**, one of them is being 
`CONFIG_PACKAGE_libunbound_libevent` which must be set **Y** otherwise you will face up following build log and failed libreswan compilation.

```
ccache_cc -DTimeZoneOffset=timezone -pthread -std=gnu99 -g  -Wall -Wextra -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wmissing-declarations -Wredundant-decls -Wnested-externs -O2 -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -fstack-protector-all -fno-strict-aliasing -fPIE -DPIE -DNSS_IPSEC_PROFILE -DXFRM_LIFETIME_DEFAULT=30 -DUSE_IKEv1 -DKERNEL_XFRM -DUSE_XFRM_INTERFACE -DUSE_DNSSEC -DDEFAULT_DNSSEC_ROOTKEY_FILE=\""/var/lib/unbound/root.key"\" -DUSE_3DES -DUSE_AES -DUSE_CAMELLIA -DUSE_CHACHA -DUSE_DH31 -DUSE_MD5 -DUSE_SHA1 -DUSE_SHA2 -DUSE_PRF_AES_XCBC -DUSE_NSS_KDF -DDEFAULT_RUNDIR=\"/var/run/pluto\" -DIPSEC_CONF=\"/etc/ipsec.conf\" -DIPSEC_CONFDDIR=\"/etc/ipsec.d\" -DIPSEC_NSSDIR=\"/etc/ipsec.d\" -DIPSEC_CONFDIR=\"/etc\" -DIPSEC_EXECDIR=\"/usr/libexec/ipsec\" -DIPSEC_SBINDIR=\"/usr/sbin\" -DIPSEC_VARDIR=\"/var\" -DPOLICYGROUPSDIR=\"/etc/ipsec.d/policies\" -DIPSEC_SECRETS_FILE=\"/etc/ipsec.secrets\" -DFORCE_PR_ASSERT -DUSE_FORK=1 -DUSE_VFORK=0 -DUSE_DAEMON=0 -DUSE_PTHREAD_SETSCHEDPRIO=1 -DGCC_LINT -DHAVE_LIBCAP_NG \
	-I. -I../../OBJ.linux../lib/libswan -I../../include -I/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/include/nss -I/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/include/nspr  \
	-DHERE_FILENAME=\"lib/libswan/version.c\" \
	-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6=libreswan-4.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto  -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/include -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/include/fortify -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/include  \
	-MF ../../OBJ.linux../lib/libswan/version.d \
	-MP -MMD -MT /usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6/OBJ.linux../lib/libswan/version.o \
	-o ../../OBJ.linux../lib/libswan/version.o \
	-c /usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6/OBJ.linux../lib/libswan/version.c
<command-line>: warning: "_FORTIFY_SOURCE" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "_FORTIFY_SOURCE" redefined
<command-line>: note: this is the location of the previous definition
/usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6/lib/libswan/unbound.c:34:10: fatal error: unbound-event.h: No such file or directory
   34 | #include <unbound-event.h>      /* see USE_UNBOUND_EVENT_H_COPY */
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[6]: *** [../../mk/rules.mk:63: unbound.o] Error 1
make[5]: *** [../../mk/targets.mk:82: all] Error 2
make[5]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6/lib/libswan'
make[4]: *** [../mk/targets.mk:82: recursive-all] Error 2
make[4]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6/lib'
make[3]: *** [/usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6/mk/targets.mk:82: recursive-all] Error 2
make[3]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6'
make[2]: *** [Makefile:123: /usr/src/openwrt/build_dir/target-x86_64_musl/libreswan-4.6/.built] Error 2
make[2]: Leaving directory '/usr/src/openwrt/feeds/packages/net/libreswan'
time: package/feeds/packages/libreswan/compile#2.14#1.50#3.03
    ERROR: package/feeds/packages/libreswan failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/libreswan/compile] Error 1
make[1]: Leaving directory '/usr/src/openwrt'
make: *** [/usr/src/openwrt/include/toplevel.mk:230: package/libreswan/compile] Error 2
```

lucize, 2022-03-06:
from what I see:

```
        config PACKAGE_libunbound_libevent
                bool "Build with expanded network resource (libevent) support."
                default y
```
so unless you had modified your config, this should work by default, this is not possible without making libunbound_libevent as a standalone package or I guess I could make libreswan-nolibunbound package variant

oskarirauta, 2022-03-08:
Yes, I opted it out in the configuration. You do not need to split packages; just make sure option is selected if user wants libreswan. For some reason I opted it out.. Can't remember why..

closed 2025-11-15 by GeorgeSapkin