openwrt/packages #17997: OpenWrt 22.0x branching in late March

hnyman, 2022-03-06:
OpenWrt will likely branch for the new 22.0x release in late March.

Packages/features/fixes to be included in the 22.0x branch (likely 22.03) should be merged well in advance of the branching.
Stuff that is merged after the branching will need to be backported to the new release branch.

PRs containing experimental stuff would preferably be merged to master **after the branching**, so that the new 22.0x release branch is as stable as possible.

See release related advice in:
* https://github.com/openwrt/openwrt/issues/9394
* https://lists.openwrt.org/pipermail/openwrt-devel/2022-February/038037.html
* https://lists.openwrt.org/pipermail/openwrt-devel/2022-March/038170.html

predators46, 2022-03-06:
@hnyman 

https://github.com/openwrt/openwrt/issues/9365

hnyman, 2022-03-20:
OpenWrt 22.03 has been branched today.

22.03 is now the current stable branch, 
21.02 is the old stable branch, and 
19.07 is supposed to be declared End-of-life soon.

predators46, 2022-03-20:
I didn't see the release 22.03

https://downloads.openwrt.org/releases/

nickberry17, 2022-03-21:
Can we get new supported devices into 22.03 or is it already in feature freeze?  i.e.: https://github.com/openwrt/openwrt/commit/7ea2f3d6e22b506d137cf255ae73ae498c176d1c

nickberry17, 2022-03-21:
> I didn't see the release 22.03
> 
> https://downloads.openwrt.org/releases/

See https://github.com/openwrt/openwrt/issues/9394 
>March 20th is not the release date of 22.0x, after the branch we'll give it more time to stabilize.

predators46, 2022-03-21:
> > I didn't see the release 22.03
> > https://downloads.openwrt.org/releases/
> 
> See [openwrt/openwrt#9394](https://github.com/openwrt/openwrt/issues/9394)
> 
> > March 20th is not the release date of 22.0x, after the branch we'll give it more time to stabilize.

Thanks

openwrtfirmwarecostum, 2022-03-30:
Nice

https://downloads.openwrt.org/releases/packages-22.03/

neheb, 2022-09-18:
Should this be unpinned?

closed 2023-12-03 by hnyman