openwrt/packages #18003: package ipset is missing for mvebu/cortexa9

ne20002, 2022-03-06:
WRT3200ACM with 21.02.2:

[image]

BKPepe, 2022-03-06:
Since you removed the issue template:

**Sorry, but posting screenshots is not enough. We don't have a crystal magic ball here to know what is going on, and even from the title, it seems you are wrong.** This is something to do with the kernel. Which version of kernel do you have, and which version does it want to install?

Would it be possible to use SSH, connect to your router, and use:
```
opkg update
opkg install ipset
```

Then post output here. Also, it would be nice to have output from this command:

```
cat /etc/os-release
```

This will help us to know if you are using any forked OpenWrt repository or not.

ne20002, 2022-03-06:
Sorry, I hoped that this may also pop up for other users.
```
root@home:~# opkg update
Downloading https://downloads.openwrt.org/releases/21.02.2/targets/mvebu/cortexa9/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/21.02.2/targets/mvebu/cortexa9/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.sig
Signature check passed.
root@home:~# opkg install ipset
Installing ipset (7.6-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.2/packages/arm_cortex-a9_vfpv3-d16/base/ipset_7.6-1_arm_cortex-a9_vfpv3-d16.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.179-1-a3723f227628558ea929b84a5ead5d14) for kmod-nf-reject
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-reject found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.179-1-a3723f227628558ea929b84a5ead5d14) for kmod-nf-ipt
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-ipt found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.179-1-a3723f227628558ea929b84a5ead5d14) for kmod-ipt-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.179-1-a3723f227628558ea929b84a5ead5d14) for kmod-nfnetlink
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nfnetlink found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.179-1-a3723f227628558ea929b84a5ead5d14) for kmod-ipt-ipset
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-ipset found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ipset:
 * 	kernel (= 5.4.179-1-a3723f227628558ea929b84a5ead5d14)
 * opkg_install_cmd: Cannot install package ipset.
root@home:~# 
root@home:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="21.02.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.2"
VERSION_ID="21.02.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16495-bf0c965af0"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3-d16"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.2 r16495-bf0c965af0"
root@home:~# 
```

It's a Linksys WRT3200ACM, upgraded from 21.02.1 to 21.02.2 with 'Attended Sysupgrade'.

ne20002, 2022-03-06:
Ok,
as this wasn't an obvious bug, I tried the following:
- made a backup of settings
- rebooted to alternate partition (which is still on 21.02.1)
- generated a new 'Attended Sysupgrade' and installed with keeping settings

Now the dependency is shown. So it seems it was a problem with the previous sysupgrade build

Sorry for inconvenience.

I'll close it.

closed 2022-03-06 by ne20002