openwrt/packages #18011: minidlna fails after ffmpeg update to 5.0

jglooije, 2022-03-07:
Maintainer: @\Ted Hess <thess@kitschensync.net>, Ian Leonard <antonlacon@gmail.com>
Environment: Openwrt trunk (latest 6 march 2022)

Description:

Minidlna fails after update ffmpeg v5.0, after reverting this patch minidlna was working again.

Error loading shared library libavformat.so.58: No such file or directory (needed by /usr/sbin/minidlnad)
Error loading shared library libavcodec.so.58: No such file or directory (needed by /usr/sbin/minidlnad)
Error loading shared library libavutil.so.56: No such file or directory (needed by /usr/sbin/minidlnad)
Error relocating /usr/sbin/minidlnad: av_strerror: symbol not found
Error relocating /usr/sbin/minidlnad: av_log_set_level: symbol not found
Error relocating /usr/sbin/minidlnad: av_reduce: symbol not found
Error relocating /usr/sbin/minidlnad: avformat_close_input: symbol not found
Error relocating /usr/sbin/minidlnad: avformat_find_stream_info: symbol not found
Error relocating /usr/sbin/minidlnad: av_dict_get: symbol not found
Error relocating /usr/sbin/minidlnad: avformat_open_input: symbol not foun
[minidlna.txt](https://github.com/openwrt/packages/files/8195674/minidlna.txt)
d

BKPepe, 2022-03-09:
Caused by this: https://github.com/openwrt/packages/pull/17968

ping: @graysky2 , @neheb 

cc: @diizzyy

neheb, 2022-03-11:
The issue is that ffmpeg 5.0 upgrades these to versions .59. Which means minidlna needs to be rebuilt.

gotthardanger, 2022-04-03:
Whats to do for rebuild minidlna? I am compiling my own image using build root for bananapi r2, but the resulting image contains old minidlna, but ffmpeg 5.x.