openwrt/packages #18035: banip: Country block don't work as expected

Nimrod-666, 2022-03-09:
Maintainer: @dibdot 
Environment: OpenWrt 21.02.2 on Linksys EA8300

Description:

Hi, i tried to set up the country blockling with banip but it don't works, or it don't works like expected. I would expect that it'll block outgoing and incoming http/https requests to countries that are blacklisted by banip.

So as a test i added poland unter "Country Selection" and reloaded the list. Then i looked after a well known polish website like "allegro.pl", made a dns lookup on the uri and geolocated the IP to be sure the webserver is in poland and not delivered on a CDN worldwide or something like that. But the IP was directly from poland and banip don't blocked the https traffic to "allegro.pl".

Do i miss something or did i misunderstood the function of the country blocking?

dibdot, 2022-03-09:
Most probably a configuration issue, provide your banIP config.

Nimrod-666, 2022-03-10:
Hi @dibdot, here's my config. Do you need something in particular?

`root@openwrt:~# uci show | grep banip
banip.global=banip
banip.global.ban_debug='0'
banip.global.ban_logsrc_enabled='0'
banip.global.ban_logdst_enabled='0'
banip.global.ban_autoblacklist='1'
banip.global.ban_autowhitelist='1'
banip.global.ban_nice='0'
banip.global.ban_maxqueue='4'
banip.global.ban_global_settype='src+dst'
banip.global.ban_target_src='DROP'
banip.global.ban_target_dst='REJECT'
banip.global.ban_loglimit='100'
banip.global.ban_proto4_enabled='1'
banip.global.ban_trigger='wan'
banip.global.ban_autodetect='0'
banip.global.ban_ifaces='guest' 'lan'
banip.global.ban_enabled='1'
banip.global.ban_monitor_enabled='1'
banip.global.ban_realtime='true'
banip.global.ban_mail_enabled='1'
banip.global.ban_mailreceiver='xxx.de'
banip.global.ban_mailsender='xxx.de'
banip.global.ban_mailtopic='banip notification'
banip.global.ban_mailprofile='default'
banip.global.ban_fetchutil='uclient-fetch'
banip.global.ban_sources='bogon' 'firehol1' 'greensnow' 'proxy' 'sslbl' 'talos' 'threat' 'tor'
banip.global.ban_countries='pl'
`

dibdot, 2022-03-10:
You haven't enabled the source "country" in the Sources Selection.

Nimrod-666, 2022-03-10:
Hi, oh my bad! But it's active now but for a tast i added poland and tried "allegro.pl" again, without any change. I can still reach the website. I made a traceroute and the first hop is my openwrt router. Please tell me what logs you need for a debugging.

dibdot, 2022-03-10:
Most probably you've misconfigured your banIP interfaces too, usually that's your wan interface(s) (and not lan or guest etc.). Maybe you should start with the interface autodetection and go from there ...

Nimrod-666, 2022-03-11:
Hi, i tried the autodetection, but it pointed to my "guest" wifi then only. Then i set manually to "wan" only and now it works fine. Thank you!

closed 2022-03-11 by Nimrod-666

Catfriend1, 2023-03-05:
Example that works:
```

config banip 'global'
	option ban_enabled '1'
	option ban_debug '0'
	option ban_autodetect '1'
	list ban_logterm 'Exit before auth from'
	list ban_logterm 'luci: failed login'
	list ban_logterm 'error: maximum authentication attempts exceeded'
	list ban_logterm 'sshd.*Connection closed by.*\[preauth\]'
	list ban_logterm 'SecurityEvent=\"ChallengeResponseFailed\".*RemoteAddress='
	list ban_dev 'wan'
	option ban_logsrc_enabled '0'
	option ban_logdst_enabled '0'
	option ban_autoblacklist '1'
	option ban_autowhitelist '1'
	option ban_nice '0'
	option ban_maxqueue '4'
	option ban_global_settype 'src'
	option ban_target_src 'DROP'
	option ban_target_dst 'DROP'
	option ban_loglimit '100'
	option ban_proto4_enabled '1'
	option ban_monitor_enabled '1'
	option ban_realtime 'true'
	option ban_mail_enabled '0'
	option ban_fetchcmd 'curl'
	list ban_countries 'ru'
	list ban_countries 'in'
	list ban_countries 'mx'
	list ban_countries 'jp'
	list ban_countries 'kr'
	list ban_countries 'cn'
	list ban_countries 'ro'
	list ban_countries 'pa'
	list ban_countries 'md'
	list ban_countries 'gr'
	list ban_countries 'cl'
	list ban_countries 'ie'
	list ban_countries 'se'
	list ban_countries 'ir'
	list ban_countries 'nl'
	list ban_countries 'de'
	list ban_countries 'gb'
	list ban_countries 'tw'
	list ban_countries 'by'
	list ban_countries 'rs'
	list ban_countries 'bg'
	list ban_countries 'cu'
	list ban_countries 'sy'
	list ban_countries 'ni'
	list ban_countries 've'
	list ban_countries 'iq'
	list ban_countries 'ly'
	list ban_countries 'af'
	list ban_countries 'kp'
	list ban_countries 'vn'
	list ban_countries 'la'
	list ban_settype_all 'blacklist'
	list ban_settype_all 'asn'
	list ban_settype_all 'bogon'
	list ban_settype_all 'cinsscore'
	list ban_settype_all 'country'
	list ban_settype_all 'darklist'
	list ban_settype_all 'debl'
	list ban_settype_all 'doh'
	list ban_settype_all 'drop'
	list ban_settype_all 'edrop'
	list ban_settype_all 'feodo'
	list ban_settype_all 'firehol2'
	list ban_settype_all 'firehol3'
	list ban_settype_all 'firehol4'
	list ban_settype_all 'greensnow'
	list ban_settype_all 'myip'
	list ban_settype_all 'nixspam'
	list ban_settype_all 'proxy'
	list ban_settype_all 'sslbl'
	list ban_settype_all 'talos'
	list ban_settype_all 'threat'
	list ban_settype_all 'tor'
	list ban_settype_all 'torrelay'
	list ban_settype_all 'voip'
	list ban_settype_all 'yoyo'
	list ban_settype_all 'countries'
	list ban_sources 'bogon'
	list ban_sources 'cinsscore'
	list ban_sources 'country'
	list ban_sources 'darklist'
	list ban_sources 'debl'
	list ban_sources 'drop'
	list ban_sources 'edrop'
	list ban_sources 'feodo'
	list ban_sources 'firehol2'
	list ban_sources 'firehol3'
	list ban_sources 'firehol4'
	list ban_sources 'greensnow'
	list ban_sources 'myip'
	list ban_sources 'nixspam'
	list ban_sources 'proxy'
	list ban_sources 'sslbl'
	list ban_sources 'talos'
	list ban_sources 'threat'
	list ban_sources 'tor'
	list ban_sources 'voip'
	list ban_sources 'yoyo'
	list ban_sources 'countries'
```