openwrt/packages #18039: OpenLDAP 2.6 libldap_r.so is missing

predators46, 2022-03-10:
I found a solution. did openwrt fix this?

https://github.com/python-ldap/python-ldap/issues/432

jefferyto, 2022-03-10:
It appears that is a workaround for python-ldap - since there is no python-ldap package here currently, there is nothing to work around for.

predators46, 2022-03-10:
> It appears that is a workaround for python-ldap - since there is no python-ldap package here currently, there is nothing to work around for.

I mean openldap must be repaired because when the process of making openldap in openwrt using openldap-2.6 libldap_r.so does not exist. all this openldap problem in version 2.5 and 2.6. I include the link for openldap repair reference

jefferyto, 2022-03-10:
Why do you need libldap_r.so? What software are you using that needs this?

closed 2022-03-12 by predators46