openwrt/packages #18053: IPROUTE2:  ERROR: package/network/utils/iproute2 failed to build (build variant: tcfull).  LIBBPF not found

okibcn, 2022-03-13:
Maintainer:  @guidosarducci  @hauke 
Environment: 
    PROFILE="linksys_e8450-ubi"
    RELEASE="snapshots "
    TARGET="mediatek"
    ARCH="mt7622"

Description:

a package (openvpn maybe?) adds iproute2-tcfull module that requires the undetected library libbpf. The build process fails with this message:
```
make[4]: Entering directory '/home/oki/kk/myopenwrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/iproute2-tcfull/iproute2-5.15.0'
TC schedulers
 ATM    no
 IPT    using xtables
 IPSET  yes

iptables modules directory: /usr/lib/iptables
libc has setns: yes
libc has name_to_handle_at: yes
SELinux support: no
libbpf support: no
        LIBBPF_FORCE=on set, but couldn't find a usable libbpf
make[4]: *** [Makefile:84: config] Error 1
make[4]: Leaving directory '/home/oki/kk/myopenwrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/iproute2-tcfull/iproute2-5.15.0'
make[3]: *** [Makefile:271: /home/oki/kk/myopenwrt/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/iproute2-tcfull/iproute2-5.15.0/.built] Error 2
make[3]: Leaving directory '/home/oki/kk/myopenwrt/openwrt/package/network/utils/iproute2'
time: package/network/utils/iproute2/tcfull/compile#0.41#0.07#0.41
    ERROR: package/network/utils/iproute2 failed to build (build variant: tcfull).
make[2]: *** [package/Makefile:119: package/network/utils/iproute2/compile] Error 1
make[2]: Leaving directory '/home/oki/kk/myopenwrt/openwrt'
make[1]: *** [package/Makefile:110: /home/oki/kk/myopenwrt/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/oki/kk/myopenwrt/openwrt'
make: *** [/home/oki/kk/myopenwrt/openwrt/include/toplevel.mk:230: world] Error 2
```

This is the diff .config used for this build run:
```
CONFIG_DEVEL=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUILD_LOG=y
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVEHISTORY=y
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVE_ON_EXIT=y
CONFIG_BUSYBOX_CONFIG_FEATURE_LESS_FLAGS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_LESS_REGEXP=y
CONFIG_BUSYBOX_CONFIG_FEATURE_LESS_WINCH=y
CONFIG_BUSYBOX_DEFAULT_STAT=y
# CONFIG_BUSYBOX_DEFAULT_FEATURE_STAT_FORMAT is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_STAT_FILESYSTEM is not set
CONFIG_DEBUG=y
CONFIG_DROPBEAR_ECC=y
CONFIG_IMAGEOPT=y
CONFIG_INCLUDE_CONFIG=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
# CONFIG_LUCI_JSMIN is not set
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
[... 204 lines cut ...]
The libbpf was properly sourced and a simple `find . -name "*libbpf**"` on the `openwrt `folder returns this:
```
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/iproute2-tcfull/iproute2-5.15.0/lib/bpf_libbpf.c
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf.c
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf_probes.c
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf.h
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf_util.h
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf.pc.template
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf_errno.c
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf.map./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf_common.h
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/lib/bpf/libbpf_internal.h
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.103/tools/build/feature/test-libbpf.c
./build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/iproute2-iptiny/iproute2-5.15.0/lib/bpf_libbpf.c
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf.c
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf_probes.c
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf.h
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf_util.h
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf.pc.template./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf_errno.c
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf.map
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf_common.h
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/lib/bpf/libbpf_internal.h
./build_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/linux-5.10.103/tools/build/feature/test-libbpf.c
```

So, most probably it is again a detection problem.

okibcn, 2022-03-14:
The failure is random. I thought the number of threads could play a role in that, but the failure happens even when using a single thread (-j1).

However, if just after the failure and without touching any file I do:
`make -j1`
or even 
`make -j(nproc)`

Then, most of the time completes the process and the image is successfully created.

solidus1983, 2022-03-15:
I had a similar issue myself with Qosify when i was trying to compile it.
Although it did download llvm-project-13.0.0.src.tar.xz it wouldn't compile because it didn't download also llvm-bpf-13.0.0.Linux-x86_64 and extract it to the ```/staging_dir/host/``` directory. Took me a week to figure what was going on, if it wasn't because i noticed that the symlink was broken in ```/staging_dir/hosts/``` i'd still be trying to fix it with clean installs of the OS and rebuilding the build enviroment.

okibcn, 2022-03-17:
Well, I was trying to isolate the issue by creating a clean openwrt structure, this is, taking the `.config `in the opening post and doing a:
```
make defconfig && make toolchain/install
rm -rf bin tmp build_dir
make defconfig

```
under this situation, I test the problems with packages, missing dependencies most of the time. So I do a
```
make V=sc package/network/utils/iprpute2/compile
```

and I got this error output:
```
make[2]: Entering directory '/build/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/build/openwrt/scripts/config'
time: target/linux/prereq#0.11#0.01#0.11
make[1]: Entering directory '/build/openwrt'
make[2]: Entering directory '/build/openwrt/package/libs/toolchain'
touch /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.prepared_460486047fa77a68464b12034d1cfc84_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain
touch /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.prepared_460486047fa77a68464b12034d1cfc84_6664517399ebbbc92a37c5bb081b5c53
rm -f /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.configured_*
rm -f /build/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.toolchain_installed
(cd /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/./; if [ -x ./configure ]; then find /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/ -name config.guess | xargs -r chmod u+w; find /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /build/openwrt/scripts/config.guess; find /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/ -name config.sub | xargs -r chmod u+w; find /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /build/openwrt/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=cortex-a53 -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/usr/include -I/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/include/fortify -I/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/include " LDFLAGS="-L/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/usr/lib -L/build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/lib -znow -zrelro "   ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.built
touch /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.built_check
cp -fpR /build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/lib/libgcc_s.so.1 /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/
touch /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.built
rm -rf /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.pkgdir/libc.installed /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.pkgdir/libc
mkdir -p /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.pkgdir/libc
install -d -m0755 /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.pkgdir/libc/lib /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.pkgdir/libc/usr/bin
cp -fpR /build/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.2.0_musl/lib/ld-musl-*.so* /build/openwrt/build_dir/target-aarch64_cortex-a53_musl/toolchain/.pkgdir/libc/lib/
[... cut ...]
time: package/kernel/gpio-button-hotplug/compile#0.14#0.00#0.12
    ERROR: package/kernel/gpio-button-hotplug failed to build.
make[1]: *** [package/Makefile:116: package/kernel/gpio-button-hotplug/compile] Error 1
make[1]: Leaving directory '/build/openwrt'
make: *** [/build/openwrt/include/toplevel.mk:230: package/network/utils/iproute2/compile] Error 2
make[1]: Leaving directory '/build/openwrt'
```

I am lost here as it seems a missing dependency somewhere, but I can't track it. Any help is welcome.

erdoukki, 2022-03-17:
https://github.com/openwrt/packages/issues/17506

> A quick fix with a single command;
> `make {toolchain,target,package/fail2ban}/compile`

okibcn, 2022-03-17:
This issue is related to the main OpenWRT repo. I have created the [issue](https://github.com/openwrt/openwrt/issues/9491) in the right place.

This issue should be closed to avoid duplicities.

closed 2022-03-17 by okibcn