openwrt/packages #18082: curl: wolfssl build without IPv6 is broken

c---, 2022-03-17:
Maintainer: @stangri 
Environment:
OpenWrt 21.02.2
mvebu-cortexa9-linksys_wrt3200acm

Description:
The 7.82 release of curl is broken when using wolfssl and IPv6 is disabled. It's patched in the latest curl source but not in the release.

Here is a patch for the 7.82 code:

```patch
--- curl-7.82.0.orig/lib/vtls/wolfssl.c
+++ curl-7.82.0.new/lib/vtls/wolfssl.c
@@ -462,9 +462,9 @@
     const char * const hostname = SSL_HOST_NAME();
     size_t hostname_len = strlen(hostname);
     if((hostname_len < USHRT_MAX) &&
-       (0 == Curl_inet_pton(AF_INET, hostname, &addr4)) &&
+       (0 == Curl_inet_pton(AF_INET, hostname, &addr4))
 #ifdef ENABLE_IPV6
-       (0 == Curl_inet_pton(AF_INET6, hostname, &addr6))
+       && (0 == Curl_inet_pton(AF_INET6, hostname, &addr6))
 #endif
       ) {
       size_t snilen;
```

stangri, 2022-03-17:
Appreciate the patch. Can you elaborate on what's broken so I can try to reproduce?

c---, 2022-03-17:
```shell
make menuconfig
# Disable IPv6
Global build settings ---> Enable IPv6 support in packages ---> N
make
```
Build fails with a C syntax error in `lib/vtls/wolfssl.c`. Sorry I don't have a log available at the moment.

You can see in the patch it moves the `&&` inside the #ifdef test for IPv6. Otherwise the && is invalid because the next character is `)`.

usver1, 2022-03-17:
log: 
OpenWrt-libtool: compile:  mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I../include -I../lib -I../lib -DBUILDING_LIBCURL -DCURL_HIDDEN_SYMBOLS -isystem /mnt/21g/21g/staging_dir/target-mips_24kc_musl/usr/include/wolfssl -isystem /mnt/21g/21g/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/usr/include -isystem /mnt/21g/21g/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include/fortify -isystem /mnt/21g/21g/staging_dir/toolchain-mips_24kc_gcc-8.4.0_musl/include -DCURL_DISABLE_NTLM -isystem /mnt/21g/21g/staging_dir/target-mips_24kc_musl/usr/include -fvisibility=hidden -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/mnt/21g/21g/build_dir/target-mips_24kc_musl/curl-7.82.0=curl-7.82.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -ffunction-sections -fdata-sections -Werror-implicit-function-declaration -Wno-system-headers -MT vtls/libcurl_la-wolfssl.lo -MD -MP -MF vtls/.deps/libcurl_la-wolfssl.Tpo -c vtls/wolfssl.c  -fPIC -DPIC -o vtls/.libs/libcurl_la-wolfssl.o
vtls/wolfssl.c: In function 'wolfssl_connect_step1':
vtls/wolfssl.c:469:7: error: expected expression before ')' token
       ) {
       ^
vtls/wolfssl.c:471:57: error: 'snilen' undeclared (first use in this function); did you mean 'strlen'?
       char *snihost = Curl_ssl_snihost(data, hostname, &snilen);
                                                         ^~~~~~
                                                         strlen
vtls/wolfssl.c:471:57: note: each undeclared identifier is reported only once for each function it appears in
make[5]: *** [Makefile:2663: vtls/libcurl_la-wolfssl.lo] Error 1

stangri, 2022-03-17:
> ```shell
> make menuconfig
> # Disable IPv6
> Global build settings ---> Enable IPv6 support in packages ---> N
> Libraries ---> libcurl ---> Selected SSL library ---> wolfSSL
> make
> ```
> 
> Build fails with a C syntax error in `lib/vtls/wolfssl.c`. Sorry I don't have a log available at the moment.
> 
> You can see in the patch it moves the `&&` inside the #ifdef test for IPv6. Otherwise the && is invalid because the next character is `)`.

Thanks again for your prompt reply. Any chance you can test it on snapshots to determine if the patch is needed on both branches or just the release?

Do you want to send the PR (just for the master branch if both are affected, I can then cherry-pick it for release)?

BKPepe, 2022-03-17:
Actually, this is the commit in the curl: https://github.com/curl/curl/commit/471d5f44c5189bd78f8839ce8ac1b4b14f4b7d61

and those changes are different than in OP.

stangri, 2022-03-17:
@BKPepe should we revert stable to 7.81 for now until the upstream patch makes it to the next release?

BKPepe, 2022-03-17:
Recently, I remember some discussion in openwrt-devel mailing list about disable IPv6 option and it seems that some users are using it. My humble opinion on this thing is that we don't need to revert it to the stable version because it does not affect everyone. We can simply backport the patch, and that's it. The user will be happy, and we will be satisfied as well. :-)

But I don't experience this issue since I don't disable IPv6, so it is up to you. If you want to wait for a new release or wait if the OP will create pull request with the patch from upstream, that's fine for my side, too.

c---, 2022-03-17:
master is affected as well. Yes, that patch could be pulled directly from the curl project as it does the exact same thing as the patch I posted above (they reworded the logic whereas mine is directly against the 7.82 source base).

I can create a PR if needed?

stangri, 2022-03-17:
> master is affected as well. Yes, that patch could be pulled directly from the curl project as it does the exact same thing as the patch I posted above (they reworded the logic whereas mine is directly against the 7.82 source base).
> 
> I can create a PR if needed?

If you could send a PR just for the master with the upstream patch that @BKPepe linked, I'd appreciate that.

neheb, 2022-03-18:
@BKPepe consensus on IRC was to make IPV6 mandatory after branching.

PR #18091 (2022-03-18): curl: Fix compiling curl wolfSSL when IPv6 is disabled
Fixes #18082

Maintainer: @stangri 
Compile tested: master mvebu-cortexa9-linksys_wrt3200acm
commit 4eb08bac (2022-03-18): curl: Fix compiling curl wolfSSL IPv6 disabled

Fixes #18082
PR #18097 (2022-03-18): [21.02] curl: Fix compiling curl wolfSSL IPv6 disabled
Fixes #18082

(cherry picked from commit 4eb08bacf309d336f03ad001194b9c8c1847ac2b)
commit f01cf663 (2022-03-18): curl: Fix compiling curl wolfSSL IPv6 disabled

Fixes #18082

(cherry picked from commit 4eb08bacf309d336f03ad001194b9c8c1847ac2b)
closed 2022-03-18 by stangri