openwrt/packages #18100: coova-chilli: error compile with redir and openssl

mikysal78, 2022-03-18:
For openwrt 19.07 and ar71xx.

If install only coova is ok, if add:
```
CONFIG_COOVACHILLI_REDIR=y
CONFIG_COOVACHILLI_OPENSSL=y
```
i have error:
```
OpenWrt-libtool: compile:  mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I../json -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/include -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/usr/include -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include/fortify -I/home/jenkins/workspace/Firmware/source/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/include -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/coova-chilli-1.4:coova-chilli-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -Wno-address-of-packed-member -MT options.lo -MD -MP -MF .deps/options.Tpo -c options.c  -fPIC -DPIC -o .libs/options.o
options.c: In function 'options_fromfd':
options.c:385:48: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_host.allocated)
                                                ^
options.c:387:48: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_path.allocated)
                                                ^
options.c:389:46: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_qs.allocated)
                                              ^
options.c: At top level:
cc1: error: unrecognized command line option '-Wno-address-of-packed-member' [-Werror]
cc1: all warnings being treated as errors
make[7]: *** [Makefile:936: options.lo] Error 1
make[7]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/coova-chilli-1.4/src'
make[6]: *** [Makefile:979: all-recursive] Error 1
make[6]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/coova-chilli-1.4/src'
make[5]: *** [Makefile:416: all-recursive] Error 1
make[5]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/coova-chilli-1.4'
make[4]: *** [Makefile:344: all] Error 2
make[4]: Leaving directory '/home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/coova-chilli-1.4'
make[3]: *** [Makefile:150: /home/jenkins/workspace/Firmware/source/build_dir/target-mips_24kc_musl/coova-chilli-1.4/.built] Error 2
make[3]: Leaving directory '/home/jenkins/workspace/Firmware/source/feeds/packages/net/coova-chilli'
time: package/feeds/packages/coova-chilli/compile#8.05#0.43#8.28
make[2]: *** [package/Makefile:113: package/feeds/packages/coova-chilli/compile] Error 2
make[2]: Leaving directory '/home/jenkins/workspace/Firmware/source'
make[1]: *** [package/Makefile:107: /home/jenkins/workspace/Firmware/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jenkins/workspace/Firmware/source'
make: *** [/home/jenkins/workspace/Firmware/source/include/toplevel.mk:227: world] Errore 2
```

teslamint, 2022-03-21:
That was patched on 1.5. See: https://github.com/coova/coova-chilli/pull/453

mikysal78, 2022-03-21:
> That was patched on 1.5. See: [coova/coova-chilli#453](https://github.com/coova/coova-chilli/pull/453)

How-to compile firmware with coova 1.5? in this repo is 1.4.
Thanks.

teslamint, 2022-03-23:
edit feeds.conf to update package feed.
See: https://openwrt.org/docs/guide-developer/feeds

BKPepe, 2022-03-23:
In this repo, there are multiple versions to be more correct. It depends on used OpenWrt versions. And to be even more precise, it is not a good idea to even suggest that users with OpenWrt 19.07 core should switch to using packages repository for OpenWrt 21.02. It is even rather better to switch it overall at once. 

Anyway, I cherry-picked several commits from OpenWrt 21.02 branch to OpenWrt 19.07 related to coova-chilli. You might want to update your feeds now and try it once more.

mikysal78, 2022-03-28:
> In this repo, there are multiple versions to be more correct. It depends on used OpenWrt versions. And to be even more precise, it is not a good idea to even suggest that users with OpenWrt 19.07 [...]

it's ok.
Thanks.

closed 2022-03-28 by mikysal78