openwrt/packages #18110: LIBOPING: fails to compile. Related to new kernel?

okibcn, 2022-03-20:
Maintainer: @jow-  (Jo-Philipp Wich)
Environment: latest snapshot, config file attached below.

## Description:
When building the attached [.config](https://github.com/openwrt/packages/files/8310584/config.txt) file below, the build **always** fails at [packages/libs/liboping](https://github.com/openwrt/packages/tree/master/libs/liboping). This started to happen today, as this config has been building for several days until today March 19 2022. The same day of the new kernel upgrade from 5.10.104 to 5.10.107. Not sure if it is related to not. but I haven't found any other relevant change since the last successful build. This package is a dependency of `collectd-mod-ping`.

This is the console dump showing the error:
```make
make: Entering directory '/build/myopenwrt/R7800/feeds/packages/libs/liboping'
rm -f /build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0/.built
touch /build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0/.built_check
CFLAGS="-Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0=liboping-1.10.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic  -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include " CXXFLAGS="-Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0=liboping-1.10.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic  -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include " LDFLAGS="-L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -znow -zrelro " make  -C /build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0=liboping-1.10.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" ;
make[1]: Entering directory '/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0'
Making all in src
make[2]: Entering directory '/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0/src'
make  all-recursive
make[3]: Entering directory '/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0/src'
Making all in mans
make[4]: Entering directory '/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0/src/mans'
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0/src/mans'
make[4]: Entering directory '/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0/src'
/bin/bash ../libtool  --tag=CC   --mode=link arm-openwrt-linux-muslgnueabi-gcc -Wall -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0=liboping-1.10.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic   -version-info 3:0:3 -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -znow -zrelro  -o liboping.la -rpath /usr/lib liboping_la-liboping.lo
OpenWrt-libtool: link: arm-openwrt-linux-muslgnueabi-gcc -shared  -fPIC -DPIC  .libs/liboping_la-liboping.o   -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib  -Os -mfloat-abi=hard -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro   -Wl,-soname -Wl,liboping.so.0 -o .libs/liboping.so.0.3.0
OpenWrt-libtool: link: (cd ".libs" && rm -f "liboping.so.0" && ln -s "liboping.so.0.3.0" "liboping.so.0")
OpenWrt-libtool: link: (cd ".libs" && rm -f "liboping.so" && ln -s "liboping.so.0.3.0" "liboping.so")
OpenWrt-libtool: link: arm-openwrt-linux-muslgnueabi-gcc-ar cru .libs/liboping.a  liboping_la-liboping.o
/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.2.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar: `u' modifier ignored since `D' is the default (see `U')
OpenWrt-libtool: link: arm-openwrt-linux-muslgnueabi-gcc-ranlib .libs/liboping.a
OpenWrt-libtool: link: ( cd ".libs" && rm -f "liboping.la" && ln -s "../liboping.la" "liboping.la" )
/bin/bash ../libtool  --tag=CC   --mode=link arm-openwrt-linux-muslgnueabi-gcc -Wall -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0=liboping-1.10.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic   -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -znow -zrelro  -o oping oping.o liboping.la -lm
OpenWrt-libtool: link: arm-openwrt-linux-muslgnueabi-gcc -Wall -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0=liboping-1.10.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -DPIC -fpic -znow -zrelro -o .libs/oping oping.o  -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/lib ./.libs/liboping.so -lm
arm-openwrt-linux-muslgnueabi-gcc -DHAVE_CONFIG_H -I.   -DUSE_NCURSES=1  -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/build/myopenwrt/R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.2.0_musl_eabi/include  -Wall -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/build/myopenwrt/R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/liboping-1.10.0=l [... cut ...]
git clone -b master https://github.com/openwrt/openwrt.git openwrt
cd openwrt
./scripts/feeds update -a
./scripts/feeds install -a
```
2. copy the [.config](https://github.com/openwrt/packages/files/8310584/config.txt) file to the openwrt folder.
3. Try to build the image with:
```bash
make defconfig
make
```
To speedup the failure, this alternative process would bring the same issue faster:
```bash
make defconfig
make -j16 {toolchain,target}/compile
make -j1 V=sc feeds/packages/libs/liboping/compile
```

Vladdrako, 2022-03-20:
[Related.](https://github.com/octo/liboping/issues/62)
Fixed with patch
<pre>
--- liboping/src/oping.c
+++ liboping/src/oping.c
@@ -1125,7 +1125,7 @@
 			wattron (ctx->window, COLOR_PAIR(color));

 		if (has_utf8())
-			mvwprintw (ctx->window, /* y = */ 3, /* x = */ x + 2, symbol);
+			mvwprintw (ctx->window, /* y = */ 3, /* x = */ x + 2, "%s", symbol);
 		else
 			mvwaddch (ctx->window, /* y = */ 3, /* x = */ x + 2, symbolc);

@@ -1223,7 +1223,7 @@
 			mvwaddch (ctx->window, /* y = */ 3, /* x = */ x + 2, ' ');
 		else if (has_utf8 ())
 			mvwprintw (ctx->window, /* y = */ 3, /* x = */ x + 2,
-					hist_symbols_utf8[index]);
+					"%s", hist_symbols_utf8[index]);
 		else
 			mvwaddch (ctx->window, /* y = */ 3, /* x = */ x + 2,
 					hist_symbols_acs[index] | A_ALTCHARSET);
@@ -1601,7 +1601,7 @@
 			HOST_PRINTF ("%zu bytes from %s (%s): icmp_seq=%u ttl=%i ",
 					data_len, context->host, context->addr,
 					sequence, recv_ttl,
-					format_qos (recv_qos, recv_qos_str, sizeof (recv_qos_str)));
+					sequence, recv_ttl);
 			if ((recv_qos != 0) || (opt_send_qos != 0))
 			{
 				HOST_PRINTF ("qos=%s ",
</pre>

ldir-EDB0, 2022-03-20:
Not new kernel - is due to the ncurses 6.3

okibcn, 2022-03-20:
@Vladdrako , your patch works fine. Are you going to do a pull request to the packages repo?

PR #18116 (2022-03-20): liboping: fix compilation with ncurses 6.3
Fix compilation with ncurses 6.3 by adjusting the printf format strings.

Apparently ncurses 6.3 introduced some new formatting tweaks that broke things.

Fixes #18110

Reference to:
* https://github.com/octo/liboping/issues/62
* https://github.com/octo/liboping/pull/61
* https://salsa.debian.org/debian/liboping/-/blob/debian/debian/patches/fix_HOST_PRINTF_format_string.patch

Maintainer:  @jow- 

Runtested with mt7622/RT3200
commit cb4d8ccb (2022-03-20): liboping: fix compilation with ncurses 6.3

Fix compilation with ncurses 6.3 by adjusting the printf format strings.

Apparently ncurses 6.3 introduced some new formatting tweaks that
broke things.

Fixes #18110

Reference to:
* https://github.com/octo/liboping/issues/62
* https://github.com/octo/liboping/pull/61
* https://salsa.debian.org/debian/liboping/-/blob/debian/debian/patches/fix_HOST_PRINTF_format_string.patch
commit a83b1f64 (2022-03-20): liboping: fix compilation with ncurses 6.3

Fix compilation with ncurses 6.3 by adjusting the printf format strings.

Apparently ncurses 6.3 introduced some new formatting tweaks that
broke things.

Fixes #18110

Reference to:
* https://github.com/octo/liboping/issues/62
* https://github.com/octo/liboping/pull/61
* https://salsa.debian.org/debian/liboping/-/blob/debian/debian/patches/fix_HOST_PRINTF_format_string.patch

(cherry picked from commit cb4d8ccb2715c33acfcfe987545f9e899056e18f)
closed 2022-03-20 by hnyman